wolfSSL/wolfssl #6767: Eliptic → Elliptic?

DimitriPapadopoulos, 2023-09-12:
### Version

master

### Description

Is the typo, `Eliptic` instead of `Elliptic`, on purpose in this line and other similar lines of [certs/ecc/genecc.sh](https://github.com/wolfSSL/wolfssl/blob/master/certs/ecc/genecc.sh)?
https://github.com/wolfSSL/wolfssl/blob/61f48eb5a11391ff8d6c8f308cd9f7070f40de6b/certs/ecc/genecc.sh#L35

anhu, 2023-09-15:
Hi @DimitriPapadopoulos , 

Thank you so much for finding this mistake.  I'm sorry my response took so long. I will look into the best way to fix it. It is probably a simple search and replace and then re-running the scripts. 

Warm regards, Anthony

anhu, 2023-09-26:
Hi @DimitriPapadopoulos , 
Sorry for the delayed response time on this issue. I've been away on business travel to trade shows.  You would think it would be a simple search and replace but that is not the case.  there are places where things are hard coded in our test that need to be manually fixed. 

Warm regards, Anthony

closed 2023-09-26 by JacobBarthelmeh