wolfSSL/wolfssl #6797: ./configure fails: cannot find input file: `Makefile.in'`

BlackMage2, 2023-09-21:
### Version

5.6.3

### Description

`./configure` fails with `config.status: error: cannot find input file: `Makefile.in'`

BlackMage2, 2023-09-21:
ok, it builds when i configure with CMake

but what version of autoconf / autoreconf / automake is required alternate?

kareem-wolfssl, 2023-09-21:
Hi @BlackMage2 ,

How did you get wolfSSL 5.6.3?  If you've cloned it from our github, try cloning it again.
Did you run ./autogen.sh?  I would recommend running it again even if you already did and seeing if it helps.

As for autoconf versions, Ubuntu 22.04 is new enough that this shouldn't be an issue.  We do test wolfSSL on Ubuntu 22.04.

BlackMage2, 2023-09-21:
from [wolfssh-1.4.13-stable.tar.gz](https://codeload.github.com/wolfSSL/wolfssh/tar.gz/refs/tags/v1.4.13-stable)

kareem-wolfssl, 2023-09-21:
You will need to download and build wolfSSL before building wolfSSH, did you already do this?
You can find our latest wolfSSL release here: https://github.com/wolfSSL/wolfssl/archive/refs/tags/v5.6.3-stable.tar.gz
To support wolfSSH, build wolfSSL with: ./configure --enable-wolfssh

BlackMage2, 2023-09-21:
is wolfSSL with openSSL compatible?

kareem-wolfssl, 2023-09-22:
Hi @BlackMage2 ,

In that case, please try rerunning ./autogen.sh and let me know if it helps.

Yes, we do offer an OpenSSL compatibility layer which you can enable with the flag `--enable-opensslextra`.  CyaSSL is the old name for wolfSSL, we still have some legacy CyaSSL files for compatibility.