wolfSSL/wolfssl #6838: pthread related failure during build on MSYS2

shanoaice, 2023-10-05:
### Version

5.8.3-stable and master

### Description

The configure flags I am using are `--enable-quic --enable-session-ticket --enable-earlydata --enable-psk --enable-harden --enable-altcertchains`. `./configure` script works fine, but `make` gives me errors about pthread mutex and stuff:
```
In file included from ../tests/api.c:134:
../tests/api.c: In function 'test_server_nofail':
../tests/api.c:5861:27: warning: implicit declaration of function 'pthread_mutex_lock' [-Wimplicit-function-declaration]
 5861 |         PTHREAD_CHECK_RET(pthread_mutex_lock(&opts->signal->mutex));
      |                           ^~~~~~~~~~~~~~~~~~
../wolfssl/test.h:362:29: note: in definition of macro 'PTHREAD_CHECK_RET'
  362 |         int _pthread_ret = (__VA_ARGS__);                            \
      |                             ^~~~~~~~~~~
../tests/api.c:5861:27: warning: nested extern declaration of 'pthread_mutex_lock' [-Wnested-externs]
 5861 |         PTHREAD_CHECK_RET(pthread_mutex_lock(&opts->signal->mutex));
      |                           ^~~~~~~~~~~~~~~~~~
../wolfssl/test.h:362:29: note: in definition of macro 'PTHREAD_CHECK_RET'
  362 |         int _pthread_ret = (__VA_ARGS__);                            \
      |                             ^~~~~~~~~~~
../tests/api.c:5861:59: error: 'tcp_ready' has no member named 'mutex'
 5861 |         PTHREAD_CHECK_RET(pthread_mutex_lock(&opts->signal->mutex));
      |                                                           ^~
../wolfssl/test.h:362:29: note: in definition of macro 'PTHREAD_CHECK_RET'
  362 |         int _pthread_ret = (__VA_ARGS__);                            \
      |
```
Complete errors are here: https://fars.ee/7n1Y

I confirm that I am using the latest winpthreads provided by MSYS2 (`mingw-w64-ucrt-x86_64-winpthreads-git`).

dgarske, 2023-10-06:
Duplicate of https://www.wolfssl.com/forums/topic2041-pthread-error-while-building-wolfssl-under-msys2.html

shanoaice, 2023-10-11:
The one on the forum is also me. Since it maybe more convienent to use GitHub issue I suppose I will post further Information here?

The log I included is from the latest stable release. https://fars.ee/8B-5 this log is the from the master branch on October 8th.

anhu, 2023-10-13:
Hello @shanoaice , I apologize for my delayed response both on our forums page and here. Have you tried building without applications and in single threaded mode?  What are you trying to accomplish?  I ask because I noticed you are encountering errors while building our tls benchmarking app which requires threading.  If you don't need threading and you don't need the tls benchmarking application, then those might be viable options.

shanoaice, 2023-10-15:
Oh, I just want to compile WolfSSL for curl http/3, then probably I don't need the benchmark suite?

lealem47, 2023-10-17:
Hi @shanoaice,

Thank you for this report. These issues should be addressed in https://github.com/wolfSSL/wolfssl/pull/6881. Let me know if anything else comes up.

Thanks,
Lealem @ wolfSSL

shanoaice, 2023-10-20:
> Hi @shanoaice,
> 
> Thank you for this report. These issues should be addressed in #6881. Let me know if anything else comes up.
> 
> Thanks, Lealem @ wolfSSL

Yes, I have tried to build with this PR and everything works smoothly. Thanks.

closed 2023-10-20 by shanoaice