wolfSSL/wolfssl #6852: [Bug]: haproxy: support for SSL_CTX_set_keylog_callback() in --enable-haproxy ?

wlallemand, 2023-10-09:
### Version

88d25036a005ba

### Description

We have a feature in haproxy which allows users to debug by dumping keys with SSL_CTX_set_keylog_callback().

It seems like the support for SSL_CTX_set_keylog_callback() is not recommended in WolfSSL where it seems to be enable in other SSL libraries (NSS, Openssl, awslc etc.) Is there any reason for that?

``` 
 --enable-keylog-export  Enable (DANGEROUS INSECURE) exporting TLS secrets to
                          an NSS keylog file (default: disabled)
```

I tried to build this feature for haproxy anyway but I was not able to.

### Reproduction steps

$ ./configure --prefix=/opt/wolfssl/ --enable-haproxy --enable-keylog-export

### Relevant log output

```shell
$ make -j8  
make -j9  all-recursive
make[1]: warning: -j9 forced in submake: resetting jobserver mode.
make[1]: Entering directory '/home/wla/projects/haproxy_tech/wolfssl'
make[2]: Entering directory '/home/wla/projects/haproxy_tech/wolfssl'
make[2]: warning: -j9 forced in submake: resetting jobserver mode.
  CC       wolfcrypt/benchmark/benchmark.o
  CC       wolfcrypt/src/src_libwolfssl_la-hmac.lo
  CC       wolfcrypt/src/src_libwolfssl_la-hash.lo
  CC       wolfcrypt/src/src_libwolfssl_la-cpuid.lo
  CC       wolfcrypt/src/src_libwolfssl_la-kdf.lo
  CC       wolfcrypt/src/src_libwolfssl_la-random.lo
  CC       wolfcrypt/src/src_libwolfssl_la-sp_int.lo
  CC       wolfcrypt/src/src_libwolfssl_la-sha256.lo
  CC       wolfcrypt/src/src_libwolfssl_la-rsa.lo
  CC       wolfcrypt/src/src_libwolfssl_la-aes.lo
  CC       wolfcrypt/src/src_libwolfssl_la-sha.lo
  CC       wolfcrypt/src/src_libwolfssl_la-sha512.lo
  CC       wolfcrypt/src/src_libwolfssl_la-sha3.lo
  CC       wolfcrypt/src/src_libwolfssl_la-logging.lo
  CC       wolfcrypt/src/src_libwolfssl_la-wc_port.lo
  CC       wolfcrypt/src/src_libwolfssl_la-error.lo
  CC       wolfcrypt/src/src_libwolfssl_la-wc_encrypt.lo
  CC       wolfcrypt/src/src_libwolfssl_la-signature.lo
  CC       wolfcrypt/src/src_libwolfssl_la-wolfmath.lo
  CC       wolfcrypt/src/src_libwolfssl_la-memory.lo
  CC       wolfcrypt/src/src_libwolfssl_la-dh.lo
  CC       wolfcrypt/src/src_libwolfssl_la-asn.lo
  CC       wolfcrypt/src/src_libwolfssl_la-coding.lo
  CC       wolfcrypt/src/src_libwolfssl_la-poly1305.lo
  CC       wolfcrypt/src/src_libwolfssl_la-md5.lo
  CC       wolfcrypt/src/src_libwolfssl_la-pwdbased.lo
  CC       wolfcrypt/src/src_libwolfssl_la-pkcs12.lo
  CC       wolfcrypt/src/src_libwolfssl_la-chacha.lo
  CC       wolfcrypt/src/src_libwolfssl_la-chacha20_poly1305.lo
  CC       wolfcrypt/src/src_libwolfssl_la-ecc.lo
  CC       src/libwolfssl_la-internal.lo
  CC       src/libwolfssl_la-wolfio.lo
  CC       src/libwolfssl_la-keys.lo
  CC       src/libwolfssl_la-ssl.lo
  CC       src/libwolfssl_la-tls.lo
  CC       src/libwolfssl_la-tls13.lo
  CC       src/libwolfssl_la-ocsp.lo
  CC       src/libwolfssl_la-crl.lo
src/tls.c:111:10: error: #warning The SHOW_SECRETS and WOLFSSL_SSLKEYLOGFILE options should only be used for debugging and never in a production environment [-Werror=cpp]
  111 |         #warning The SHOW_SECRETS and WOLFSSL_SSLKEYLOGFILE options should only be used for debugging and never in a production environment
      |          ^~~~~~~
  CC       wolfcrypt/test/test.o
  CC       examples/benchmark/tls_bench.o
  CC       examples/client/client-client.o
  CC       examples/echoclient/echoclient.o
  CC       examples/echoserver/echoserver.o
  CC       examples/server/server-server.o
  CC       examples/asn1/asn1.o
  CC       examples/pem/pem.o
  CC       wolfcrypt/test/testsuite_testsuite_test-test.o
  CC       examples/client/testsuite_testsuite_test-client.o
  CC       examples/echoclient/testsuite_testsuite_test-echoclient.o
cc1: all warnings being treated as errors
make[2]: *** [Makefile:7046: src/libwolfssl_la-tls.lo] Error 1
make[2]: *** Waiting for unfinished jobs....
make[2]: Leaving directory '/home/wla/projects/haproxy_tech/wolfssl'
make[1]: *** [Makefile:7823: all-recursive] Error 1
make[1]: Leaving directory '/home/wla/projects/haproxy_tech/wolfssl'
make: *** [Makefile:4661: all] Error 2
```

kareem-wolfssl, 2023-10-09:
Hi @wlallemand ,

We do support this, but we strongly recommend against using it in production as the SSL key log file allows an attacker to compromise security.  We only recommend using this option for debugging.  For your use case in haproxy, since you are gating this behind a runtime option, it is safe to enable this at build time in wolfSSL.
Looks like your build is getting stuck on -Wall.  Try adding `CFLAGS='-Wno-error=cpp'` in front of your ./configure command for wolfSSL, then make clean and make.

wlallemand, 2023-10-09:
I was able to build with this but it's not possible to build wolfSSL with -Werror this way unfortunately :/
It would be nice to have this feature within --enable-haproxy directly.

It looks like the SSL_CTX_set_keylog_callback() does not support the TLSv1.3 variables and only dumps the <= TLSv1.2 ones, unlike it is done with OpenSSL. Any chance it will appear in this standard callback instead of having to use wolfSSL_set_tls13_secret_cb() ?

kareem-wolfssl, 2023-10-12:
Hi @wlallemand ,

My colleague douzzer is working on resolving your first point in https://github.com/wolfSSL/wolfssl/pull/6861.  With this PR, you will be able to build with --enable-keylog-export and not run into an error.

For your second point, I will work on this, we do have some older work that integrated TLS 1.3 keylogging into SSL_CTX_set_keylog_callback but it was not upstreamed.

wlallemand, 2023-10-30:
Thanks, I'm now able to build successfully wolfssl with both `--enable-haproxy` and `--enable-keylog-export`, can we have the `--enable-keylog-export` by default in `--enable-haproxy` since that it works correctly? Thanks!

Good to know that you are goging to integrated the TLS1.3 support, it will be useful!

wlallemand, 2024-05-24:
Hello, Any update on this?

kareem-wolfssl, 2024-05-24:
Apologies, you are correct.  I will work on fixing this in the near future.

julek-wolfssl, 2025-12-18:
This was fixed in https://github.com/wolfSSL/wolfssl/pull/9171. Thank you for the report.