wolfSSL/wolfssl #6874: [Bug]: mingw cross-compilation fails on Linux

Fierelier, 2023-10-16:
### Version

v5.6.3-stable

### Description

* Target: Windows (i586)
* Build-environment: Devuan 5 (roughly equivalent to Debian 12)

I get 2 different exceptions when trying to compile. The first is:
```
In file included from wolfcrypt/src/wc_port.c:28:
wolfcrypt/src/wc_port.c: In function 'wc_ReadDirFirst':
./wolfssl/wolfcrypt/types.h:674:35: error: 'strncpy' output may be truncated copying between 0 and 255 bytes from a string of length 259 [-Werror=stringop-truncation]
  674 |         #define XSTRNCPY(s1,s2,n) strncpy((s1),(s2),(n))
      |                                   ^~~~~~~~~~~~~~~~~~~~~~
wolfcrypt/src/wc_port.c:635:13: note: in expansion of macro 'XSTRNCPY'
  635 |             XSTRNCPY(ctx->name + pathLen + 1,
      |             ^~~~~~~~
wolfcrypt/src/wc_port.c: In function 'wc_ReadDirNext':
./wolfssl/wolfcrypt/types.h:674:35: error: 'strncpy' output may be truncated copying between 0 and 255 bytes from a string of length 259 [-Werror=stringop-truncation]
  674 |         #define XSTRNCPY(s1,s2,n) strncpy((s1),(s2),(n))
      |                                   ^~~~~~~~~~~~~~~~~~~~~~
wolfcrypt/src/wc_port.c:788:13: note: in expansion of macro 'XSTRNCPY'
  788 |             XSTRNCPY(ctx->name + pathLen + 1,
      |             ^~~~~~~~
cc1: all warnings being treated as errors
make[2]: *** [Makefile:6351: wolfcrypt/src/src_libwolfssl_la-wc_port.lo] Error 1
```

I recklessly added `-Wno-error=stringop-truncation` to progress the build further. With this option, it fails like this:
```
In file included from tests/api.c:134:
tests/api.c: In function ‘test_server_nofail’:
tests/api.c:5861:27: error: implicit declaration of function ‘pthread_mutex_lock’ [-Werror=implicit-function-declaration]
 5861 |         PTHREAD_CHECK_RET(pthread_mutex_lock(&opts->signal->mutex));
      |                           ^~~~~~~~~~~~~~~~~~
./wolfssl/test.h:362:29: note: in definition of macro ‘PTHREAD_CHECK_RET’
  362 |         int _pthread_ret = (__VA_ARGS__);                            \
      |                             ^~~~~~~~~~~
tests/api.c:5861:27: error: nested extern declaration of ‘pthread_mutex_lock’ [-Werror=nested-externs]
 5861 |         PTHREAD_CHECK_RET(pthread_mutex_lock(&opts->signal->mutex));
      |                           ^~~~~~~~~~~~~~~~~~
./wolfssl/test.h:362:29: note: in definition of macro ‘PTHREAD_CHECK_RET’
  362 |         int _pthread_ret = (__VA_ARGS__);                            \
      |                             ^~~~~~~~~~~
tests/api.c:5861:59: error: ‘tcp_ready’ has no member named ‘mutex’
 5861 |         PTHREAD_CHECK_RET(pthread_mutex_lock(&opts->signal->mutex));
      |                                                           ^~
./wolfssl/test.h:362:29: note: in definition of macro ‘PTHREAD_CHECK_RET’
  362 |         int _pthread_ret = (__VA_ARGS__);                            \
      |                             ^~~~~~~~~~~
  CC       tests/unit_test-hash.o
tests/api.c:5863:27: error: implicit declaration of function ‘pthread_cond_signal’ [-Werror=implicit-function-declaration]
 5863 |         PTHREAD_CHECK_RET(pthread_cond_signal(&opts->signal->cond));
      |                           ^~~~~~~~~~~~~~~~~~~
./wolfssl/test.h:362:29: note: in definition of macro ‘PTHREAD_CHECK_RET’
  362 |         int _pthread_ret = (__VA_ARGS__);                            \
      |                             ^~~~~~~~~~~
tests/api.c:5863:27: error: nested extern declaration of ‘pthread_cond_signal’ [-Werror=nested-externs]
 5863 |         PTHREAD_CHECK_RET(pthread_cond_signal(&opts->signal->cond));
      |                           ^~~~~~~~~~~~~~~~~~~
./wolfssl/test.h:362:29: note: in definition of macro ‘PTHREAD_CHECK_RET’
  362 |         int _pthread_ret = (__VA_ARGS__);                            \
      |                             ^~~~~~~~~~~
tests/api.c:5863:60: error: ‘tcp_ready’ has no member named ‘cond’
 5863 |         PTHREAD_CHECK_RET(pthread_cond_signal(&opts->signal->cond));
      |                                                            ^~
./wolfssl/test.h:362:29: note: in definition of macro ‘PTHREAD_CHECK_RET’
  362 |         int _pthread_ret = (__VA_ARGS__);                            \
      |                             ^~~~~~~~~~~
tests/api.c:5864:27: error: implicit declaration of function ‘pthread_mutex_unlock’ [-Werror=implicit-function-declaration]
 5864 |         PTHREAD_CHECK_RET(pthread_mutex_unlock(&opts->signal->mutex));
      |                           ^~~~~~~~~~~~~~~~~~~~
./wolfssl/test.h:362:29: note: in definition of macro ‘PTHREAD_CHECK_RET’
  362 |         int _pthread_ret = (__VA_ARGS__);                            \
      |                             ^~~~~~~~~~~
tests/api.c:5864:27: error: nested extern declaration of ‘pthread_mutex_unlock’ [-Werror=nested-externs]
 5864 |         PTHREAD_CHECK_RET(pthread_mutex_unlock(&opts->signal->mutex));
      |                           ^~~~~~~~~~~~~~~~~~~~
./wolfssl/test.h:362:29: note: in definition of macro ‘PTHREAD_CHECK_RET’
  362 |         int _pthread_ret = (__VA_ARGS__);                            \
      |                             ^~~~~~~~~~~
tests/api.c:5864:61: error: ‘tcp_ready’ has no member named ‘mutex’
 5864 |         PTHREAD_CHECK_RET(pthread_mutex_unlock(&opts->signal->mutex));
      |                                                             ^~
./wolfssl/test.h:362:29: note: in definition of macro ‘PTHREAD_CHECK_RET’
  362 |         int _pthread_ret = (__VA_ARGS__);                            \
      |                             ^~~~~~~~~~~
  CC       tests/unit_test-w64wrapper.o
cc1: all warnings being treated as errors
make[2]: *** [Makefile:7079: tests/unit_test-api.o] Error 1
```

### Reproduction steps

1. For the second exception, put this script in the root of the git clone:
```bash
#!/usr/bin/env bash
set -e
cd "$(dirname "$(realpath "$BASH_SOURCE")")"
export CFLAGS="-march=i586 -Wno-error=stringop-truncation"
export CPPFLAGS="$CFLAGS"
export CXXFLAGS="$CFLAGS"
HOST="i686-w64-mingw32"
./autogen.sh
./configure --host="$HOST" --prefix="$PWD/usr" --disable-asm
make clean
make
make install
```
2. Run it. You will get my second exception.
3. To check out the first exception, do the same again but remove `-Wno-error=stringop-truncation` from the file.

embhorn, 2023-10-17:
Hi @Fierelier 

Thanks for contacting wolfSSL Support. Cross compiling can be tricky. Here are some notes:
https://www.wolfssl.com/documentation/manuals/wolfssl/chapter02.html#building-with-configure-with-cross-compile

Please try adding the configure option `--enable-singlethreaded` to resolve the pthread issues.

If you are only using the cryptography (wolfCrypt) not the TLS stack (wolfSSL), then consider using the configure option `--enable-cryptonly`. The string functions are only used by wolfSSL (the TLS stack).

Thanks,
@embhorn - wolfSSL Support

Fierelier, 2023-10-17:
I suspect I have found the source of my Pentium I issue: The mingw compiler tools, and the DLLs it links to are compiled for i686 on Debian. I will use a toolchain compiled for i486 instead and report back.

lealem47, 2023-10-17:
Hi @Fierelier,

Thank you for this report.

Glad to hear you got a successful build. https://github.com/wolfSSL/wolfssl/pull/6881 should hopefully resolve the pthreads issues you were seeing as well as the `stringop-truncation` issue.

Keep us updated about the crashing issue. Also, did you mean to say a toolchain for `i486` or for `i586`?

Thanks,
Lealem @ wolfSSL

Fierelier, 2023-10-18:
Hello @lealem47 

I meant `i486`, I've used Gentoo i486 to compile the MinGW toolchain and I got a seemingly successful i586 build of wolfSSL -- no crash on Pentium I! So no issue on wolfSSL's part, on this front.

The rest of the issues would likely be resolved by https://github.com/wolfSSL/wolfssl/pull/6881. I will close this issue, since it's basically a duplicate of issue https://github.com/wolfSSL/wolfssl/issues/6838 now. Feel free to re-open it if you feel it still holds relevance.

I appreciated the help, thank you.

closed 2023-10-18 by Fierelier