wolfSSL/wolfssl #6876: [Bug]: wolfSSL_dtls_export requires buffer larger than actually used

nunojpg, 2023-10-17:
### Version

master

### Description

My DTLS v1.2 (NOPEER) exports done with wolfSSL_dtls_export are 366 bytes long.
But wolfSSL_dtls_export will report a error if I call it with a buf len < 430.

This comes from the checks in the beggining of wolfSSL_session_export_internal, and I wonder if that calculation is missing the #ifdef for WOLFSSL_SESSION_EXPORT_NOPEER.

embhorn, 2023-10-17:
Hi @nunojpg 

Thanks for this report. I have requested a review by the wolfSSL team.

rizlik, 2023-10-20:
Hi @nunojpg ,
Thanks for the reporting. It may be that the code considers the maximum amount of space needed for the key, so the required space is more than what is required. Can you provide more information about your setup? (options used and ciphers used). Having a minimal reproducer would help a lot in debugging the issue.

Thanks,
Marco

nunojpg, 2023-10-21:
Hi @rizlik, sure,

I build master with "./configure --enable-shared=no --enable-sessionexport=nopeer --enable-dtls".

Let's just use server-dtls.c example with this patch:

```
diff --git a/dtls/server-dtls.c b/dtls/server-dtls.c
index 5e4c13e..96d3d4a 100644
--- a/dtls/server-dtls.c
+++ b/dtls/server-dtls.c
@@ -195,6 +195,18 @@ int main(int argc, char** argv)
             }
             else {
                 printf("Sending reply.\n");
+                printf("%s\n", wolfSSL_get_cipher(ssl));
+
+                char buf[512];
+                int sz = sizeof(buf);
+                int ret;
+                printf("export for buffer with size %d: ", sz);
+                ret = wolfSSL_dtls_export(ssl, buf, &sz);
+                printf("return %4d used size %d\n", ret, sz);
+                printf("export for buffer with size %d: ", sz);
+                ret = wolfSSL_dtls_export(ssl, buf, &sz);
+                printf("return %4d used size %d\n", ret, sz);
+
             }
         }
```
When running it against client-dtls.c I get:

TLS_ECDHE_RSA_WITH_AES_256_GCM_SHA384
export for buffer with size 512: return  382 used size 382
export for buffer with size 382: return -202 used size 446
Sending reply.

So just repeating the command again with the size previously required always fails.

Out of curiosity the size of the exports are:

397 TLS_ECDHE_RSA_WITH_AES_256_GCM_SHA384 (configured with peer)
382 TLS_ECDHE_RSA_WITH_AES_256_GCM_SHA384 (configured with no peer)

My previously reported size is because I use another cipher:

366 TLS_ECDHE_ECDSA_WITH_CHACHA20_POLY1305_SHA256

rizlik, 2023-10-23:
Hi @nunojpg, thanks for the quick reply. 

Yes, indeed the function checks that the buffer is big enough to hold the **maximum** size of the session, which is often more than needed. 
Ideally, the function should return the exact number of bytes needed if invoked with `buf == NULL` and not fail if invoked with a big enough buffer.
We add this to our feature request list. 
Would you mind to share why this is important to your use case?  Feel free to move the discussion to support@wolfssl.com if you want.

Marco

rizlik, 2023-10-23:
Yes, the code (and comments) need to be refreshed for sure. 
Thanks a lot for the report, it is appreciated.