wolfSSL/wolfssl #6913: Unable to create RSA from private key and PEM Phrase

eylul-commits, 2023-10-26:
### Version

v5.5.4-stable

### Description

Hello! I have a simple licensing project which uses private encrypt and public decrypt. I started the project with OpenSSL but now I am trying to migrate from it. Since WolfSSL has an OpenSSL module, I  changed necessary function and variable names (they start with wolfSSL_ now). Also changed  the library I included. 

```
  wolfSSL_PEM_read_bio_RSAPrivateKey(bio, &rsa, NULL, NULL);  //This part returns NULL.
```
However, unlike `PEM_read_bio_RSAPrivateKey` in OpenSSL, `wolfSSL_PEM_read_bio_RSAPrivateKey` does not ask for PEM pass phrase and it directly returns NULL.

I also tried to manually give the pass phrase. This works with OpenSSL functions, but does not work with WolfSSL (no errors but returns NULL instead of RSA).
```
    char* passphrase = "passphrase"
    wolfSSL_PEM_read_bio_RSAPrivateKey(bio, &rsa, NULL, passphrase);
```
I enabled openSSL, enckeys and pwdbased. Nothing changed.
`./configure --enable-opensslall --enable-enckeys --enable-pwdbased`

anhu, 2023-10-26:
Hello @eylul-commits , 

Thank you so much for using wolfSSL!  My name is Anthony Hu and I am a member of the wolfSSL team. 

As a starting point to better understand the problem and where the issue might lie, can you please try using our `examples/pem/pem` app to see if you can use the `--pass` functionality there to successfully open your file? Please report back what you see when you try that. 

Can you also please also let us know about yourself?  For example, Is this project for academic, professional or personal interest? Are you associated with an organization?  what are you trying to achieve?  What are your goals with the project? 

I await your results.  

Warm regards, Anthony

eylul-commits, 2023-10-31:
Hi,

`examples/pem/pem` app does not exist on the version I am using which is v5.5.4. Also my private key isn't in a file, I keep it as a char array. I'm trying to generate encrypted text by using 'wolfSSL_RSA_private_encrypt' function.

I'm trying to make an example licensing project for a company app. I have an RSA public and private key. I'm generating a encrypted text (license) by using private key, then I put the license to my application configuration, then my application will decrypt it by using it's own public key to start working.

Here is my code to generate encrypted text (license): 

```
    WOLFSSL_RSA* create_rsa_with_bio(unsigned char* key) {
        char* passphrase = "password";
        WOLFSSL_RSA* rsa = NULL;
        WOLFSSL_BIO* bio = wolfSSL_BIO_new_mem_buf(key, -1);
        if (bio == NULL) throw_error("Failed to create BIO");
        WOLFSSL_RSA* rsa = wolfSSL_PEM_read_bio_RSAPrivateKey(bio, &rsa, NULL, passphrase);
        if (rsa == NULL) throw_error("Failed to create RSA (check PEM pass phrase)");
        return rsa;
    }

    int private_encrypt(unsigned char* data, int length, unsigned char* key, unsigned char* encrypted) {
        return wolfSSL_RSA_private_encrypt(length, data, encrypted, create_rsa_with_bio(key), RSA_PKCS1_PADDING);
    }

   int main() {
        unsigned char private_key[]="-----BEGIN RSA PRIVATE KEY-----\n"\
            "Proc-Type: 4,ENCRYPTED\n"\
            "DEK-Info: DES-EDE3-CBC,4F2535FAADEBA9EC\n\n";
            // ...
        unsigned char encrypted[4096];
        int encrypted_length = private_encrypt("text to encrypt", 15, private_key, encrypted);
        // encrypted array must be filled
    }
```

This worked when I used OpenSSL. But when I used wolfssl libraries instead (my only change is "wolfssl_" prefixes), my rsa becomes NULL.

anhu, 2023-10-31:
Hi, 

I have consulted with a couple of my colleagues and they pointed out that triple-DES is disabled by default so you might need to explicitly enable it.  Can you please add `--enable-des3` to your `./configure` command? 

If that doesn't help, logs would be great!  In your  example licensing project , Before you call any wolfSSL APIs, please call `wolfSSL_Debugging_ON();`.  That will give logs that might help in determining the problem. 

Warm regards, Anthony

anhu, 2023-10-31:
On another note, may I ask where you geographically located? Country and nearest large city should be sufficient.

Warm regards, Anthony

eylul-commits, 2023-11-02:
Yes, `--enable-des3` worked. Thank you for your help. I am located in Istanbul, Turkey by the way.
Best Wishes!

closed 2023-11-02 by eylul-commits

anhu, 2023-11-02:
Hi @eylul-commits , 

Great to hear you got it working.  If you are interested in commercial licensing, please do not hesitate to reach out to us at support@wolfssl.com. We'll be happy to start a discussion there. 

Warm regards, Anthony