wolfSSL/wolfssl #6914: [Bug]: cannot package on CentOS 8 aarch

sergey-safarov, 2023-10-26:
### Contact Details

s.safarov@gmail.com

### Version

5.6.3

### Description

I want to create RPM package of the `wolftls` and to do this use command
```
rpmbuild -bb --undefine=_disable_source_fetch wolftls.spec
```
The `wolftls.spec` is renamed `rpm/spec`.
During packaging, i got the error listed below.

### Reproduction steps

```
rpmbuild -bb --undefine=_disable_source_fetch wolftls.spec
```
Where `wolftls.spec` is renamed `rpm/spec`.

### Relevant log output

```shell
+ /usr/bin/mkdir -p /root/rpmbuild/BUILDROOT/wolfssl-5.6.3-1.aarch64/usr/share/doc/wolfssl
+ cp -pr AUTHORS /root/rpmbuild/BUILDROOT/wolfssl-5.6.3-1.aarch64/usr/share/doc/wolfssl
+ cp -pr ChangeLog.md /root/rpmbuild/BUILDROOT/wolfssl-5.6.3-1.aarch64/usr/share/doc/wolfssl
+ cp -pr COPYING /root/rpmbuild/BUILDROOT/wolfssl-5.6.3-1.aarch64/usr/share/doc/wolfssl
+ cp -pr README /root/rpmbuild/BUILDROOT/wolfssl-5.6.3-1.aarch64/usr/share/doc/wolfssl
+ cp -pr README.md /root/rpmbuild/BUILDROOT/wolfssl-5.6.3-1.aarch64/usr/share/doc/wolfssl
+ exit 0

RPM build errors:
    Downloading http://wolfssl.com/wolfssl-5.6.3.tar.gz to /root/rpmbuild/SOURCES/wolfssl-5.6.3.tar.gz
    File not found: /root/rpmbuild/BUILDROOT/wolfssl-5.6.3-1.aarch64/usr/share/doc/wolfssl/taoCert.txt
    File not found: /root/rpmbuild/BUILDROOT/wolfssl-5.6.3-1.aarch64/usr/share/doc/wolfssl/example/echoserver.c
    File not found: /root/rpmbuild/BUILDROOT/wolfssl-5.6.3-1.aarch64/usr/share/doc/wolfssl/example/server.c
    File not found: /root/rpmbuild/BUILDROOT/wolfssl-5.6.3-1.aarch64/usr/share/doc/wolfssl/example/echoclient.c
    File not found: /root/rpmbuild/BUILDROOT/wolfssl-5.6.3-1.aarch64/usr/share/doc/wolfssl/example/client.c
    File not found: /root/rpmbuild/BUILDROOT/wolfssl-5.6.3-1.aarch64/usr/share/doc/wolfssl/example/sctp-client.c
    File not found: /root/rpmbuild/BUILDROOT/wolfssl-5.6.3-1.aarch64/usr/share/doc/wolfssl/example/sctp-server.c
    File not found: /root/rpmbuild/BUILDROOT/wolfssl-5.6.3-1.aarch64/usr/share/doc/wolfssl/example/sctp-client-dtls.c
    File not found: /root/rpmbuild/BUILDROOT/wolfssl-5.6.3-1.aarch64/usr/share/doc/wolfssl/example/sctp-server-dtls.c
    File not found: /root/rpmbuild/BUILDROOT/wolfssl-5.6.3-1.aarch64/usr/share/doc/wolfssl/example/tls_bench.c
    File not found: /root/rpmbuild/BUILDROOT/wolfssl-5.6.3-1.aarch64/usr/share/doc/wolfssl/README.txt
    File not found: /root/rpmbuild/BUILDROOT/wolfssl-5.6.3-1.aarch64/usr/share/doc/wolfssl/QUIC.md
    File not found: /root/rpmbuild/BUILDROOT/wolfssl-5.6.3-1.aarch64/usr/lib64/libwolfssl.so
    File not found: /root/rpmbuild/BUILDROOT/wolfssl-5.6.3-1.aarch64/usr/lib64/libwolfssl.so.35
    File not found: /root/rpmbuild/BUILDROOT/wolfssl-5.6.3-1.aarch64/usr/lib64/libwolfssl.so.35.5.1
[root@sbc-stage-a0 SPECS]# 
```
```

JacobBarthelmeh, 2023-10-27:
My CentOS VM is older (version 7) but it did not hit the same issue. The pathing on those files looks suspicious though.

Steps done to try and reproduce it:

```
cd wolfssl
./autogen.sh
./configure
make rpm
```

```
 rpmbuild -bb --undefine=_disable_source_fetch wolfssl/rpm/spec
```

Is there other environment setup needed to reproduce? I see -aarch64 listed, maybe I need a cross compiler for aarch64 installed on the system or to be building natively on aarch64?

sergey-safarov, 2023-10-28:
Thanks for providing instructions.
I was able to compile wolffssl RPM package on CentOS/x86, Centos8/x86 and CentOS/8aarch.

Full command list
```
podman run -it --network host centos:stream8
docker run -it --network host quay.io/centos/centos:stream8
dnf install -y git autoconf automake libtool make rpm-build rpmdevtools
cd /tmp
git clone https://github.com/wolfSSL/wolfssl.git
cd wolfssl
git checkout tags/v5.6.3-stable
git log -n 1
./autogen.sh
./configure
make rpm
rpmbuild -bb --undefine=_disable_source_fetch rpm/spec
```
Closing

closed 2023-10-28 by sergey-safarov