wolfSSL/wolfssl #6928: [Bug]: 5.6.4 introduces a soname change

dvzrv, 2023-10-31:
### Contact Details

dvzrv@archlinux.org

### Version

5.6.4

### Description

Hi! I package this project for Arch Linux.

When building 5.6.4 I noticed, that it introduces a [soname](https://en.wikipedia.org/wiki/Soname) change (coming from 5.6.3). As these are very atypical for patch-level releases, I wanted to raise this here to make sure this was not done by mistake.

```
libwolfssl.so=35-64 -> libwolfssl.so=41-64
```

[wolfssl-5.6.4-1-x86_64.pkg.tar.zst-namcap.log](https://github.com/wolfSSL/wolfssl/files/13214602/wolfssl-5.6.4-1-x86_64.pkg.tar.zst-namcap.log)
[wolfssl-5.6.4-1-x86_64-package.log](https://github.com/wolfSSL/wolfssl/files/13214603/wolfssl-5.6.4-1-x86_64-package.log)
[wolfssl-5.6.4-1-x86_64-check.log](https://github.com/wolfSSL/wolfssl/files/13214604/wolfssl-5.6.4-1-x86_64-check.log)
[wolfssl-5.6.4-1-x86_64-build.log](https://github.com/wolfSSL/wolfssl/files/13214605/wolfssl-5.6.4-1-x86_64-build.log)

### Reproduction steps

https://gitlab.archlinux.org/archlinux/packaging/packages/wolfssl/-/blob/main/PKGBUILD?ref_type=heads

### Relevant log output

```shell
See attached log files above.
```

dvzrv, 2023-10-31:
Addendum:

Looking at the generated library and symlinks for 5.6.3 and 5.6.4, we can observe, that the soname will change every time [LIBTOOL_CURRENT](https://github.com/wolfSSL/wolfssl/blob/37884f864d6fd9b04f44677cb04da15d0c9d6526/CMakeLists.txt#L44-L51) is incremented.

What is puzzling for me is the jump from 35 -> 41 within the range of a patch level release.

5.6.3:
```
lrwxrwxrwx root/root         0 2023-07-14 12:17 usr/lib/libwolfssl.so -> libwolfssl.so.35
lrwxrwxrwx root/root         0 2023-07-14 12:17 usr/lib/libwolfssl.so.35 -> libwolfssl.so.35.5.1
-rwxr-xr-x root/root    800728 2023-07-14 12:17 usr/lib/libwolfssl.so.35.5.1
```

5.6.4:
```
lrwxrwxrwx root/root         0 2023-10-31 10:23 usr/lib/libwolfssl.so -> libwolfssl.so.41
lrwxrwxrwx root/root         0 2023-10-31 10:23 usr/lib/libwolfssl.so.41 -> libwolfssl.so.41.0.0
-rwxr-xr-x root/root    817112 2023-10-31 10:23 usr/lib/libwolfssl.so.41.0.0
```

Where the intermediate sonames just not released or is this a bug?

dvzrv, 2023-10-31:
Upon further investigation, I noticed that the sources for 5.6.3 specify `LIBTOOL_CURRENT` as `40` and not as `35`.
However, when building from those sources the soname is reproducibly `35` as outlined above.

Reproduce with:

```
git clone https://github.com/wolfSSL/wolfssl/ && cd wolfssl
git checkout v5.6.3-stable
grep -R "set(LIBTOOL_CURRENT" CMakeLists.txt
cmake -B build -D CMAKE_BUILD_TYPE=None -D CMAKE_INSTALL_PREFIX=/usr -D WOLFSSL_CURVE25519=ON -D WOLFSSL_CURVE448=ON -D WOLFSSL_ED25519=ON -D WOLFSSL_ED448=ON -D WOLFSSL_REPRODUCIBLE_BUILD=ON -D WOLFSSL_SNI=ON -D WARNING_C_FLAGS="-march=x86-64 -mtune=generic -O2 -pipe -fno-plt -fexceptions -Wp,-D_FORTIFY_SOURCE=2 -W
format -Werror=format-security -fstack-clash-protection -fcf-protection" -W no-dev -S wolfssl && cmake --build build --verbose && DESTDIR=foo cmake --install build
ls -lah foo/usr/lib/libwolfssl.so*
```

```
git clone https://github.com/wolfSSL/wolfssl/ && cd wolfssl
git checkout v5.6.4-stable
grep -R "set(LIBTOOL_CURRENT" CMakeLists.txt
cmake -B build -D CMAKE_BUILD_TYPE=None -D CMAKE_INSTALL_PREFIX=/usr -D WOLFSSL_CURVE25519=ON -D WOLFSSL_CURVE448=ON -D WOLFSSL_ED25519=ON -D WOLFSSL_ED448=ON -D WOLFSSL_REPRODUCIBLE_BUILD=ON -D WOLFSSL_SNI=ON -D WARNING_C_FLAGS="-march=x86-64 -mtune=generic -O2 -pipe -fno-plt -fexceptions -Wp,-D_FORTIFY_SOURCE=2 -W
format -Werror=format-security -fstack-clash-protection -fcf-protection" -W no-dev -S wolfssl && cmake --build build --verbose && DESTDIR=foo cmake --install build
ls -lah foo/usr/lib/libwolfssl.so*
```

JacobBarthelmeh, 2023-10-31:
Hi @dvzrv,

Thanks for bringing this up and handling the Arch Linux package! Had used abi-tracker to check the difference from 5.6.4 to 5.6.3 using the build (--enable-distro --disable-examples) and it listed 3 API as removed. One was converted to a macro and the other 2 looked like they had macro guards around them being altered for when they get compiled in. Since it reported some removed API though I incremented the current and reset the age and revision.

As you noted, the previous release 5.6.3 had a current of 40 but since age and revision had not been reset the soname once libtool gets done with it was still 35.

Regards,
Jacob

dvzrv, 2023-11-07:
> As you noted, the previous release 5.6.3 had a current of 40 but since age and revision had not been reset the soname once libtool gets done with it was still 35.

Just to clarify this for me as a packager of this project: Going forward is it expected, that further breaking changes in the library will be released in patch-level releases?

I'm asking, because doing so is highly unusual and I am required to rebuild all dependents of `libwolfssl.so`.

lukastribus, 2023-11-08:
Related: OpenWRT moved away from wolfssl, one of the reasons being lack of LTS and ABI stability:

http://lists.openwrt.org/pipermail/openwrt-announce/2023-October/000047.html

JacobBarthelmeh, 2023-11-08:
@dvzrv there is ABI support on specific functions that have WOLFSSL_ABI in front of their declarations (includes the TLS 1.3 method functions). That is correct though, that currently there is no guarantee that the SONAME of the library will not get updated with future releases. LTS and ABI support is something that's actively being discussed internally.

dvzrv, 2023-11-09:
> @dvzrv there is ABI support on specific functions that have WOLFSSL_ABI in front of their declarations (includes the TLS 1.3 method functions). 

Thanks for the info. As a packager for a downstream, doing that kind of work in my free time, I do not think that comparing function declarations, with what is used in dependents of wolfssl is at all feasible though.
I'll add a big **WARNING** to the build script, so that hopefully others will know this too, going forward.

> That is correct though, that currently there is no guarantee that the SONAME of the library will not get updated with future releases.

It changing is not the problem I guess, but it changing outside the confines of what is usually accepted is. :)

> LTS and ABI support is something that's actively being discussed internally.

That is good to hear and would be much appreciated (including by me). Please announce any such change! :)
It would be great if rebuilds of dependents would not have to happen on arbitrary releases (e.g. patch-level release).

Thanks for clarifying the open questions and thanks @lukastribus for additional info. I'll close this now, as currently there is no further actionable on this item.

closed 2023-11-09 by dvzrv