wolfSSL/wolfssl #6960: AddressSanitizer reports stack-buffer-overflow in wc_InitRsaKey_ex()

Kartatz, 2023-11-10:
### Version

5.6.4

### Description

AddressSanitizer is reporting a stack-buffer-overflow on a simple test program.

I compiled wolfSSL on Android with CMake:

```bash
cmake -DWOLFSSL_KEYGEN=ON \
	-DWOLFSSL_CERTGEN=ON \
	-DCMAKE_BUILD_TYPE=Debug \
	-DWOLFSSL_CERTEXT=ON \
	-DWOLFSSL_CERTREQ=ON \
	-DWOLFSSL_ASN=ON \
	-DCMAKE_C_FLAGS='-DWOLFSSL_ASN_TEMPLATE -DWOLFSSL_CUSTOM_OID -DHAVE_OID_ENCODING'
```

### Reproduction steps

Write a simple test program:

```c
#include <wolfssl/options.h>
#include <wolfssl/wolfcrypt/settings.h>
#include <wolfssl/wolfcrypt/rsa.h>

int main() {
	
	RsaKey private_key = {0};
	wc_InitRsaKey(&private_key, 0);
	
}
```

Then compile with `cc ... -lwolfssl -g -O0 -fsanitize=address`.

### Relevant log output

```shell
=================================================================
==16229==ERROR: AddressSanitizer: stack-buffer-overflow on address 0x007fe65818d0 at pc 0x0075b78df378 bp 0x007fe657f790 sp 0x007fe657ef80
WRITE of size 8368 at 0x007fe65818d0 thread T0
    #0 0x75b78df374 in memset out/lib/compiler-rt-aarch64/out/llvm-project/compiler-rt/lib/asan/../sanitizer_common/sanitizer_common_interceptors.inc:869:3
    #1 0x75bd2bc294 in wc_InitRsaKey_ex /data/data/com.termux/files/home/wolfssl-5.6.4-stable/wolfcrypt/src/rsa.c:157:5
    #2 0x75bd2bc3dc in wc_InitRsaKey /data/data/com.termux/files/home/wolfssl-5.6.4-stable/wolfcrypt/src/rsa.c:251:12
    #3 0x614592c4a0 in main /sdcard/test.c:8:2
    #4 0x75b9fd9a30 in __libc_init (/apex/com.android.runtime/lib64/bionic/libc.so+0x4ca30) (BuildId: 64ab50d2969db4bf7fbb3794d3d06201)

Address 0x007fe65818d0 is located in stack of thread T0 at offset 8368 in frame
    #0 0x614592c354 in main /sdcard/test.c:5

  This frame has 1 object(s):
    [32, 8368) 'private_key' (line 7) <== Memory access at offset 8368 overflows this variable
HINT: this may be a false positive if your program uses some custom stack unwind mechanism, swapcontext or vfork
      (longjmp and C++ exceptions *are* supported)
SUMMARY: AddressSanitizer: stack-buffer-overflow out/lib/compiler-rt-aarch64/out/llvm-project/compiler-rt/lib/asan/../sanitizer_common/sanitizer_common_interceptors.inc:869:3 in memset
Shadow bytes around the buggy address:
  0x001ffccb02c0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x001ffccb02d0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x001ffccb02e0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x001ffccb02f0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x001ffccb0300: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
=>0x001ffccb0310: 00 00 00 00 00 00 00 00 00 00[f3]f3 f3 f3 f3 f3
  0x001ffccb0320: f3 f3 f3 f3 f3 f3 f3 f3 f3 f3 f3 f3 f3 f3 f3 f3
  0x001ffccb0330: f3 f3 f3 f3 f3 f3 f3 f3 f3 f3 f3 f3 00 00 00 00
  0x001ffccb0340: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x001ffccb0350: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x001ffccb0360: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Shadow byte legend (one shadow byte represents 8 application bytes):
  Addressable:           00
  Partially addressable: 01 02 03 04 05 06 07 
  Heap left redzone:       fa
  Freed heap region:       fd
  Stack left redzone:      f1
  Stack mid redzone:       f2
  Stack right redzone:     f3
  Stack after return:      f5
  Stack use after scope:   f8
  Global redzone:          f9
  Global init order:       f6
  Poisoned by user:        f7
  Container overflow:      fc
  Array cookie:            ac
  Intra object redzone:    bb
  ASan internal:           fe
  Left alloca redzone:     ca
  Right alloca redzone:    cb
==16229==ABORTING
```

dgarske, 2023-11-10:
Hi @Kartatz ,

This is likely due to your the struct RsaKey not matching between the wolfSSL build and your application. I believe it is because `HAVE___UINT128_T` is not getting set in the options.h coming from CMake.

@ejohnstown this sounds a lot like the other CMake issue where `HAVE___UINT128_T` isn't defined in the CMake generated options.h. Can you please look at fixing this properly in wolfSSL? See https://github.com/wolfSSL/wolfssh/issues/594

Thanks,
David Garske, wolfSSL

anhu, 2023-11-10:
Doesn't a call to `wc_InitRsaKey()`  require a corresponding call to `wc_FreeRsaKey()` ?

ejohnstown, 2023-11-11:
@anhu, it should but for this simple of a test it is a don't care. The issue is seen in the wc_InitRsaKey() call.

dgarske, 2023-11-11:
Hi @Kartatz ,

Can you please test with this cmake patch? https://github.com/wolfSSL/wolfssl/pull/6965#event-10930715685

Thanks,
David Garske, wolfSSL

Kartatz, 2023-11-11:
> Can you please test with this cmake patch?  

I did some tests and this indeed fixes the issue. Thanks @ejohnstown and @dgarske!

PR #6965 (2023-11-11): uint128 and CMake
# Description

Add to the check for HAVE___UINT128_T adding it to the list of items that get dumped into the options.h file.

Fixes #6960.

# Testing

I built the library following the process in #6960 and compiled the sample code. Also built wolfSSH against this wolfSSL change and it didn't have issues.

# Checklist

 - [ ] added tests
 - [ ] updated/added doxygen
 - [ ] updated appropriate READMEs
 - [ ] Updated manual and documentation
closed 2023-11-11 by dgarske