wolfSSL/wolfssl #6969: [bug]: Missing support for STM32H725 in wolfSSL.I-CUBE-wolfSSL_conf.h

Makin-Things, 2023-11-14:
### Contact Details

simon@makin-things.com

### Version

5.6.4

### Description

The is no 

```
#elif defined(STM32H725xx)
    #define WOLFSSL_STM32H7
    #define HAL_CONSOLE_UART huart3
```
in the generated wolfSSL.I-CUBE-wolfSSL_conf.h which causes it to not build.

dgarske, 2023-11-14:
@lealem47 can you please add this to the default.ftl? Thanks

closed 2023-11-16 by dgarske

Makin-Things, 2024-01-18:
There doesn't appear to be any update available yet within the STM32CubeIDE. When I manually add this it works until the IOC regenerates the project. Is there a timeframe for when this will be made available? In the interim, how can I add it in such a way that it doesn't constantly get removed?