wolfSSL/wolfssl #6983: Enable wolfSSL DTLS-SRTP support using cmake

farazrbx, 2023-11-18:
### Version

current master branch

### Description

How do I enable wolfSSL DTLS-SRTP support using cmake? Looks like the option is missing.

lealem47, 2023-11-20:
Hi @farazrbx,

Please try with https://github.com/wolfSSL/wolfssl/pull/6991 using the command `cmake -DWOLFSSL_SRTP=yes`.

Thanks,
Lealem @ wolfSSL

farazrbx, 2023-11-22:
Thanks!
`--enable-quic` and `--enable-curl` options also seem to be missing in CMake.

lealem47, 2023-11-22:
@farazrbx Autotools is our primary build system so unfortunately, our CMake is missing quite a few options. If you give me a full list of the features you need, I can make sure that they are added.

farazrbx, 2023-11-27:
Thank you! I think that's it for me. I'm trying to replace openssl with wolfssl in my project that depends on curl, webrtc, and libsrtp.

lealem47, 2023-11-27:
@farazrbx You're welcome! I'll keep you updated on the CMake additions. Can I ask what type of project you're working on?

farazrbx, 2023-11-30:
Thank you for your help. My project is a game engine. I think we have a commercial license for WolfSSL too. 
I'm also working on PlayStation compatibility here: https://github.com/wolfSSL/wolfssl/compare/master...farazrbx:wolfssl:master
It would be nice if we could merge some of the changes to the upstream.

lealem47, 2023-12-01:
@farazrbx Thank you for the context. 

PlayStation compatibility sounds great! I encourage you to put up a pull request once you've finalized the changes.

lealem47, 2023-12-08:
Hi @farazrbx ,

The pull request is up at https://github.com/wolfSSL/wolfssl/pull/7049. Thanks for the patience

lealem47, 2023-12-11:
Closing this now that the PR has merged.

closed 2023-12-11 by lealem47