wolfSSL/wolfssl #6999: Is there a function for get_dest_addr ?

ManonBillet, 2023-11-23:
### Version

5.6.3

### Description

I'm making a custom dtls server with multithread and I need to find the ip address for the destination. 
Have you a function who allows this feature ?

anhu, 2023-11-23:
Hi @ManonBillet 

the way you asked your question seems a bit vague.  You are making a DTLS server and would like to know ip address of the destination.  Do you mean when you get a connection request from a client, you would like to know the ip address for the client? 

If this is the question you are asking, then perhaps this related post is useful: 

https://www.linuxquestions.org/questions/programming-9/how-to-get-client-ip-address-at-udp-server-304252/

Note that in DTLS the address is a UDP level concept. 

Warm regards, Anthony

ManonBillet, 2023-11-24:
Hello, 

In my server I have the ip of my routers but I can't find my own ip address. 
So, when I receive a client request, I would like to know the ip of my interface which is connected to the client. 

King regards, 

Manon Billet 
5ème année - Département TC 
INSA Lyon 

[ https://www.linuxquestions.org/questions/programming-9/how-to-get-client-ip-address-at-udp-server-304252/ | https://www.linuxquestions.org/questions/programming-9/how-to-get-client-ip-address-at-udp-server-304252/ ]

anhu, 2025-11-17:
Hi @ManonBillet , 

I'm going to guess that similar to me, you have lost track of this issue. As such, as a matter of housekeeping, I'm going to close this ticket. Please send re-open it if you are interested in pursuing this again.

Warm regards, Anthony