wolfSSL/wolfssl #7000: [Bug]: DTLS 1.2 export/import broken since aed715cb2c8

nunojpg, 2023-11-24:
### Version

master

### Description

I just have a test failing in the session export/import feature and I did bisect it down to 
Transmission appears to work ok, but on reception wolfSSL_read yields WOLFSSL_ERROR_WANT_READ.
I am only using DTLS 1.2. Configure with --enable-shared=no --enable-sessionexport=nopeer --enable-dtls.

`aed715cb2c80842c5717d94b987d20ccc1b53f97 is the first bad commit
commit aed715cb2c80842c5717d94b987d20ccc1b53f97
Author: Juliusz Sosinowicz <juliusz@wolfssl.com>
Date:   Thu Oct 26 17:11:40 2023 +0200

    dtls 1.3: allow to skip cookie exchange on resumption
    
    tls 1.3: do cookie exchange when asked too even when found a matching cipher`

anhu, 2023-11-24:
Hello @nunojpg , 

Thanks for letting us know we have a problem here. I see you have provided configuration flags, but can you please let us know how to reproduce what you are seeing?  For example, you say you have a test failing, but I tried `make check` after checkout our master branch and configuring with your flags and all our tests passed.  

Can you provide us the test that failed? 

Warm regards, Anthony

nunojpg, 2023-11-25:
Hi, this happens when using Botan as a DTLS client and I can't at this time provide a standalone test case.
I have tracked the bug further to a change in this commit on src/internal.c:

```
if (!ssl->options.dtlsStateful) {
            WOLFSSL_MSG("Drop non-handshake record when not stateful");
             return 1;
}
```

So of course the source of the bug can be anywhere else, but I'm ending up at this stateful check which is not relevant for a build without DTLS 1.3 or any use of DTLS 1.3.

Let me know if this helps.

nunojpg, 2023-11-25:
Trying this patch on the examples repo:

```
--- a/dtls/server-dtls.c
+++ b/dtls/server-dtls.c
@@ -176,6 +176,14 @@ int main(int argc, char** argv)
             break;
         }
         while (1) {
+            unsigned int ss = 1024;
+            unsigned char buf[1024];
+            wolfSSL_dtls_export(ssl, buf, &ss);
+            wolfSSL_free(ssl);
+            ssl = wolfSSL_new(ctx);
+            wolfSSL_dtls_import(ssl, buf, ss);
+            wolfSSL_set_fd(ssl, listenfd);
```

The server works before aed715cb2c8 but not after.

After some investigation I note ssl->options.dtlsStateful is set in the original session but not after the import and this commit requires it to be set due to the change in src/internal.c:

```
if (!ssl->options.dtlsStateful) {
             WOLFSSL_MSG("Drop non-handshake record when not stateful");
             return 1;
} 
```

So either this variable is not being exported/imported correctly, or this check is not intended for DTLS 1.2.

Hope this helps out.

julek-wolfssl, 2023-12-05:
Hi @nunojpg,

thank you for your report. I have opened pull request that fixes this issue here https://github.com/wolfSSL/wolfssl/pull/7030. Please try this branch to see if it fixes session exporting for you.

Juliusz

closed 2023-12-11 by dgarske