wolfSSL/wolfssl #7008: Missing SSL_set_mtu() and DTLS_set_link_mtu()

farazrbx, 2023-11-28:
### Version

latest

### Description

`SSL_set_mtu()` and `DTLS_set_link_mtu()` functions are missing in WolfSSL.
Do you have any plans to implement them in the future?

anhu, 2023-11-29:
Hi @farazrbx , 

I will need to consult with my team on that.   Please stay tuned. 

Warm regards, Anthony

anhu, 2023-11-29:
Note to self: APIs for OpenSSL Compat Layer.

anhu, 2023-11-30:
Note that we do have `wolfSSL_dtls_set_mtu()` .  How strong is your interest in DTLS_set_link_mtu()? Would you like to register a feature request for it?

douzzer, 2024-11-17:
@farazrbx Note that since merge of #7648 we have `SSL_set_mtu()`, which is a compat layer alias for `wolfSSL_set_mtu_compat()`.

These interfaces are available in release 5.7.4 or later.

anhu, 2025-11-11:
Hi, Looks like this is resolved!  I will now close this issue. 
Warm regards, Anthony