wolfSSL/wolfssl #7024: Warning: inline asm clobber list contains reserved registers R7 [-Winline-asm]

waqasakrambueid, 2023-12-01:
### Version

v5.6.4-stable

### Description

I am compiling for Android [armeabi-v7a] and having different warnings like shown below.

```
sp_armthumb.c:99153:9: warning: inline asm clobber list contains reserved registers: R7 [-Winline-asm]
        "movs   r2, #0\n\t"
 note: Reserved registers on the clobber list may not be preserved across the asm statement, and clobbering them may lead to undefined behaviour.
 
 sp_armthumb.c:97988:9: warning: inline asm clobber list contains reserved registers: R7 [-Winline-asm]
        "movs   r4, #0\n\t"
 note: Reserved registers on the clobber list may not be preserved across the asm statement, and clobbering them may lead to undefined behaviour.
```
 
 
My user_settings.h:

```
#ifndef USER_SETTINGS_BASE_H
#define USER_SETTINGS_BASE_H

//
// -- WolfSSL / WolfCrypt setup
//

//
// General stuff
//
#define WOLFCRYPT_ONLY
#define SINGLE_THREADED
#define WOLFSSL_STATIC_MEMORY

// Encryption stuff
#define HAVE_ENCRYPT_THEN_MAC
#define HAVE_ECC
#define HAVE_ECC_SIGN
#define HAVE_ECC_VERIFY
#define HAVE_ECC_ENCRYPT
#define HAVE_ECC_CHECK_KEY
#define HAVE_ECC_DHE
#define HAVE_ECC_KEY_IMPORT
#define HAVE_ECC_KEY_EXPORT
#define ECC_SHAMIR
#define ECC_TIMING_RESISTANT
#define TFM_TIMING_RESISTANT
#define HAVE_HKDF
#define ECC_USER_CURVES
#define WOLFSSL_ECIES_ISO18033
#define WOLFSSL_DES_ECB
// #define HAVE_AES_ECB
// #define WOLFSSL_AES_DIRECT

#define DEFAULT_ECC_KEY_CURVE ECC_SECP256R1
#define DEFAULT_ECC_KEY_CURVE_SIZE 32

// Setup math stuff
#ifdef WOLFSSL_SP_MATH
#define WOLFSSL_SP_MATH_ALL
#define WOLFSSL_HAVE_SP_DH
#define WOLFSSL_HAVE_SP_ECC
#define WOLFSSL_SP_SMALL
#define WOLFSSL_SP_NO_MALLOC
#define WOLFSSL_SP_NO_2048
#define WOLFSSL_SP_NO_3072
#define SP_INT_BITS 256
#else
#define TFM_ECC256
#endif

// Remove unnecessary or not available features
#define NO_WOLFSSL_CLIENT
#define NO_WOLFSSL_SERVER
#define NO_FILESYSTEM
#define NO_ERROR_STRINGS
#define NO_DSA
#define NO_MD4
#define NO_MD5
#define NO_SHA
#define NO_PSK
#define NO_PWDBASED
#define NO_RC4
#define NO_SESSION_CACHE
#define NO_TLS
#define NO_RSA
#define NO_DH
#define NO_ASN_TIME
#define NO_AES_192
#define NO_AES_256
#define WC_NO_RSA_OAEP
#define NO_DEV_URANDOM
#define NO_WOLFSSL_DIR
#define NO_SESSION_CACHE
#define WOLFSSL_NO_SIGALG
#define NO_RESUME_SUITE_CHECK
#define NO_OLD_TLS
#define WOLFSSL_AEAD_ONLY
#define WOLFSSL_NO_TLS12
#define WOLFSSL_SP_NO_2048
#define WOLFSSL_SP_NO_3072

#define XTIME(t) NULL
#endif

```

waqasakrambueid, 2023-12-01:
Is there a way to fix this warning?

dgarske, 2023-12-01:
Hi @waqasakrambueid ,

Does this recent PR fix your issue?
https://github.com/wolfSSL/wolfssl/pull/7013

Thanks,
David Garske, wolfSSL

SparkiDev, 2023-12-03:
Hi @waqasakrambueid

Please try the compile option: -Wno-inline-asm
This should stop the compiler complaining about use of R7

Thanks,
Sean Parkinson, wolfSSL

waqasakrambueid, 2023-12-05:
@dgarske I tested in master, It still existed. I have attached complete logs below.

`wolfssl/wolfcrypt/src/sp_int.c:8841:17: warning: inline asm clobber list contains reserved registers: R7 [-Winline-asm]
                SP_ASM_MUL_ADD(l, h, o, dp[0], b->dp[j]);
                ^
/Users/waqasakram/blueidng/blueid-access/src/ext/wolfssl/wolfcrypt/src/sp_int.c:1520:9: note: expanded from macro 'SP_ASM_MUL_ADD'
        "uxth   r6, %[a]                \n\t"            \
        ^
/Users/waqasakram/blueidng/blueid-access/src/ext/wolfssl/wolfcrypt/src/sp_int.c:8841:17: note: Reserved registers on the clobber list may not be preserved across the asm statement, and clobbering them may lead to undefined behaviour.
/Users/waqasakram/blueidng/blueid-access/src/ext/wolfssl/wolfcrypt/src/sp_int.c:1520:9: note: expanded from macro 'SP_ASM_MUL_ADD'
        "uxth   r6, %[a]                \n\t"            \
        ^
/Users/waqasakram/blueidng/blueid-access/src/ext/wolfssl/wolfcrypt/src/sp_int.c:8852:17: warning: inline asm clobber list contains reserved registers: R7 [-Winline-asm]
                SP_ASM_MUL_ADD(l, h, o, a->dp[i], dp[0]);
                ^
/Users/waqasakram/blueidng/blueid-access/src/ext/wolfssl/wolfcrypt/src/sp_int.c:1520:9: note: expanded from macro 'SP_ASM_MUL_ADD'
        "uxth   r6, %[a]                \n\t"            \
        ^
/Users/waqasakram/blueidng/blueid-access/src/ext/wolfssl/wolfcrypt/src/sp_int.c:8852:17: note: Reserved registers on the clobber list may not be preserved across the asm statement, and clobbering them may lead to undefined behaviour.
/Users/waqasakram/blueidng/blueid-access/src/ext/wolfssl/wolfcrypt/src/sp_int.c:1520:9: note: expanded from macro 'SP_ASM_MUL_ADD'
        "uxth   r6, %[a]                \n\t"            \
        ^
/Users/waqasakram/blueidng/blueid-access/src/ext/wolfssl/wolfcrypt/src/sp_int.c:8919:17: warning: inline asm clobber list contains reserved registers: R7 [-Winline-asm]
                SP_ASM_MUL_ADD(l, h, o, a->dp[i], b->dp[j]);
                ^
/Users/waqasakram/blueidng/blueid-access/src/ext/wolfssl/wolfcrypt/src/sp_int.c:1520:9: note: expanded from macro 'SP_ASM_MUL_ADD'
        "uxth   r6, %[a]                \n\t"            \
        ^
/Users/waqasakram/blueidng/blueid-access/src/ext/wolfssl/wolfcrypt/src/sp_int.c:8919:17: note: Reserved registers on the clobber list may not be preserved across the asm statement, and clobbering them may lead to undefined behaviour.
/Users/waqasakram/blueidng/blueid-access/src/ext/wolfssl/wolfcrypt/src/sp_int.c:1520:9: note: expanded from macro 'SP_ASM_MUL_ADD'
        "uxth   r6, %[a]                \n\t"            \
        ^
/Users/waqasakram/blueidng/blueid-access/src/ext/wolfssl/wolfcrypt/src/sp_int.c:8930:17: warning: inline asm clobber list contains reserved registers: R7 [-Winline-asm]
                SP_ASM_MUL_ADD(l, h, o, a->dp[i], b->dp[j]);
                ^
/Users/waqasakram/blueidng/blueid-access/src/ext/wolfssl/wolfcrypt/src/sp_int.c:1520:9: note: expanded from macro 'SP_ASM_MUL_ADD'
        "uxth   r6, %[a]                \n\t"            \
        ^
/Users/waqasakram/blueidng/blueid-access/src/ext/wolfssl/wolfcrypt/src/sp_int.c:8930:17: note: Reserved registers on the clobber list may not be preserved across the asm statement, and clobbering them may lead to undefined behaviour.
/Users/waqasakram/blueidng/blueid-access/src/ext/wolfssl/wolfcrypt/src/sp_int.c:1520:9: note: expanded from macro 'SP_ASM_MUL_ADD'
        "uxth   r6, %[a]                \n\t"            \
        ^
/Users/waqasakram/blueidng/blueid-access/src/ext/wolfssl/wolfcrypt/src/sp_int.c:14556:17: warning: inline asm clobber list contains reserved registers: R7 [-Winline-asm]
                SP_ASM_MUL_ADD2(l, h, o, a->dp[i], a->dp[j]);
                ^
/Users/waqasakram/blueidng/blueid-access/src/ext/wolfssl/wolfcrypt/src/sp_int.c:1633:9: note: expanded from macro 'SP_ASM_MUL_ADD2'
        "uxth   r6, %[a]                \n\t"            \
        ^
/Users/waqasakram/blueidng/blueid-access/src/ext/wolfssl/wolfcrypt/src/sp_int.c:14556:17: note: Reserved registers on the clobber list may not be preserved across the asm statement, and clobbering them may lead to undefined behaviour.
/Users/waqasakram/blueidng/blueid-access/src/ext/wolfssl/wolfcrypt/src/sp_int.c:1633:9: note: expanded from macro 'SP_ASM_MUL_ADD2'
        "uxth   r6, %[a]                \n\t"            \
        ^
/Users/waqasakram/blueidng/blueid-access/src/ext/wolfssl/wolfcrypt/src/sp_int.c:14567:17: warning: inline asm clobber list contains reserved registers: R7 [-Winline-asm]
                SP_ASM_MUL_ADD2(l, h, o, a->dp[i], a->dp[j]);
                ^
/Users/waqasakram/blueidng/blueid-access/src/ext/wolfssl/wolfcrypt/src/sp_int.c:1633:9: note: expanded from macro 'SP_ASM_MUL_ADD2'
        "uxth   r6, %[a]                \n\t"            \
        ^
/Users/waqasakram/blueidng/blueid-access/src/ext/wolfssl/wolfcrypt/src/sp_int.c:14567:17: note: Reserved registers on the clobber list may not be preserved across the asm statement, and clobbering them may lead to undefined behaviour.
/Users/waqasakram/blueidng/blueid-access/src/ext/wolfssl/wolfcrypt/src/sp_int.c:1633:9: note: expanded from macro 'SP_ASM_MUL_ADD2'
        "uxth   r6, %[a]                \n\t"            \
        ^
[... cut ...]  to undefined behaviour.
/Users/waqasakram/blueidng/blueid-access/src/ext/wolfssl/wolfcrypt/src/sp_armthumb.c:98178:9: warning: inline asm clobber list contains reserved registers: R7 [-Winline-asm]
        "movs   r7, #0\n\t"
        ^
/Users/waqasakram/blueidng/blueid-access/src/ext/wolfssl/wolfcrypt/src/sp_armthumb.c:98178:9: note: Reserved registers on the clobber list may not be preserved across the asm statement, and clobbering them may lead to undefined behaviour.
14 warnings generated.
`

SparkiDev, 2023-12-05:
Hi @waqasakrambueid

Note that in the error log:
   ... R7 [-Winline-asm] ...

Please add -Wno-inline-asm to the compile flags to stop the compiler warning about this.

Sean

SparkiDev, 2023-12-06:
Hi @waqasakrambueid

We have had other customers use the inline assembly code on Android using this compile flag.
But do raise a new issue if the code isn't working.

Sean

closed 2023-12-06 by SparkiDev