wolfSSL/wolfssl #7033: Multiple Compilation Errors in CCS Project with wolfSSL and FreeRTOS Dependencies

Can-Yildirim, 2023-12-06:
### Version

FreeRTOS V8.2.3  lwip-1.4.1 wolfssh-1.4.14 wolfssl-5.6.4

### Description

We are encountering several compilation issues in our project, these problems primarily involve the integration of third-party libraries, specifically wolfSSL and FreeRTOS. Below is a detailed list of the errors we have identified:

    Error regarding the need for at least one key agreement algorithm in wolfssh (internal.h, line 176).
    Compilation error due to the order of including FreeRTOS.h before task.h (FreeRTOS, task.h, line 75).
    Multiple instances of the 'expected a type specifier' error in wolfssl's ssl.h (lines 1867, 2180, 2182, 2184, 2185, 2187, 2230) and internal.h (lines 6121, 6380, 6381, 6391, 6392, 6393, 6610).
    Errors expecting either a definition or a tag name in wolfssl's ssl.h (lines 168, 215).
We also encountered a directive error in wolfssl's ssl.c (line 63) stating no cipher suites are defined due to disabled DH and ECC, and no static suites being defined.
   
    #35 #error directive: "You need at least one key agreement algorithm." .ccsproject /PR-SSFT-2272 line 176, external location: C:\ti5\third_party\wolfssh\internal.h C/C++ Problem
    #35 #error directive: "include FreeRTOS.h must appear in source files before include task.h" .ccsproject /PR-SSFT-2272 line 75, external location: C:\ti5\third_party\FreeRTOS\Source\include\task.h C/C++ Problem
    #80 expected a type specifier .ccsproject /PR-SSFT-2272 line 1867, external location: C:\ti5\third_party\wolfssl\ssl.h C/C++ Problem
    #111 expected either a definition or a tag name .ccsproject /PR-SSFT-2272 line 215, external location: C:\ti5\third_party\wolfssl\ssl.h C/C++ Problem
    #111 expected either a definition or a tag name .ccsproject /PR-SSFT-2272 line 168, external location: C:\ti5\third_party\wolfssl\ssl.h C/C++ Problem
    #80 expected a type specifier .ccsproject /PR-SSFT-2272 line 2182, external location: C:\ti5\third_party\wolfssl\ssl.h C/C++ Problem
    #80 expected a type specifier .ccsproject /PR-SSFT-2272 line 2180, external location: C:\ti5\third_party\wolfssl\ssl.h C/C++ Problem
    #80 expected a type specifier .ccsproject /PR-SSFT-2272 line 2184, external location: C:\ti5\third_party\wolfssl\ssl.h C/C++ Problem
    #80 expected a type specifier .ccsproject /PR-SSFT-2272 line 2185, external location: C:\ti5\third_party\wolfssl\ssl.h C/C++ Problem
    #80 expected a type specifier .ccsproject /PR-SSFT-2272 line 2187, external location: C:\ti5\third_party\wolfssl\ssl.h C/C++ Problem
    #35 #error directive: "No cipher suites defined because DH disabled, ECC disabled, and no static suites defined. Please see top of README" ssl.c /PR-SSFT-2272/third_party/wolfssl/src line 63 C/C++ Problem
    #80 expected a type specifier .ccsproject /PR-SSFT-2272 line 2230, external location: C:\ti5\third_party\wolfssl\ssl.h C/C++ Problem
    #80 expected a type specifier .ccsproject /PR-SSFT-2272 line 6121, external location: C:\ti5\third_party\wolfssl\internal.h C/C++ Problem
    #80 expected a type specifier .ccsproject /PR-SSFT-2272 line 6380, external location: C:\ti5\third_party\wolfssl\internal.h C/C++ Problem
    #80 expected a type specifier .ccsproject /PR-SSFT-2272 line 6381, external location: C:\ti5\third_party\wolfssl\internal.h C/C++ Problem
    #80 expected a type specifier .ccsproject /PR-SSFT-2272 line 6391, external location: C:\ti5\third_party\wolfssl\internal.h C/C++ Problem
    #80 expected a type specifier .ccsproject /PR-SSFT-2272 line 6392, external location: C:\ti5\third_party\wolfssl\internal.h C/C++ Problem
    #80 expected a type specifier .ccsproject /PR-SSFT-2272 line 6393, external location: C:\ti5\third_party\wolfssl\internal.h C/C++ Problem
    #80 expected a type specifier .ccsproject /PR-SSFT-2272 line 6610, external location: C:\ti5\third_party\wolfssl\internal.h C/C++ Problem
    
    
 Additionally, our configuration file user_settings.h is set up as follows: 
    
```
#ifndef USER_SETTINGS_H
#define USER_SETTINGS_H

#define WOLFSSL_USER_SETTINGS

// Basic Settings
#define SINGLE_THREADED
#define WOLFSSL_GENERAL_ALIGNMENT 4

// Enable FreeRTOS and LwIP
#define FREERTOS
#define WOLFSSL_LWIP

// Enable specific cryptographic functionalities as needed
#define WOLFSSL_TLS
#define WOLFSSL_AES
#define WOLFSSL_SHA256
#define WOLFSSL_RSA
#define WOLFSSL_ECC
#define WOLFSSL_DH

// Uncomment if using Fast Math library
// #define USE_FAST_MATH

// Uncomment to enable timing resistance
// #define TFM_TIMING_RESISTANT
// #define ECC_TIMING_RESISTANT
// #define WC_RSA_BLINDING

// Uncomment if you're not using filesystem
// #define NO_FILESYSTEM

// Uncomment if you're not using writev()
// #define NO_WRITEV

// Uncomment if you're not using any of these algorithms
// #define NO_MD5
// #define NO_SHA
// #define NO_DES3
// #define NO_DSA
// #define NO_RC4

// Uncomment to disable server functionality
// #define NO_WOLFSSL_SERVER

// Uncomment to disable client functionality
// #define NO_WOLFSSL_CLIENT

// Advanced Settings (uncomment and modify as needed)
// #define WOLFSSL_STATIC_MEMORY
// #define WOLFSSL_STATIC_MEMORY_SMALL

// Debugging (uncomment to enable debug messages)
// #define DEBUG_WOLFSSL
// #define WOLFSSL_DEBUG

#endif /* USER_SETTINGS_H */
```
We believe that these issues may be related to configuration settings or compatibility problems between these libraries within our Code Composer Studio project environment. Any guidance on resolving these errors, especially considering our specific library versions and user_settings.h configuration, would be immensely helpful.

dgarske, 2023-12-06:
Hi @Can-Yildirim ,

You are not using valid build option macros or combinations of build options. Please see some user_settings.h examples here: https://github.com/wolfSSL/wolfssl/tree/master/examples/configs

You must have at least one asymmetric algorithm enabled. RSA and DH are on by default unless NO_RSA or NO_DH are set. ECC is enabled with HAVE_ECC. ED25519 is enabled with HAVE_ED25519.

You cannot have SINGLE_THREADED and FREERTOS both defined. Single threaded means bare-metal, no RTOS.

Thanks,
David Garske, wolfSSL

Can-Yildirim, 2023-12-07:
Hi David,

Quick update:

1. Build options issue resolved by correcting macro to `#ifndef WOLFSSL_USER_SETTINGS_H` and updating `user_settings_min_ecc.h`.
2. Enabled asymmetric algorithm 
3. Conflicts between `SINGLE_THREADED` and `FREERTOS` resolved; wolfSSL build successful.

**4. New issue: undefined "heap" identifier in wolfSSH's `internal.c` line 1209.** 
```
#ifndef WOLFSSH_USER_SETTINGS_H
#define WOLFSSH_USER_SETTINGS_H

#ifdef __cplusplus
extern "C" {
#endif

#include <wolfssl/wolfcrypt/types.h>

// Enable FreeRTOS and LwIP
#define FREERTOS
#define WOLFSSH_LWIP
#undef WOLFSSL_USER_IO

#undef WOLFSSH_USER_IO

#ifndef EWOULDBLOCK
#define EWOULDBLOCK  -1
#endif

#ifndef ECONNRESET
#define ECONNRESET   -2
#endif

#ifndef ECONNREFUSED
#define ECONNREFUSED -3
#endif

#ifndef ECONNABORTED
#define ECONNABORTED -4
#endif

// Enable ECC and specify key agreement algorithms
#define HAVE_ECC
#define HAVE_ECC_DHE  // Enable ECC Diffie-Hellman Ephemeral for key agreement
#define HAVE_ECDH     // If you want to use ECDH key exchange

// You can specify the curves you want to use, e.g., NIST P-256 (secp256r1)
#define HAVE_ECC256
// Add other curves as needed, for example, HAVE_ECC384 for 384-bit curves

// Enable SFTP and SCP features
//#define WOLFSSH_SFTP
//#define WOLFSSH_SCP

// Disable various test drivers
#define NO_APITEST_MAIN_DRIVER
#define NO_TESTSUITE_MAIN_DRIVER
#define NO_UNITTEST_MAIN_DRIVER
#define NO_MAIN_DRIVER

// Miscellaneous settings
#define WS_NO_SIGNAL
//#define WS_USE_TEST_BUFFERS
#define NO_WOLFSSL_DIR
#define WOLFSSH_NO_NONBLOCKING

// Window size and buffer settings
#define DEFAULT_WINDOW_SZ (128 * 128)
#define WOLFSSH_MAX_SFTP_RW 8192

#undef WOLFSSH_NO_ECDH_SHA2_NISTP256

#ifdef __cplusplus
}
#endif

#endif /* WOLFSSH_USER_SETTINGS_H */

```

Advice needed for the wolfSSH heap issue.

Best,
Can

dgarske, 2023-12-07:
Hi @Can-Yildirim ,

That sounds like excellent progress. The "heap" hint is used for static memory cases, where heap is not available.

I tried to locate line 1209, but it doesn't line up. Can you confirm which version of wolfSSH you are on?

Perhaps you can find the line reporting the error in current master?
https://github.com/wolfSSL/wolfssh/blob/master/src/internal.c#L1209

Thanks,
David Garske, wolfSSL

dgarske, 2023-12-07:
Hi @Can-Yildirim ,

I see above you mentioned wolfSSH v1.4.14. Here is internal.c line 1209 -> https://github.com/wolfSSL/wolfssh/blob/31c98b8c68acb10b755c3bfc008d6294de017586/src/internal.c#L1208

This is a valid report... current master is line 1247.  @ejohnstown will you review this?

Thanks,
David Garske, wolfSSL

Can-Yildirim, 2023-12-07:
Hi David,

Thanks for your response. We're currently using version 1.4.14-stable of wolfSSH. The error we mentioned is found in this version at this specific line: https://github.com/wolfSSL/wolfssh/blob/31c98b8c68acb10b755c3bfc008d6294de017586/src/internal.c#L1209.

Here's the relevant code snippet from lines 1184 to 1232 in `internal.c`:

```c
static int SetHostPrivateKey(WOLFSSH_CTX* ctx,
        byte keyId, byte* der, word32 derSz, int dynamicType)
{
    word32 destIdx = 0;
    int ret = WS_SUCCESS;

    /* Look for the specified keyId. Add it if not present,
    * replace it if present. Call UpdateHostCertificate().
    */

    while (destIdx < ctx->privateKeyCount
            && ctx->privateKey[destIdx].publicKeyFmt != keyId) {
        destIdx++;
    }

    if (destIdx >= WOLFSSH_MAX_PVT_KEYS) {
        ret = WS_CTX_KEY_COUNT_E;
    }
    else {
        WOLFSSH_PVT_KEY* pvtKey = ctx->privateKey + destIdx;

        if (pvtKey->publicKeyFmt == keyId) {
            if (pvtKey->key != NULL) {
                ForceZero(pvtKey->key, pvtKey->keySz);
                WFREE(pvtKey->key, heap, dynamicType);
            }
        }
        else {
            ctx->privateKeyCount++;
            pvtKey->publicKeyFmt = keyId;
        }

        pvtKey->key = der;
        pvtKey->keySz = derSz;

        #ifdef WOLFSSH_CERTS
        if (ret == WS_SUCCESS) {
            ret = UpdateHostCertificates(ctx, destIdx, WOLFSSH_MAX_PVT_KEYS);
        }
        #endif
        if (ret == WS_SUCCESS) {
            RefreshPublicKeyAlgo(ctx);
        }
    }

    WOLFSSH_UNUSED(dynamicType);

    return ret;
}
```

Along with the previous information, here's our `wolfssh/user_settings.h`

```c
#ifndef WOLFSSH_USER_SETTINGS_H
#define WOLFSSH_USER_SETTINGS_H

#ifdef __cplusplus
extern "C" {
#endif

#include <wolfssl/wolfcrypt/types.h>

// Enable FreeRTOS and LwIP
#define FREERTOS
#define WOLFSSH_LWIP
#undef WOLFSSL_USER_IO

#undef WOLFSSH_USER_IO

#ifndef EWOULDBLOCK
#define EWOULDBLOCK  -1
#endif

#ifndef ECONNRESET
#define ECONNRESET   -2
#endif

#ifndef ECONNREFUSED
#define ECONNREFUSED -3
#endif

#ifndef ECONNABORTED
#define ECONNABORTED -4
#endif

// Enable ECC and specify key agreement algorithms
#define HAVE_ECC
#define HAVE_ECC_DHE  // Enable ECC Diffie-Hellman Ephemeral for key agreement
#define HAVE_ECDH     // If you want to use ECDH key exchange

// You can specify the curves you want to use, e.g., NIST P-256 (secp256r1)
#define HAVE_ECC256
// Add other curves as needed, for example, HAVE_ECC384 for 384-bit curves

// Enable SFTP and SCP features
//#define WOLFSSH_SFTP
//#define WOLFSSH_SCP

// Disable various test drivers
#define NO_APITEST_MAIN_DRIVER
#define NO_TESTSUITE_MAIN_DRIVER
#define NO_UNITTEST_MAIN_DRIVER
#define NO_MAIN_DRIVER

// Miscellaneous settings
#define WS_NO_SIGNAL
//#define WS_USE_TEST_BUFFERS
#define NO_WOLFSSL_DIR
#define WOLFSSH_NO_NONBLOCKING

// Window size and buffer settings
#define DEFAULT_WINDOW_SZ (128 * 128)
#define WOLFSSH_MAX_SFTP_RW 8192

#undef WOLFSSH_NO_ECDH_SHA2_NISTP256

#ifdef __cplusplus
}
#endif

#endif /* WOLFSSH_USER_SETTINGS_H */
```

Your insight into the "heap" error in this context would be very helpful.

Thanks,
Can

dgarske, 2023-12-07:
Hi @Can-Yildirim ,

I was wrong. It is already fixed in latest master. We are putting together a wolfSSH release soon. Until then can you please try with the latest master branch?

Thanks,
David Garske, wolfSSL

Can-Yildirim, 2023-12-07:
> Hi @Can-Yildirim ,
> 
> I was wrong. It is already fixed in latest master. We are putting together a wolfSSH release soon. Until then can you please try with the latest master branch?
> 
> Thanks, David Garske, wolfSSL

Hi @dgarske,

Thanks! wolfSSH master branch worked perfectly, and we were able to build both wolfSSL and wolfSSH.

Best regards,
Can

ejohnstown, 2023-12-07:
Yes, that heap variable wasn't being used in the build typically used for testing. That `heap` wasn't used by the `WMALLOC` macro, and it wasn't declared, so the build test missed it. We are now using `ctx->heap` as it should be.

I'm glad the master branch is working better for you. Thanks!