wolfSSL/wolfssl #7047: [Bug]: TLS post-quantum cryptography crash (invalid memory free) when algorithm does not match

mspiegel, 2023-12-08:
### Contact Details

michael@hushmesh.com

### Version

5.6.4

### Description

When testing the post-quantum cryptography capabilities (`HAVE_PQC` enabled) we found the TLS server will crash when a client tries to connect and the algorithm does not match.  Google Chrome 116 will attempt to use X25519Kyber768 and that is an easy way to trigger the behavior. I have patched the issue locally and can provide our patch as a pull request. If it's easier, I will also describe the problem and you can patch it in your end.

In `src/tls.c` the functions `server_generate_pqc_ciphertext` and `TLSX_KeyShare_ProcessPqc`. Both of these functions invoke `wc_ecc_free(&eccpubkey);` and `wc_KyberKey_Free(kem);` unconditionally before the return statement at the end of the function. In both functions, there are circumstances where the functions `wc_ecc_init_ex()` and `wc_KyberKey_Init()` are either skipped or they are called and return failure. The calls to `wc_ecc_free()` and `wc_KyberKey_Free()` cause a crash under those conditions.

anhu, 2023-12-08:
Hi @mspiegel ,
Please have a look at https://github.com/wolfSSL/wolfssl/pull/7048 .   Let me know if it fixes your issue. 
Warm regards, Anthony.

mspiegel, 2023-12-08:
Hi Anthony. That fixes our issue. Thanks for the quick reply.

anhu, 2023-12-08:
Hi @mspiegel , 

That's excellent news.  Again, thank you so much for your interest and taking the time to report it.  Can you please let us know a bit about your self, your organization and your project? We always love to know how people are using our software. 

Warm regards, Anthony

closed 2023-12-12 by SparkiDev