wolfSSL/wolfssl #7088: Change of SONAME in a stable patch release

jirutka, 2023-12-21:
### Version

5.6.6

### Description

Breaking ABI compatibility / changing SONAME in a patch release (5.6.4 -> 5.6.6) is really not cool.

```diff
-usr/lib/libwolfssl.so.41.0.0:    SONAME               libwolfssl.so.41
+usr/lib/libwolfssl.so.42.0.0:    SONAME               libwolfssl.so.42
```

anhu, 2023-12-21:
Hello @jirutka , 

Thank you for reaching out to us.  We do guarantee ABI compatibility for those function marked `WOLFSSL_ABI` .  We do our best to stay backwards compatible and there is active discussion on a plan for your concerns internally.

JacobBarthelmeh, 2023-12-21:
Related to https://github.com/wolfSSL/wolfssl/issues/6928

Hi @jirutka, can you provide us some details about your use case with wolfSSL and how ABI would assist with it?

jirutka, 2023-12-25:
I’m the maintainer of the [wolfssl](https://pkgs.alpinelinux.org/packages?name=wolfssl) package in Alpine Linux. It’s in the community repository, so it’s in the stable releases (two a year) and we are committed to supporting it at least until the next stable release – security fixes without breaking backwards compatibility.

The change of SONAME is considered a breaking change (binaries linked against it must be rebuilt). Doing it in a patch release is against the semantic versioning and generally accepted best practices.

space88man, 2024-01-17:
+1 for soname stability — packaging for Fedora, ELx  a hypothetical  wolfssl-5.6.4 would provide libwolfssl.so.41 and wolfssl-5.6.6 would provide libwolfssl.so.42.

You could not "dnf upgrade" from wolfssl-5.6.4 to wolfssl-5.6.6.

Debian has libwolfssl35 libwolfssl41 libwolfssl42  etc — this is not sustainable.

closed 2025-12-10 by JacobBarthelmeh