wolfSSL/wolfssl #7093: [Bug]: SSL_read_early_data returns OUT_OF_ORDER_E after SSL_write

tatsuhiro-t, 2023-12-26:
### Version

3.6.6

### Description

./configure --enable-all --enable-quic --enable-aesni --enable-harden --enable-debug
linux amd64

### Reproduction steps

My server reads TLSv1.3 early data with SSL_read_early_data (I use OpenSSL compatible header to port my server from OpenSSL to wolfSSL).
If I understand TLSv1.3 early data correctly, server can send messages including application data without reading all early data.
This works for other TLS stacks, like OpenSSL and BoringSSL.  SSL_write is used for this purpose, and this works fine and client gets this data.  The problem is , after SSL_write, SSL_read_early_data returns OUT_OF_ORDER_E, and sends TLS alert, and connection is gone.

While I was poking around wolfSSL source code, I found that wolfSSL_write resets ssl->earlyData to no_early_data:
https://github.com/wolfSSL/wolfssl/blob/daf1d1728f8022d99464abb6c2917492345b24eb/src/ssl.c#L3255

This makes SSL_read_early_data misbehaving because  early data is suddenly disabled.
This line is probably there for some purpose, but it prevents sending application data before reading all early data from client.

embhorn, 2023-12-26:
Hello @tatsuhiro-t 

Thanks for providing this report. I am requesting a review of this issue by our engineers.

julek-wolfssl, 2023-12-27:
Hi @tatsuhiro-t, can you attach the debug logs of your test case? I am seeing a similar but different behaviour in our code. In my tests we are just ignoring the additional early data and not sending an alert. If you have a reproducer, please attach that as well.

tatsuhiro-t, 2023-12-27:
Thank you for reply.  I'll collect logs and share it later this day.  As for ignoring part of early data, TLSv1.3 also does not allow it.  Only server can do is either reject entirely, or accept all early data, or send HRR.

tatsuhiro-t, 2023-12-28:
Sorry I reported the wrong version: it should be 5.6.6.

tatsuhiro-t, 2023-12-28:
Attached wolfSSL log from nghttpx built from this branch: https://github.com/nghttp2/nghttp2/tree/wolfssl + wolfSSL_Debugging_ON(): [log.txt](https://github.com/wolfSSL/wolfssl/files/13783464/log.txt)

Send HTTP requests using [quictls](https://github.com/quictls/openssl) 1.1.1 client with the following patch:

```diff
diff --git a/apps/s_client.c b/apps/s_client.c
index 00effc8037..c54799ca6d 100644
--- a/apps/s_client.c
+++ b/apps/s_client.c
@@ -2745,7 +2745,7 @@ int s_client_main(int argc, char **argv)
         }
 
         while (!finish) {
-            if (!BIO_read_ex(edfile, cbuf, BUFSIZZ, &readbytes))
+            if (!BIO_read_ex(edfile, cbuf, 100, &readbytes))
                 finish = 1;
 
             while (!SSL_write_early_data(con, cbuf, readbytes, &writtenbytes)) {
@@ -2762,6 +2762,8 @@ int s_client_main(int argc, char **argv)
                     goto shut;
                 }
             }
+
+            usleep(10000);
         }
 
         BIO_free(edfile);
```

I think this patch should apply to vanilla openssl as well.

What this patch does is send early data in 2 packets to give server a chance to write before reading all early data.

early data is [http-req.txt](https://github.com/wolfSSL/wolfssl/files/13783465/http-req.txt)

julek-wolfssl, 2023-12-28:
@tatsuhiro-t Please try again with https://github.com/wolfSSL/wolfssl/pull/7102. This should fix your issue.

tatsuhiro-t, 2023-12-29:
It does fix my issue.  Thank you very much for quick fix.

closed 2024-01-04 by SparkiDev