wolfSSL/wolfssl #7095: Cannot Load Falcon or Sphincs certificates  with example server

NomanNasirMinhas, 2023-12-27:
### Version

5.6.6

### Description

I have built wolfssl with liboqs on WSL Ubuntu. I generated Sphincs, Falcon, and Dilithium certificates by using oqs-ossl3. The dilithum certificates are working fine with wolfssl but Falcon and Sphincs ones are failing with `wolfSSL error: can't load ca file, Please run from wolfSSL home dir`.
[image]

anhu, 2023-12-27:
Hello @NomanNasirMinhas , 

Thank you for letting us know about your problem. I'm glad at least Dilithium is working. 

Please note that we do not support SPHINCS+ in TLS 1.3 as it is simply not practical for online applications. 

Falcon should be working, but perhaps there has been some sort of change in recent versions of liboqs.  Can you please let me know what version of liboqs you are building against?  Just so you know, we build against  version 0.8.0 of liboqs. 

Warm regards, Anthony

Frauschi, 2024-01-04:
I think the mentioned issue with the Falcon certificates is related to the updates OIDs of Falcon in the current OQS code. With the updated OIDs (see #7109) I can load Falcon certificates generated by current OpenSSL with the OQS Provider.

dgarske, 2024-01-18:
Hi @NomanNasirMinhas , 

Please let us know if the fixes in https://github.com/wolfSSL/wolfssl/pull/7109 resolve this issue.

Thanks,
David Garske, wolfSSL

closed 2025-11-11 by anhu