wolfSSL/wolfssl #7105: I found MAX_MTU is 1400 when Sending data using DTLS1.2 protocol

QiangFan-cpu, 2024-01-03:
### Version

WolfSSL commercial v5.6.4

### Description

I found MAX_MTU is 1400 when Sending data using DTLS1.2.  I want to know about the value of MAX_MTU,
why set it to 1400.
Now,I using scene, I want to call "wolfssl_write()" to send 1500 byte data, but wolfssl checked the length of data and the size of MTU, If the length of data > 1400(MAX_MTU),would display error. I want to know the mentod of using way about sending data.

embhorn, 2024-01-03:
Hello @QiangFan-cpu 

Thanks for contacting wolfSSL Support. I am requesting a review of this issue by our engineers.

If this is a high priority request, please consider using our commercial support by emailing support@wolfssl.com

Thanks,
@embhorn - wolfSSL Support

ejohnstown, 2024-08-09:
The value of 1400 was chosen by subtracting 100 from the usual MTU size of 1500. The MTU size includes the size of the ethernet frame header (14), size of the IP header (40 for IPv6), the UDP header size (8), and the payload size. I then rounded it up to 100 to be conservative. The payload is going to include the size of the DTLS record header, padding, IV, and MAC. That means less than 1400 will be able to be transmit. If your network's MTU is greater than 1500 set that value higher. If you know you are using IPv4, set the MTU higher.