wolfSSL/wolfssl #7116: WolfSSL compilation on Zephyr RTOS fails for arm64 architecture boards

ickochar, 2024-01-05:
### Version

5.6.6

### Description

Following are metioned steps for compiling Zephyr application using wolfssl :
https://github.com/wolfSSL/wolfssl/tree/master/zephyr

Build command : west build -b qemu_kvm_arm64 modules/crypto/wolfssl/zephyr/samples/wolfssl_benchmark -d build -p

Error Logs :
[3/247] Generating include/generated/version.h
-- **Zephyr version: 3.4.0** (/p/psg/swip/w/ckochar/zephyrproject_3_3/zephyr), build: zephyr-v3.4.0-419-g564e20b0ae03
[237/247] Linking C executable zephyr/zephyr_pre0.elf
FAILED: zephyr/zephyr_pre0.elf zephyr/zephyr_pre0.map 
: && ccache /build/zephyr-sdk-0.15.1/aarch64-zephyr-elf/bin/aarch64-zephyr-elf-gcc  -gdwarf-4 zephyr/CMakeFiles/zephyr_pre0.dir/misc/empty_file.c.obj -o zephyr/zephyr_pre0.elf  -fuse-ld=bfd  -Wl,-T  zephyr/linker_zephyr_pre0.cmd  -Wl,-Map=/p/psg/swip/w/ckochar/zephyrproject_3_3/build/zephyr/zephyr_pre0.map  -Wl,--whole-archive  app/libapp.a  zephyr/libzephyr.a  zephyr/arch/common/libarch__common.a  zephyr/arch/arch/arm64/core/libarch__arm64__core.a  zephyr/lib/libc/minimal/liblib__libc__minimal.a  zephyr/lib/libc/common/liblib__libc__common.a  zephyr/lib/posix/liblib__posix.a  zephyr/drivers/interrupt_controller/libdrivers__interrupt_controller.a  zephyr/drivers/console/libdrivers__console.a  zephyr/drivers/serial/libdrivers__serial.a  zephyr/drivers/timer/libdrivers__timer.a  modules/wolfssl/lib..__modules__crypto__wolfssl__zephyr.a  -Wl,--no-whole-archive  zephyr/kernel/libkernel.a  zephyr/CMakeFiles/offsets.dir/./arch/arm64/core/offsets/offsets.c.obj  -L"/build/zephyr-sdk-0.15.1/aarch64-zephyr-elf/bin/../lib/gcc/aarch64-zephyr-elf/12.1.0"  -L/p/psg/swip/w/ckochar/zephyrproject_3_3/build/zephyr  -lgcc  zephyr/arch/common/libisr_tables.a  -mabi=lp64  -Wl,--gc-sections  -Wl,--build-id=none  -Wl,--sort-common=descending  -Wl,--sort-section=alignment  -Wl,-u,_OffsetAbsSyms  -Wl,-u,_ConfigAbsSyms  -nostdlib  -static  -Wl,-X  -Wl,-N  -Wl,--orphan-handling=warn  -Wl,-no-pie && cd /p/psg/swip/w/ckochar/zephyrproject_3_3/build/zephyr && /usr/local/bin/cmake -E echo
/build/zephyr-sdk-0.15.1/aarch64-zephyr-elf/bin/../lib/gcc/aarch64-zephyr-elf/12.1.0/../../../../aarch64-zephyr-elf/bin/ld.bfd: modules/wolfssl/lib..__modules__crypto__wolfssl__zephyr.a(random.c.obj): in function `sys_rand_get':
/nfs/png/disks/swuser_work_ckochar/zephyrproject_3_3/build/zephyr/include/generated/syscalls/rand32.h:57: undefined reference to `z_impl_sys_rand_get'
/nfs/png/disks/swuser_work_ckochar/zephyrproject_3_3/build/zephyr/include/generated/syscalls/rand32.h:57:(.text.wc_RNG_GenerateBlock+0xe8): relocation truncated to fit: R_AARCH64_CALL26 against undefined symbol `z_impl_sys_rand_get'
/build/zephyr-sdk-0.15.1/aarch64-zephyr-elf/bin/../lib/gcc/aarch64-zephyr-elf/12.1.0/../../../../aarch64-zephyr-elf/bin/ld.bfd: /nfs/png/disks/swuser_work_ckochar/zephyrproject_3_3/build/zephyr/include/generated/syscalls/rand32.h:57: undefined reference to `z_impl_sys_rand_get'
/nfs/png/disks/swuser_work_ckochar/zephyrproject_3_3/build/zephyr/include/generated/syscalls/rand32.h:57:(.text._InitRng+0x138): relocation truncated to fit: R_AARCH64_CALL26 against undefined symbol `z_impl_sys_rand_get'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: /usr/local/bin/cmake --build /nfs/png/disks/swuser_work_ckochar/zephyrproject_3_3/build

embhorn, 2024-01-05:
Hi @ickochar 

Thanks for contacting wolfSSL Support. Does your Zephyr project configuration enable entropy?
https://github.com/wolfSSL/wolfssl/blob/d5d476a3a1973fb9554f87fa84e1b587bdea70a0/zephyr/samples/wolfssl_benchmark/prj.conf#L27C1-L29C41

Could you please tell us a bit about your project using wolfSSL?

Thanks,
@embhorn - wolfSSL Support

ickochar, 2024-01-06:
Hello @embhorn ,
Thanks for the direction.

It seem like I didn't had entropy enabled. I have enabled in proj.cong :
# Entropy
**CONFIG_TEST_RANDOM_GENERATOR=y**
CONFIG_ENTROPY_GENERATOR=y
CONFIG_ENTROPY_DEVICE_RANDOM_GENERATOR=y

Actually I am trying to run benchmark results on my ARMv8 device with WOLFSSL_ARMASM defined in my setting file. Now below issues was observed. Am I missing any other config or setting ?

```
[251/261] Linking C executable zephyr/zephyr_pre0.elf
FAILED: zephyr/zephyr_pre0.elf zephyr/zephyr_pre0.map 
: && ccache /build/zephyr-sdk-0.15.1/aarch64-zephyr-elf/bin/aarch64-zephyr-elf-gcc  -gdwarf-4 zephyr/CMakeFiles/zephyr_pre0.dir/misc/empty_file.c.obj -o zephyr/zephyr_pre0.elf  -fuse-ld=bfd  -Wl,-T  zephyr/linker_zephyr_pre0.cmd  -Wl,-Map=/p/psg/swip/w/ckochar/zephyrproject_3_3/zephyr/build/zephyr/zephyr_pre0.map  -Wl,--whole-archive  app/libapp.a  zephyr/libzephyr.a  zephyr/arch/common/libarch__common.a  zephyr/arch/arch/arm64/core/libarch__arm64__core.a  zephyr/lib/libc/minimal/liblib__libc__minimal.a  zephyr/lib/libc/common/liblib__libc__common.a  zephyr/lib/posix/liblib__posix.a  zephyr/subsys/random/libsubsys__random.a  zephyr/drivers/interrupt_controller/libdrivers__interrupt_controller.a  zephyr/drivers/clock_control/libdrivers__clock_control.a  zephyr/drivers/console/libdrivers__console.a  zephyr/drivers/pm_cpu_ops/libdrivers__pm_cpu_ops.a  zephyr/drivers/reset/libdrivers__reset.a  zephyr/drivers/serial/libdrivers__serial.a  zephyr/drivers/syscon/libdrivers__syscon.a  zephyr/drivers/timer/libdrivers__timer.a  modules/wolfssl/lib..__modules__crypto__wolfssl__zephyr.a  -Wl,--no-whole-archive  zephyr/kernel/libkernel.a  zephyr/CMakeFiles/offsets.dir/./arch/arm64/core/offsets/offsets.c.obj  -L"/build/zephyr-sdk-0.15.1/aarch64-zephyr-elf/bin/../lib/gcc/aarch64-zephyr-elf/12.1.0"  -L/p/psg/swip/w/ckochar/zephyrproject_3_3/zephyr/build/zephyr  -lgcc  zephyr/arch/common/libisr_tables.a  -mcpu=cortex-a55  -mabi=lp64  -Wl,--gc-sections  -Wl,--build-id=none  -Wl,--sort-common=descending  -Wl,--sort-section=alignment  -Wl,-u,_OffsetAbsSyms  -Wl,-u,_ConfigAbsSyms  -nostdlib  -static  -Wl,-X  -Wl,-N  -Wl,--orphan-handling=warn  -Wl,-no-pie && cd /p/psg/swip/w/ckochar/zephyrproject_3_3/zephyr/build/zephyr && /usr/local/bin/cmake -E echo
/build/zephyr-sdk-0.15.1/aarch64-zephyr-elf/bin/../lib/gcc/aarch64-zephyr-elf/12.1.0/../../../../aarch64-zephyr-elf/bin/ld.bfd: app/libapp.a(benchmark.c.obj): in function `bench_aesgcm_internal':
/p/psg/swip/w/ckochar/zephyrproject_3_3/modules/crypto/wolfssl/wolfcrypt/benchmark/benchmark.c:4203: undefined reference to `wc_AesGcmSetKey'
/p/psg/swip/w/ckochar/zephyrproject_3_3/modules/crypto/wolfssl/wolfcrypt/benchmark/benchmark.c:4203:(.text.bench_aesgcm_internal.constprop.0+0x8c): relocation truncated to fit: R_AARCH64_CALL26 against undefined symbol `wc_AesGcmSetKey'
/build/zephyr-sdk-0.15.1/aarch64-zephyr-elf/bin/../lib/gcc/aarch64-zephyr-elf/12.1.0/../../../../aarch64-zephyr-elf/bin/ld.bfd: /p/psg/swip/w/ckochar/zephyrproject_3_3/modules/crypto/wolfssl/wolfcrypt/benchmark/benchmark.c:4259: undefined reference to `wc_AesGcmSetKey'
/p/psg/swip/w/ckochar/zephyrproject_3_3/modules/crypto/wolfssl/wolfcrypt/benchmark/benchmark.c:4259:(.text.bench_aesgcm_internal.constprop.0+0x134): relocation truncated to fit: R_AARCH64_CALL26 against undefined symbol `wc_AesGcmSetKey'
/build/zephyr-sdk-0.15.1/aarch64-zephyr-elf/bin/../lib/gcc/aarch64-zephyr-elf/12.1.0/../../../../aarch64-zephyr-elf/bin/ld.bfd: /p/psg/swip/w/ckochar/zephyrproject_3_3/modules/crypto/wolfssl/wolfcrypt/benchmark/benchmark.c:4220: undefined reference to `wc_AesGcmEncrypt'
/p/psg/swip/w/ckochar/zephyrproject_3_3/modules/crypto/wolfssl/wolfcrypt/benchmark/benchmark.c:4220:(.text.bench_aesgcm_internal.constprop.0+0x240): relocation truncated to fit: R_AARCH64_CALL26 against undefined symbol `wc_AesGcmEncrypt'
/build/zephyr-sdk-0.15.1/aarch64-zephyr-elf/bin/../lib/gcc/aarch64-zephyr-elf/12.1.0/../../../../aarch64-zephyr-elf/bin/ld.bfd: /p/psg/swip/w/ckochar/zephyrproject_3_3/modules/crypto/wolfssl/wolfcrypt/benchmark/benchmark.c:4275: undefined reference to `wc_AesGcmDecrypt'
/p/psg/swip/w/ckochar/zephyrproject_3_3/modules/crypto/wolfssl/wolfcrypt/benchmark/benchmark.c:4275:(.text.bench_aesgcm_internal.constprop.0+0x298): relocation truncated to fit: R_AARCH64_CALL26 against undefined symbol `wc_AesGcmDecrypt'
/build/zephyr-sdk-0.15.1/aarch64-zephyr-elf/bin/../lib/gcc/aarch64-zephyr-elf/12.1.0/../../../../aarch64-zephyr-elf/bin/ld.bfd: app/libapp.a(benchmark.c.obj): in function `bench_sha512_256':
/p/psg/swip/w/ckochar/zephyrproject_3_3/modules/crypto/wolfssl/wolfcrypt/benchmark/benchmark.c:6586: undefined reference to `wc_Sha512_256Free'
/p/psg/swip/w/ckochar/zephyrproject_3_3/modules/crypto/wolfssl/wolfcrypt/benchmark/benchmark.c:6586:(.text.bench_sha512_256+0x90): relocation truncated to fit: R_AARCH64_CALL26 against undefined symbol `wc_Sha512_256Free'
/build/zephyr-sdk-0.15.1/aarch64-zephyr-elf/bin/../lib/gcc/aarch64-zephyr-elf/12.1.0/../../../../aarch64-zephyr-elf/bin/ld.bfd: /p/psg/swip/w/ckochar/zephyrproject_3_3/modules/crypto/wolfssl/wolfcrypt/benchmark/benchmark.c:6504: undefined reference to `wc_InitSha512_256_ex'
/p/psg/swip/w/ckochar/zephyrproject_3_3/modules/crypto/wolfssl/wolfcrypt/benchmark/benchmark.c:6504:(.text.bench_sha512_256+0xb8): relocation truncated to fit: R_AARCH64_CALL26 against undefined symbol `wc_InitSha512_256_ex'
[... cut ...]
/build/zephyr-sdk-0.15.1/aarch64-zephyr-elf/bin/../lib/gcc/aarch64-zephyr-elf/12.1.0/../../../../aarch64-zephyr-elf/bin/ld.bfd: /p/psg/swip/w/ckochar/zephyrproject_3_3/modules/crypto/wolfssl/wolfcrypt/src/random.c:387: undefined reference to `wc_Sha256Free'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: /usr/local/bin/cmake --build /nfs/png/disks/swuser_work_ckochar/zephyrproject_3_3/zephyr/build
ckochar@ckochar-container-577df98c49-2g6vc:/p/psg/swip/w/ckochar/zephyrproject_3_3/zephyr
```

embhorn, 2024-01-23:
@philljj - was there a Zephyr benchmark issue resolved recently?

ickochar, 2024-01-24:
> Just so I understand your context better, as a sanity test have you built without `WOLFSSL_ARMASM`?
Yes, i was able to build without `WOLFSSL_ARMASM`

dgarske, 2024-03-18:
Hi @ickochar ,

I pushed support for the ARM and Intel assembly support to PR #7325.

Note: The `qemu_kvm_arm64` is a Cortex-A53 without crypto extensions, so our speedups that rely on `aes` and `sha256` instructions won't be available.

Let me know if you have any issues. Can you tell us more about your project?

Thanks,
David Garske, wolfSSL

ickochar, 2024-03-19:
> Let me know if you have any issues. Can you tell us more about your project?
We are trying to benchmark crypto operation performance  on our cortex-A53 hardware.

I tried to get your code and compile for our hardware. Seems like the changes are not compiling as expected. 
Error observed :
`/tmp/cclwNdTX.s: Assembler messages:
/tmp/cclwNdTX.s:58: Error: selected processor does not support 'sha256h q12,q13,v0.4s'`

Temporally I updated file with diff mentioned below and it was compiling. Not sure how to final fix will look like : 

> --- a/cmake/gcc-m-cpu.cmake
> +++ b/cmake/gcc-m-cpu.cmake
> @@ -72,7 +72,7 @@ if("${ARCH}" STREQUAL "arm")
>    endif()
>  elseif("${ARCH}" STREQUAL "arm64")
>    if(CONFIG_CPU_CORTEX_A53)
> -    set(GCC_M_CPU cortex-a53)
> +    set(GCC_M_CPU cortex-a53+crypto -mstrict-align)
>    elseif(CONFIG_CPU_CORTEX_A55)
>      set(GCC_M_CPU cortex-a55)
>    elseif(CONFIG_CPU_CORTEX_A76)

dgarske, 2024-08-22:
Hi @ickochar ,

Was this issue resolved? Can we mark this closed?