wolfSSL/wolfssl #7121: [Bug]: WOLFSSL_X509_STORE::param::check_times has no effect

shimunn, 2024-01-09:
### Version

5.6.6

### Description

The `check_time` does not seem to have an effect, the code only works if the system time is rewound(to make the cert appear to be valid). The "same" code written for OpenSSL works as expected.

I also couldn't find any instances in the wolfSSL source where `check_time` is used for verification.

### Reproduction steps

1. Generate an S/MIME signature signed by an expired CA Cert
2. try to verify the S/MIME signature with the expired cert while setting `WOLFSSL_X509_STORE::param::check_time` when adding the cert to the store

```c
  int ret = 0;
  WOLFSSL_BIO *in __attribute__((cleanup(wolfSSL_BIO_free_ptr))) =
      wolfSSL_BIO_new(wolfSSL_BIO_s_mem());
  wolfSSL_BIO_write(in, manifest_smime, manifest_smime_len);
  WOLFSSL_BIO *signed_data = NULL;
  WOLFSSL_PKCS7 *p7 =
      (WOLFSSL_PKCS7 *)wolfSSL_SMIME_read_PKCS7(in, &signed_data);
  if (p7 == NULL) {
    puts("pkcs7 fail");
    return;
  }
  unsigned char *manifest;
  if (signed_data != NULL) {
    int manifest_len = wolfSSL_BIO_get_mem_data(signed_data, &manifest);
    manifest[manifest_len] = '\0';
    printf("manifest (unverified): %s\n", manifest);
  } else {
    puts("failed to extract payload");
    return;
  }
  p7->pkcs7.devId = INVALID_DEVID;
  const unsigned char *pt = maintenance_ca_crt_der;
  WOLFSSL_X509 *x509 = wolfSSL_d2i_X509(NULL, &pt, maintenance_ca_crt_der_len);
  if (x509 == NULL) {
    puts("x509");
    return;
  }
  WOLFSSL_X509_STORE *store = wolfSSL_X509_STORE_new();
  if (store == NULL)
    puts("store");
  store->param->check_time = (time_t)1622020523;
  wolfSSL_X509_VERIFY_PARAM_set_flags(store->param, WOLFSSL_USE_CHECK_TIME);
  //wolfSSL_X509_VERIFY_PARAM_set_flags(store->param, WOLFSSL_NO_CHECK_TIME);
  printf("verify flags: %ld, time: %ld\n", store->param->flags,
         store->param->check_time);
  ret = wolfSSL_X509_STORE_add_cert(store, x509);
  if (ret != WOLFSSL_SUCCESS) {
    printf("failed to add cert %d\n", ret);
    return;
  }
  WOLFSSL_BIO *content __attribute__((cleanup(wolfSSL_BIO_free_ptr))) =
      wolfSSL_BIO_new(wolfSSL_BIO_s_mem());
  if (wolfSSL_Debugging_ON() == NOT_COMPILED_IN) {
    puts("no debug\n");
  }
  ret = wolfSSL_PKCS7_verify((PKCS7 *)p7, NULL, store, signed_data, content, 0);
  if (ret == WOLFSSL_SUCCESS) {
    printf("manifest: %s\n", manifest);
  } else {
    printf("verify: %d\n", ret);
    /* print out certificate that could not be verified */
    int i;
    byte *pt = p7->pkcs7.verifyCert;

    printf("Could not verify certificate: ");
    for (i = 0; i < p7->pkcs7.verifyCertSz; i++) {
      printf("%02X", pt[i]);
    }
    printf("\n");
    ret = -1;
  }
```

### Relevant log output

```shell
wolfSSL_PKCS7_verify returns 0
```

JacobBarthelmeh, 2024-02-28:
Hi @shimunn,

Thanks for the report! I have been reviewing this. It does look as though use of the check_time value is not integrated into the low level before/after certificate date checks. Am investigating into a fix.

Warm Regards,
Jacob

shimunn, 2024-05-07:
Has there been any progres towards an solution?

JacobBarthelmeh, 2024-05-08:
Hi shimunn, thank you for your patience on this. A code change or workaround has not been completed yet but we should be picking this back up soon.

Could you tell us some more about the use case and urgency level? Is this currently a blocker on the project working on?

JacobBarthelmeh, 2025-12-15:
The code change to support setting `check_time` was merged in with PR 9522.