wolfSSL/wolfssl #7125: [Bug]: syntax errors when compiling on Windows with QUIC enabled

farazrbx, 2024-01-10:
### Version

wolfSSL Release 5.6.6

### Description

Build fails on windows using msvc with some syntax errors when quic is enabled.
To fix it, I had to add `#include <stdint.h>` in `quic.h` for `uint8_t`
And define `#define STDERR_FILENO 2` in `api.c`

### Reproduction steps

````
mkdir build
cd build
cmake -DWOLFSSL_QUIC=yes ..
-- Building for: Visual Studio 17 2022
-- Selecting Windows SDK version 10.0.22621.0 to target Windows 10.0.22631.
-- The C compiler identification is MSVC 19.37.32825.0
-- The ASM compiler identification is MSVC
-- Found assembler: C:/Program Files/Microsoft Visual Studio/2022/Community/VC/Tools/MSVC/14.37.32822/bin/Hostx64/x64/cl.exe
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: C:/Program Files/Microsoft Visual Studio/2022/Community/VC/Tools/MSVC/14.37.32822/bin/Hostx64/x64/cl.exe - skipped
...

cmake --build .
```

### Relevant log output

```shell
C:\git\wolfssl-5.6.6\wolfssl\quic.h:60: error: C2143: syntax error: missing ')' before '*'
C:\git\wolfssl-5.6.6\wolfssl\quic.h:60: error: C2143: syntax error: missing ';' before '*'
C:\git\wolfssl-5.6.6\wolfssl\quic.h:60: error: C2059: syntax error: '*'
C:\git\wolfssl-5.6.6\wolfssl\quic.h:62: error: C2059: syntax error: ')'
C:\git\wolfssl-5.6.6\wolfssl\quic.h:68: error: C2143: syntax error: missing ')' before '*'
C:\git\wolfssl-5.6.6\wolfssl\quic.h:68: error: C2143: syntax error: missing '{' before '*'
...

C:\git\wolfssl-5.6.6\tests\api.c:56577: error: Use of undeclared identifier 'STDERR_FILENO'
C:\git\wolfssl-5.6.6\tests\api.c:56638: error: Use of undeclared identifier 'STDERR_FILENO'
```

dgarske, 2024-01-10:
https://github.com/wolfSSL/wolfssl/pull/7072/files/26b79008f1ec2513f382110e2a395a52f82952b7..5f74d01802a49fbb6493ca78ed2586f81586c8a7

farazrbx, 2024-03-01:
Fixed by https://github.com/wolfSSL/wolfssl/pull/7072

closed 2024-03-01 by farazrbx