wolfSSL/wolfssl #7130: [Question] Should SNI be parsed in early in a ClientHello callback a la OpenSSL?

space88man, 2024-01-16:
### Version

master

### Description

OpenSSL recommends manually parsing SNI in the ClientHello callback and use `SSL_CTX_set_tlsext_servername_callback`  for acknowledgement only. This supersedes using only `SSL_CTX_set_tlsext_servername_callback` and foregoes helper functions like `SSL_get_servername` as that is not available during the ClientHello callback. The reason for this new recommendation is not entirely clear but related to the order of parsing and that the SNI server callback happens quite late.

Does wolfSSL also follow have the same recommendation?

anhu, 2024-01-16:
Hello @space88man , 

These APIs are part of our OpenSSL compatibility layer. We recommend that you use these APIs as they would be expected to be used if you were building against OpenSSL.  

Please let me know if you have any further questions. 

Warm regards, Anthony

space88man, 2024-01-17:
Thanks @anhu : in my code I'm not expected to use the OpenSSL compatibility layer — what is the "correct" wolfSSL-way of doing SNI (and switching out the WOLFSSL_CTX *)?

> Hello @space88man ,
> 
> These APIs are part of our OpenSSL compatibility layer. We recommend that you use these APIs as they would be expected to be used if you were building against OpenSSL.
> 
> Please let me know if you have any further questions.
> 
> Warm regards, Anthony

anhu, 2024-01-17:
I think I understand now, 
in `doc/dox_comments/header_files/ssl.h` please look up the following APIs: 

- `wolfSSL_UseSNI`
- `wolfSSL_CTX_UseSNI`
- `wolfSSL_SNI_SetOptions`
- `wolfSSL_CTX_SNI_SetOptions`
- `wolfSSL_SNI_GetFromBuffer`
- `wolfSSL_SNI_Status`
- `wolfSSL_SNI_GetRequest`

That file show examples for how to use each of these APIs. You can find further examples of how to use this APIs in `examples/server/server.c`, `examples/client/client.c` and `tests/api.c` .  I hope this helps!!

Warm regards, Anthony

closed 2024-01-17 by space88man