wolfSSL/wolfssl #7153: Using DTLS with the example server

smeyer198, 2024-01-22:
### Version

5.6.6

### Description

I would like to use WolfSSL's example implementation of a DTLS server ([this one](https://github.com/wolfSSL/wolfssl/blob/master/examples/server/server.c)) to test to my DTLS client implementation. Using the CLI I try to start the server with the arguments `-p 8888 -c <path_to_my_certificate> -k <path_to_my_key> -u`. For some reason, I get the error
```
Start Server!
wolfSSL error: Bad SSL version
exit status 1
Server terminated! (4ms)
```
According to the documentation (and `--help`) option, the flag `-u` enables UDP and DTLS, and the default version is `DTLS 1.2` (i.e. `-v 3`). However, it seems that the parser still checks for an SSL version, which gives the error above. I tried to use different explicit version flags (i.e. `-v 2` and `-v 3`) without success. Is there anything I do wrong? Do I have to disable TLS to make sure that the versions are parsed for the correct DTLS version?

I hope someone can help me with my issue.

anhu, 2024-01-22:
Hi @smeyer198 , 

When you build wolfSSL, during the configure step, do you use `--enable-dtls` ? 

Warm regards, Anthony

smeyer198, 2024-01-22:
Hi,
you were right. I did not specify the `--enable-dtls` flag.

Thank you very much for your help! From my side the problem is solved.

closed 2024-01-22 by smeyer198

anhu, 2024-01-22:
Hi @smeyer198 , 

Glad to see you found success!!  Can I ask you about yourself and your project? We love knowing how people are using our code.  Please tell me a bit about what you are doing. 

Warm regards, Anthony.

smeyer198, 2024-01-23:
The project is part of my Masters Thesis in Computer Science. On a high level, I am doing some research on TLS and DTLS libraries, and implementations in general. Since wolfSSL provides example client and server implementations, it is ideal to test my server and client implementations, which will be used to analyze implementations in the wild. I would prefer to not going into detail on what I am doing exactly because I am not finished yet.

I selected wolfSSL for my studies because it is well documented and it is (as far as I know) the only open-source library that supports DTLS 1.3

I hope this answers your questions.

Kind regards, Sven.

anhu, 2024-01-23:
Hello @smeyer198 , 

Excellent.  thank you for all the information.  Please do get back to us with your thesis once it has been made public!  Here at wolfSSL, we love to highlight how the academic world is using our software  I might even be able to write a short blog post about your paper.  You can send it to anthony@wolfssl.com or facts@wolfssl.com . 

Warm regards, Anthony