wolfSSL/wolfssl #7160: [doc] Document that a BIO pair may need 2-pass read to retrieve all bytes due to ring buffer

space88man, 2024-01-22:
### Version

master

### Description

When a BIO pair has known NN bytes available from (`wolfSSL_BIO_pending)` it may still need two-pass `wolfSSL_BIO_read` to get all bytes due to ring buffer, i.e, the first read may be short, then the 2nd read will get the remainder.

This is an implementation detail as `wolfSSL_BIO_BIO_read` uses `wolfSSL_BIO_nread0` under the cover and does not perform wrap around (i.e it does only 1 x `XMEMCPY`).

This is different behaviour  from OpenSSL where the BIO_read on a BIO pair will do wrap around.
(The OpenSSL nread behaviour is the same as wolfSSL's nread behaviour). OpenSSL used two implementations - a BIO_read that does wrap around(and provides all bytes in one pass) and a BIO_nread that does not do wrap around.

julek-wolfssl, 2024-01-24:
Thank you for your report. I opened a pull request that solves your issue at https://github.com/wolfSSL/wolfssl/pull/7169. Please try it and let us know your feedback.

space88man, 2024-01-25:
Great - LGTM.

> Thank you for your report. I opened a pull request that solves your issue at #7169. Please try it and let us know your feedback.

closed 2024-01-25 by space88man