wolfSSL/wolfssl #7165: [Question] is wolfSSL_set_tmp_dh discouraged in wolfSSL(like OpenSSL)?

space88man, 2024-01-24:
### Version

master

### Description

I am maintaining some legacy code that does `wolfSSL_CTX_set_tmp_dh()`. The equivalent function in OpenSSL  is discouraged and it is recommended to use built-in  well-known DH parameters enabled with `SSL_CTX_set_dh_auto`.

There is no equivalent of `SSL_CTX_set_dh_auto` in woflSSL: is it necessary to set DH parameters or are there well-known defaults?

embhorn, 2024-01-24:
Hi @space88man 

I'll review this with the team tomorrow and provide an update.

Thanks,
@embhorn

embhorn, 2024-01-24:
Hi @space88man 

For TLS, by default wolfSSL uses predefined DH param groups. You can over-ride this using `wolfSSL_CTX_set_tmp_dh`.

space88man, 2024-02-04:
@embhorn - how does this work in practice?

In examples/echoserver/echoserver.c
```
332        #if !defined(NO_FILESYSTEM) && !defined(NO_DH) && !defined(NO_ASN)
333           wolfSSL_SetTmpDH_file(ssl, dhParamFile, WOLFSSL_FILETYPE_PEM);
334        #elif !defined(NO_DH)
335            SetDH(ssl);  /* will repick suites with DHE, higher than PSK */
336      #endi
```

If I comment out the line with `wolfSSL_SetTmpDH...` then any connection attempt with a `DHE_RSA_XXX` cipher fails. From the echoserver perspective

```
# client sends
openssl s_client -connect localhost:11111 -tls1_2 -cipher DHE-RSA-AES256-SHA256

# server reports
./echoserver 
# I dump cipher list just to be sure DHE_RSA_XXX is supported
wolfSSL supported ciphers = TLS13-AES128-GCM-SHA256:TLS13-AES256-GCM-SHA384:TLS13-CHACHA20-POLY1305-SHA256:TLS13-AES128-CCM-SHA256:TLS13-AES128-CCM-8-SHA256:TLS13-AES128-CCM8-SHA256:TLS13-SHA256-SHA256:TLS13-SHA384-SHA384:DHE-RSA-AES128-SHA:DHE-RSA-AES256-SHA:DHE-PSK-AES256-GCM-SHA384:DHE-PSK-AES128-GCM-SHA256:DHE-PSK-AES256-CBC-SHA384:DHE-PSK-AES128-CBC-SHA256:DHE-PSK-AES128-CCM:DHE-PSK-AES256-CCM:DHE-PSK-NULL-SHA384:DHE-PSK-NULL-SHA256:ECDHE-ECDSA-AES128-CCM:ECDHE-ECDSA-AES128-CCM-8:ECDHE-ECDSA-AES128-CCM8:ECDHE-ECDSA-AES256-CCM-8:ECDHE-ECDSA-AES256-CCM8:ECDHE-RSA-AES128-SHA:ECDHE-RSA-AES256-SHA:ECDHE-ECDSA-AES128-SHA:ECDHE-ECDSA-AES256-SHA:ECDHE-RSA-RC4-SHA:ECDHE-RSA-DES-CBC3-SHA:ECDHE-ECDSA-RC4-SHA:ECDHE-ECDSA-DES-CBC3-SHA:DHE-RSA-AES128-SHA256:DHE-RSA-AES256-SHA256:DHE-RSA-AES128-GCM-SHA256:DHE-RSA-AES256-GCM-SHA384:ECDHE-RSA-AES128-GCM-SHA256:ECDHE-RSA-AES256-GCM-SHA384:ECDHE-ECDSA-AES128-GCM-SHA256:ECDHE-ECDSA-AES256-GCM-SHA384:DHE-RSA-CAMELLIA128-SHA:DHE-RSA-CAMELLIA256-SHA:DHE-RSA-CAMELLIA128-SHA256:DHE-RSA-CAMELLIA256-SHA256:ECDHE-RSA-AES128-SHA256:ECDHE-ECDSA-AES128-SHA256:ECDHE-RSA-AES256-SHA384:ECDHE-ECDSA-AES256-SHA384:ECDHE-RSA-CHACHA20-POLY1305:ECDHE-ECDSA-CHACHA20-POLY1305:DHE-RSA-CHACHA20-POLY1305:ECDHE-RSA-CHACHA20-POLY1305-OLD:ECDHE-ECDSA-CHACHA20-POLY1305-OLD:DHE-RSA-CHACHA20-POLY1305-OLD:ADH-AES128-SHA:ADH-AES256-GCM-SHA384:ECDHE-ECDSA-NULL-SHA:ECDHE-PSK-NULL-SHA256:ECDHE-PSK-AES128-CBC-SHA256:ECDHE-PSK-AES128-GCM-SHA256:PSK-CHACHA20-POLY1305:ECDHE-PSK-CHACHA20-POLY1305:DHE-PSK-CHACHA20-POLY1305:EDH-RSA-DES-CBC3-SHA:WDM-NULL-SHA256
SSL_accept error = -501, can't match cipher suite
SSL_accept failed
```

> Hi @space88man
> 
> For TLS, by default wolfSSL uses predefined DH param groups. You can over-ride this using `wolfSSL_CTX_set_tmp_dh`.

embhorn, 2024-02-21:
This question has been answered. Please feel free to reopen if there are further concerns.

closed 2024-02-21 by embhorn