wolfSSL/wolfssl #7196: [Bug]: Write-Dup with chaha poly1305 write fails with "error fatal"

deforation, 2024-02-01:
### Contact Details

remo.schweizer.it@gmail.com

### Version

5.6.6

### Description

It seems the write-duplication does not work in combination with Chacha Poly1305.
All wolfssl_write or send calls fail with "-1" Fatal error.

After tracking the issue down, it turns out, 
`ssl->auth.poly1305` is never set on the write-dup ssl.
Thus in the end the poly1305 encryption will fail due to the fact that that above pointer is NULL.

Not sure if this is a proper fix, but the following modification solved it for me.
```
// Extending the DupSSL function in ssl.c with the following lines

#ifdef HAVE_ONE_TIME_AUTH
    dup->auth.setup = ssl->auth.setup;
	#ifdef HAVE_POLY1305
	    dup->auth.poly1305 = malloc(sizeof(Poly1305));
	    XMEMCPY(&dup->auth.poly1305, &ssl->auth.poly1305, sizeof(Poly1305));
	#endif
#endif
```

Another change that had to be made is the following:
The current master of the function `wolfSSL_write_dup` looks like this.
```
        if ( (ret = InitSSL(dup, ssl->ctx, 1)) < 0) {
            FreeSSL(dup, ssl->ctx->heap);
            dup = NULL;
        } else if ( (ret = DupSSL(dup, ssl)) < 0) {
            FreeSSL(dup, ssl->ctx->heap);
            dup = NULL;
        }
```
From my understanding it should more look like this:
```
        if ( (ret = InitSSL(dup, ssl->ctx, 1)) < 0) {
            FreeSSL(dup, ssl->ctx->heap);
            dup = NULL;
        } 
        // else removed
        if ( (ret = DupSSL(dup, ssl)) < 0) {
            FreeSSL(dup, ssl->ctx->heap);
            dup = NULL;
        }
```
Otherwise internal data will never be copied.

Not sure if the issue also exists in combination with other cipher suites.
Just checked this one so far as it was auto negotiated and just didn't work till above fix was applied.

### Reproduction steps

1. ./configure --enable-session-ticket --enable-rwlock --enable-savecert --enable-savesession --enable-threadlocal --enable-error-queue-per-thread --enable-enckeys --disable-harden --enable-sslv3 --disable-aesgcm --disable-sp --enable-staticmemory --enable-nullcipher --enable-intelasm --enable-static=yes --enable-writedup --enable-64bit --enable-singlethreaded --enable-fastmath --enable-fasthugemath --enable-fast-rsa
2. Create a socket and tls connection that connects to a server with the ciphersuite chacha poly1305
3. Use the duplicated ssl handle returned by wolfSSL_write_dup for any send or write operation

dgarske, 2024-02-01:
@julek-wolfssl can you review this bug report?

deforation, 2024-02-01:
Made an adjustment to the copy part.
As it had an error in it.
Also removed the second wrong statement about the dup function in general.

julek-wolfssl, 2024-02-02:
Hi @deforation,

thank you for the report. Please try the fix posted in https://github.com/wolfSSL/wolfssl/pull/7206.

Juliusz

deforation, 2024-02-04:
@julek-wolfssl Thanks, the posted merge requests solves the issue for me.

closed 2024-02-07 by dgarske