wolfSSL/wolfssl #7197: [Bug]: [haproxy] blocking when using chroot + wolfssl

wlallemand, 2024-02-01:
### Version

5.6.6

### Description

HAProxy has a "chroot" primitive which is often used by users. With OpenSSL, Rand_Bytes()  is called before chroot() so OpenSSL is able to open /dev/urandom and keep the FD. Once HAProxy has done its chroot(), the random is fed from this FD.

With WolfSSL, its seems that wc_GenerateSeed() is not keeping the fd and is closing it each time, which means once chroot'ed, haproxy does not have access anymore to the random source, and every requests are blocking.

It looks like the only way to make this work, is to stop using /dev/urandom and use getrandom(), by building wolfSSL with WOLFSSL_GETRANDOM.

Is there a way to keep to the /dev/urandom open during init and keep using it?

Thanks

### Relevant log output

```shell
https://github.com/wolfSSL/wolfssl/blob/master/wolfcrypt/src/random.c#L3775
```

wlallemand, 2024-05-24:
Hello, Any update on this?

cervajs, 2024-05-24:
+1 for this. wolfssl is raising as very good alternative to OpenSSL for http/3 in HAproxy and this problem can be "confusing" for users

closed 2025-12-30 by douzzer