wolfSSL/wolfssl #7210: esp32_sha errors -> 5.6.6-stable-update2-esp32

AchimPieters, 2024-02-03:
### Version

5.6.6

### Description

```
FAILED: esp-idf/wolfssl/CMakeFiles/__idf_wolfssl.dir/wolfcrypt/src/port/Espressif/esp32_sha.c.obj 
ccache /opt/esp/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/xtensa-esp32-elf-gcc -DESP_PLATFORM -DIDF_VER=\"v5.3-dev-1353-gb3f7e2c8a4\" -DLIBWOLFSSL_VERSION_GIT_BRANCH=\"main\" -DLIBWOLFSSL_VERSION_GIT_HASH=\"eb4487cf7e68dcf34a4a0781d933c8cc1e05c9d7\" -DLIBWOLFSSL_VERSION_GIT_HASH_DATE="\"'Mon Jan 22 17:41:44 2024 +0100'\"" -DLIBWOLFSSL_VERSION_GIT_ORIGIN=\"https://github.com/AchimPieters/esp32-homekit-demo.git\" -DLIBWOLFSSL_VERSION_GIT_SHORT_HASH=\"eb4487c\" -DLIBWOLFSSL_VERSION_WOLFSSL_ROOT=\"/project/components/wolfssl\" -DSOC_MMU_PAGE_SIZE=CONFIG_MMU_PAGE_SIZE -DSOC_XTAL_FREQ_MHZ=CONFIG_XTAL_FREQ -DWOLFSSL_USER_SETTINGS_DIR=\"//project//examples//led//components//wolfssl//include//user_settings.h\" -D_GLIBCXX_HAVE_POSIX_SEMAPHORE -D_GLIBCXX_USE_POSIX_SEMAPHORE -D_GNU_SOURCE -D_POSIX_READER_WRITER_LOCKS -I/project/examples/led/build/config -I/project/components/wolfssl/include -I/project/components/wolfssl -I/project/components/wolfssl/wolfssl -I/project/components/wolfssl/wolfssl/wolfcrypt -I/project/components/wolfssl/wolfssl/wolfcrypt/port/Espressif -I/opt/esp/idf/components/freertos/FreeRTOS-Kernel/include/freertos -I/opt/esp/idf/components/newlib/platform_include -I/opt/esp/idf/components/freertos/config/include -I/opt/esp/idf/components/freertos/config/include/freertos -I/opt/esp/idf/components/freertos/config/xtensa/include -I/opt/esp/idf/components/freertos/FreeRTOS-Kernel/include -I/opt/esp/idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/include -I/opt/esp/idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/include/freertos -I/opt/esp/idf/components/freertos/esp_additions/include -I/opt/esp/idf/components/esp_hw_support/include -I/opt/esp/idf/components/esp_hw_support/include/soc -I/opt/esp/idf/components/esp_hw_support/include/soc/esp32 -I/opt/esp/idf/components/esp_hw_support/port/esp32/. -I/opt/esp/idf/components/heap/include -I/opt/esp/idf/components/log/include -I/opt/esp/idf/components/soc/include -I/opt/esp/idf/components/soc/esp32 -I/opt/esp/idf/components/soc/esp32/include -I/opt/esp/idf/components/hal/platform_port/include -I/opt/esp/idf/components/hal/esp32/include -I/opt/esp/idf/components/hal/include -I/opt/esp/idf/components/esp_rom/include -I/opt/esp/idf/components/esp_rom/include/esp32 -I/opt/esp/idf/components/esp_rom/esp32 -I/opt/esp/idf/components/esp_common/include -I/opt/esp/idf/components/esp_system/include -I/opt/esp/idf/components/esp_system/port/soc -I/opt/esp/idf/components/esp_system/port/include/private -I/opt/esp/idf/components/xtensa/esp32/include -I/opt/esp/idf/components/xtensa/include -I/opt/esp/idf/components/xtensa/deprecated_include -I/opt/esp/idf/components/lwip/include -I/opt/esp/idf/components/lwip/include/apps -I/opt/esp/idf/components/lwip/include/apps/sntp -I/opt/esp/idf/components/lwip/lwip/src/include -I/opt/esp/idf/components/lwip/port/include -I/opt/esp/idf/components/lwip/port/freertos/include -I/opt/esp/idf/components/lwip/port/esp32xx/include -I/opt/esp/idf/components/lwip/port/esp32xx/include/arch -I/opt/esp/idf/components/lwip/port/esp32xx/include/sys -mlongcalls -Wno-frame-address  -DWOLFSSL_USER_SETTINGS -DWOLFSSL_CMAKE_SYSTEM_NAME_LINUX -fdiagnostics-color=always -ffunction-sections -fdata-sections -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-variable -Wno-error=unused-but-set-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -Wno-enum-conversion -gdwarf-4 -ggdb -Og -fno-shrink-wrap -fmacro-prefix-map=/project/examples/led=. -fmacro-prefix-map=/opt/esp/idf=/IDF -fstrict-volatile-bitfields -fno-jump-tables -fno-tree-switch-conversion -std=gnu17 -Wno-old-style-declaration -MD -MT esp-idf/wolfssl/CMakeFiles/__idf_wolfssl.dir/wolfcrypt/src/port/Espressif/esp32_sha.c.obj -MF esp-idf/wolfssl/CMakeFiles/__idf_wolfssl.dir/wolfcrypt/src/port/Espressif/esp32_sha.c.obj.d -o esp-idf/wolfssl/CMakeFiles/__idf_wolfssl.dir/wolfcrypt/src/port/Espressif/esp32_sha.c.obj -c /project/components/wolfssl/wolfcrypt/src/port/Espressif/esp32_sha.c
/project/components/wolfssl/wolfcrypt/src/port/Espressif/esp32_sha.c: In function 'esp_sha_ctx_copy':
/project/components/wolfssl/wolfcrypt/src/port/Espressif/esp32_sha.c:531:12: error: invalid use of undefined type 'struct wc_Sha'
  531 |     if (src->ctx.mode == ESP32_SHA_HW) {
      |            ^~
/project/components/wolfssl/wolfcrypt/src/port/Espressif/esp32_sha.c:545:16: error: invalid use of undefined type 'struct wc_Sha'
  545 |             dst->ctx.mode = ESP32_SHA_HW_COPY;
      |                ^~
/project/components/wolfssl/wolfcrypt/src/port/Espressif/esp32_sha.c:548:37: error: invalid use of undefined type 'struct wc_Sha'
  548 |             ret = esp_sha_init(&(dst->ctx), WC_HASH_TYPE_SHA);
      |                                     ^~
/project/components/wolfssl/wolfcrypt/src/port/Espressif/esp32_sha.c:559:16: error: invalid use of undefined type 'struct wc_Sha'
  559 |         if (dst->ctx.mode == ESP32_SHA_SW) {
      |                ^~
/project/components/wolfssl/wolfcrypt/src/port/Espressif/esp32_sha.c:585:12: error: invalid use of undefined type 'struct wc_Sha'
  585 |         dst->ctx.initializer = &(dst->ctx); /* assign new breadcrumb to dst */
      |            ^~
/project/components/wolfssl/wolfcrypt/src/port/Espressif/esp32_sha.c:585:37: error: invalid use of undefined type 'struct wc_Sha'
  585 |         dst->ctx.initializer = &(dst->ctx); /* assign new breadcrumb to dst */
      |                                     ^~
/project/components/wolfssl/wolfcrypt/src/port/Espressif/esp32_sha.c:589:16: error: invalid use of undefined type 'struct wc_Sha'
  589 |             dst->ctx.task_owner = 0;
      |                ^~
ninja: build stopped: subcommand failed.
ninja failed with exit code 1, output of the command is in the /project/examples/led/build/log/idf_py_stderr_output_41832 and /project/examples/led/build/log/idf_py_stdout_output_41832
root@22fb157df2ff:/project/examples/led# 
```

For reproduction and latest changes, see: https://github.com/AchimPieters/esp32-homekit-demo

gojimmypi, 2024-02-03:
Hello @AchimPieters - I've been able to reproduce the problem you are seeing.

The cause appears to be that with hardware acceleration enabled (by default), the ESP32 libraries do not properly handle [this](https://github.com/AchimPieters/esp32-homekit-demo/blob/26cf558a46b3a9f10b79b711a9209326aa7932a3/components/wolfssl/include/user_settings.h#L143) define in your `user_settings.h`:

```
#define NO_SHA
```

Remove that line should fix the compile problem you are seeing. 

Additionally, a few other observations and recommendations:

Your example contains an [entire copy of wolfSSL](https://github.com/AchimPieters/esp32-homekit-demo/tree/main/components/wolfssl). I've (painfully) learned this can be problematic to maintain and troublehoot. The latest versions of the `CMakeLists.txt` allow for specification of where the wolfSSL source code is located, by setting `WOLFSSL_ROOT`. See the example [template](https://github.com/wolfSSL/wolfssl/blob/851f0590239475be5f00858123a5ded4e0a21c3f/IDE/Espressif/ESP-IDF/examples/template/components/wolfssl/CMakeLists.txt#L53).

There's also a potential problem of distributing wolfSSL. I'm not a licensing expert, so it is probably best to contact support@wolfssl.com

It is best to include the wolfSSL `settings.h` in _every file_ that uses wolfSSL. Include this before referencing any other wolfssl files. Do not explicitly include `user_settings.h` yourself as the `settings.h.` pulls it in as appropriate. Later inclusion of the `user_settings.h` may undo some of the `settings.h` actions, leading to unpredictable results.

I'm leaving this issue open, as the `NO_SHA` should be valid regardless of whether hardware encryption is enabled or not.

AchimPieters, 2024-02-04:
@gojimmypi The No SHA did the trick, git some errors that are hardware specific related, but not with Wolfssl.

I'm going to see if I can add wolfssl as a submodule: https://components.espressif.com/components/wolfssl/wolfssl
(License: [Custom](https://components.espressif.com/components/wolfssl/wolfssl/versions/5.6.6-stable-update2-esp32/license))

If you have any recommendations, I would like to here them.

AchimPieters, 2024-02-07:
@gojimmypi what is the direct link, for adding the wolfssl as a submodule e.g. https://components.espressif.com/components/wolfssl/wolfssl?

gojimmypi, 2024-02-07:
Hello @AchimPieters !

> I'm going to see if I can add wolfssl as a submodule ... If you have any recommendations, I would like to here them.

Yes, I really need to put a blog together to better explain that.

First: I'm not so much a fan of using [submodules](https://git-scm.com/book/en/v2/Git-Tools-Submodules), as least not as a developer. It's great for distribution of a final product but does not lend itself to being able to easily make changes.

Regarding the link you mentioned, that's an [Espressif Managed Component](https://docs.espressif.com/projects/idf-component-manager/en/latest/). It is a wonderful tool for easily getting started, but that too - does not lend itself to making changes as a developer. In fact, it is even less friendly than a submodule: as soon as a change is detected, it complains. It is great for maintaining code integrity, but not for making changes.

In your example, I deleted everything in this directory except for the `CMakeLists.txt` file and the `include` directory:

```
esp32-homekit-demo\components\wolfssl
```
 
I then made one change to the `CMakeLists.txt` file, pointing to my `wolfssl-arduino` directory. (I have wolfssl cloned there for Arduino-specific changes).

```
# Optionally set your source to wolfSSL in your project CMakeLists.txt like this:
set(WOLFSSL_ROOT "c:/workspace/wolfssl-arduino" )
```

I was able to successfully compile the example LED project that references wolfSSL.

The cool thing here that as a developer, we can more easily see changes and contribute. The library can be shared among as many projects as desired, and versions can easily be changed either with git hashes or even pointing to a completely different directory.

Regarding the more recent SRP errors you are seeing: is that for the LED or some other example? I'm wondering if there's an issue finding the `user_settings.h` and/or the settings therein. I would need more details.

I might try isolating the project and using wolfSSL as a local component as seen in the [examples](https://github.com/wolfSSL/wolfssl/tree/master/IDE/Espressif/ESP-IDF/examples).

AchimPieters, 2024-02-08:
Hello @gojimmypi,

`I was able to successfully compile the example LED project that references wolfSSL.` 

Also for the esp32C3 ?

`In your example, I deleted everything in this directory except for the CMakeLists.txt file and the include directory:`

Did you do so with my current version, or my `old` version?

Maybe you can guide my through your steps and resources?

gojimmypi, 2024-02-12:
> Did you do so with my current version, or my old version?

I did this with the same code clone mentioned above, so within the last couple of weeks.

> Also for the esp32C3 or ESPC2?

I only tested for the ESP32, but would expect similar successful results with others such the -C3 ad -C2.  Did you see different results?
 
>but noting works any more?

I'll need a little more detail, please.

> Maybe clone my latest repro to see what I mean?

I'll revisit switch your latest code soon. I just need to finish up a few customer requests. 

Any additional information would be helpful, such as specific error messages, please.

AchimPieters, 2024-02-13:
Reproduction steps

Open a terminal window:
Install docker
Install IDF Latest - docker pull espressif/idf:latest
Clone repository - git clone --recursive https://github.com/AchimPieters/esp32-homekit-demo.git
start IDF -  docker run -it -v ~/esp32-homekit-demo:/project -w /project espressif/idf:latest
cd examples 
cd led
idf.py set-target esp32
idf.py menuconfig
 - Serial flasher config  ---> Flash size (2 MB)  ---> (X) 4 MB
 - StudioPieters  ---> (myssid) WiFi SSID and (mypassword) WiFi Password
 -  Component config  --->  HomeKit  ---> [*] Debug output
Save configuration
idf.py build

Open a new (second) terminal window:
cd /Users/USERNAME/esp32-homekit-demo/examples/led
esptool.py erase_flash
python -m esptool --chip esp32 -b 460800 --before default_reset --after hard_reset write_flash --flash_mode dio --flash_size 4MB --flash_freq 40m 0x1000 build/bootloader/bootloader.bin 0x8000 build/partition_table/partition-table.bin 0x10000 build/main.bin

screen /dev/tty.usbserial-01FD1166 115200 

```
I (3828) IP_EVENT: WiFI ready
>>> HomeKit: Starting server
>>> HomeKit: Using existing accessory ID: 25:1B:8A:18:79:D5
>>> HomeKit: Configuring mDNS
>>> homekit_setup_mdns: Accessory Setup ID = 1QJ8
>>> homekit_run_server: Starting HTTP server
>>> homekit_server_accept_client: Free heap: 210832
>>> HomeKit: [Client 1] Got new client connection from 192.168.178.145
>>> homekit_client_process: [Client 1] Got 132 incoming data
>>> homekit_server_on_pair_setup: Pair Setup
>>> homekit_server_on_pair_setup: Free heap: 211220
>>> tlv_debug: Got following TLV values:
>>> tlv_debug: Type 0 value (1 bytes): \x00
>>> tlv_debug: Type 6 value (1 bytes): \x01
>>> HomeKit: [Client 1] Pair Setup Step 1/3
>>> homekit_server_on_pair_setup: Free heap: 211116
>>> crypto_srp_new: Initializing SRP
>>> homekit_server_on_pair_setup: [Client 1] Initializing crypto
>>> homekit_server_on_pair_setup: Free heap: 202140
>>> homekit_server_on_pair_setup: [Client 1] Using user-specified password: 338-77-883
>>> crypto_srp_init: Generating salt
>>> crypto_srp_init: Setting SRP username
>>> crypto_srp_init: Setting SRP params
>>> crypto_srp_init: Setting SRP password
>>> crypto_srp_init: Getting SRP verifier
>>> crypto_srp_init: Setting SRP verifier
>>> crypto_srp_get_public_key: Calculating public key
>>> client_sendv: [Client 1] Sending payload: HTTP/1.1 200 OK\x0D\x0AContent-Type: application/pairing+tlv8\x0D\x0ATransfer-Encoding: chunked\x0D\x0AConnection: keep-alive\x0D\x0A\x0D\x0A
>>> client_sendv: [Client 1] Sending payload: \x31\x39\x39\x0D\x0A\x03\xFF\x6D\x0B\xF3\x51\x14\x0C\x43\xEB\xD6\x2E\x05\x72\x98\x24\x22\xB3\x2A\x8E\xAB\x6F\x9C\x72\x16\x92\x2F\xE1\xF1\xFF\xD2\x95\xCC\xAD\x22\x19\x0A\xD8\xCE\x67\x08\xE8\x1C\x1D\x5F\xCE\x17\x22\x91\xA8\x3E\xA7\xBF\xBA\x77\x25\x0E\xC9\x25\x07\x5B\x8A\x4F\xF7\x27\x69\x74\x99\x54\x51\x79\xF6\x06\xF8\x00\x4E\x18\xEC\x26\x3B\xC2\x38\x2F\xFF\xAB\x1D\x6F\x67\x90\xFF\x98\x31\x94\xA5\x75\x38\x7B\x1B\x7D\x44\x4D\xB3\xBD\x97\x31\xB4\xCE\x3D\x28\x84\xD5\xF1\x67\x84\x0A\xF8\x95\x94\xFE\x7D\xB3\xA9\x0C\xAA\xBF\x2E\x76\x62\x73\x18\xC0\x8D\xBD\x40\x75\x6B\xA5\x86\x2C\x55\x7E\x1D\x51\xBE\xCE\xF9\x95\x52\x4C\x4E\x0C\x5B\xCD\xCF\x83\x65\xAE\xEA\x70\x37\x06\xC8\x14\xAF\x61\x7A\xFC\x89\x90\xA0\x28\xBC\x34\x89\x27\x2A\xEF\x86\xA5\x9C\x31\xCB\xE5\xC4\x10\x4B\x88\x3D\xF7\x16\xD3\x03\xD9\x34\xA1\x2E\x10\xF0\x66\x96\x73\x6F\x41\x22\x27\x15\x7B\x1F\x6D\xC8\x6D\x71\xEE\x26\x9D\xCE\x61\xAB\x92\x5B\xD5\xD2\xC5\xFE\x72\xFF\x68\x6B\x11\x51\xE8\x68\xC8\x0D\xDC\x0E\xE8\xC7\x1F\xF7\x3A\x4E\x1A\x0E\xEE\x0F\x29\x72\xA0\xD7\x36\x23\x83\x4D\x70\xE2\xE3\x03\x81\x9E\xEB\x34\xC7\x98\xEC\x42\x2F\xED\xCC\xCD\x04\x45\x02\x18\x05\x49\x0E\xFE\x81\x09\x06\xC6\x38\xD1\x27\xBA\xC1\x3A\xAB\xAF\x0C\x80\x75\xAC\x79\x0D\xC9\xFA\x9E\x9D\x24\xEC\x3F\x66\x4E\x6F\x6D\x90\xEA\x81\x80\x38\xCE\x77\x72\x34\xF7\x0B\x74\x26\x77\xAE\xBE\xA3\xD4\x56\x95\x02\x21\x33\x8D\x39\xD4\x46\x56\x3D\xAB\x66\x5B\xAA\xDC\xDD\xC0\x64\x0E\x45\x5F\xF3\xF8\x61\x05\x6B\x8A\x59\x95\xDD\xAD\xEC\x89\xD4\xED\x92\xF0\x72\xCB\xA4\xF7\xFA\xC2\x45\x9A\x2F\x25\x98\xE9\xC9\x05\xB9\xFD\xF9\x00\xBF\x5D\x96\xED\x4F\xF0\xE0\x02\x10\xFA\x4E\x1F\xCE\x96\xBB\x7E\xE6\xEE\x9B\x4D\xE4\x0E\xCA\xFD\xCF\x06\x01\x02\x0D\x0A
>>> client_sendv: [Client 1] Sending payload: \x30\x0D\x0A\x0D\x0A
>>> homekit_client_process: [Client 1] Finished processing
>>> homekit_client_process: [Client 1] Got 585 incoming data
>>> homekit_server_on_pair_setup: Pair Setup
>>> homekit_server_on_pair_setup: Free heap: 201820
>>> tlv_debug: Got following TLV values:
>>> tlv_debug: Type 6 value (1 bytes): \x03
>>> tlv_debug: Type 3 value (384 bytes): \xB6\xF5\x5E\x3C\x57\xE0\x44\xEE\xC9\xD3\xBD\x78\xBD\x66\x2E\x77\x5F\x13\x5C\x80\x4E\x98\xE7\x2F\x42\x86\xE8\x61\x63\xC6\xA3\xF9\x1A\x02\xCB\x54\xA4\x9A\x79\xFD\x75\x79\xBF\x60\x7A\xF0\x50\x06\x1E\x77\x97\xCC\x7D\xBA\x26\x83\xBE\x6F\x45\x58\xD2\x8C\x60\x07\x76\xB0\xE8\x43\xB7\xF1\x53\x8A\xDD\xB3\x29\xA4\xB6\x05\xEB\x05\xEF\xF8\x79\xD6\xAE\x12\x5E\xAE\x89\x2B\x07\x31\x26\xB5\x96\xD1\xEB\x79\x32\x54\xCD\x53\xCB\x07\x22\x05\x05\x30\xF7\xE3\x37\x0E\x52\x6E\xA4\x1E\x7F\x80\x3A\x71\x87\x3A\x2C\xAC\xF5\xB6\x82\x98\xAD\x8F\x17\x3C\x63\x7C\x9A\xF0\xEA\x56\x48\xC1\x7C\x17\x85\x4A\x4D\x22\x44\x1B\x4A\x58\x1A\x39\x62\x68\x75\xE4\x0C\x89\xEA\x8A\xDF\x7A\x18\x24\xCB\x6F\xD7\x1F\xA7\x89\x32\x17\x99\xB7\x1D\x79\x34\x12\x76\x66\x41\x4D\xBB\x14\x7B\x5E\xBE\xCC\x38\xF8\x76\x71\x85\x80\x32\xF3\xA2\x92\x83\xF2\x04\x53\x74\xEE\x19\x89\xA5\xCE\xDB\x82\x34\x38\x4C\x15\x69\x31\x9D\x9A\x68\x4C\x36\x42\x [... cut ...]
>>> client_sendv: [Client 1] Sending payload: \x34\x35\x0D\x0A\x06\x01\x04\x04\x40\x74\x56\x8F\x9C\x88\xBC\x4D\xF2\x66\x46\xD2\x61\xA9\xBB\x82\xD2\x23\xC2\xF6\x80\xD2\x0B\xA6\x0C\x0C\x9B\x22\x9D\x5C\xB0\xFD\x2D\xB8\x3F\x3B\x47\xB3\xC7\xAA\x9E\xE7\xDA\xA9\xBD\xB1\x0F\xCF\x62\x59\xA7\xF5\x5A\x76\x2F\xAB\x97\xF0\x59\x35\x1B\x25\xDE\xD5\x4C\x0D\x0A
>>> client_sendv: [Client 1] Sending payload: \x30\x0D\x0A\x0D\x0A
>>> homekit_client_process: [Client 1] Finished processing
>>> HomeKit: [Client 1] Closing client connection from 192.168.178.145

```

in HomeKit app -> Unable to add Accessory

gojimmypi, 2024-02-14:
Hello @AchimPieters 

It looks like you are making excellent progress on your project! 

I've reviewed your log, above: it appears wolfSSL is compiling and working properly for you now. What was the problem & solution with your SRP initializer?

Recall I don't have any Apple products, so I am unable to test the [HomeKit app](https://developer.apple.com/design/human-interface-guidelines/homekit) with your examples. 

The message `Unable to add Accessory` is rather vague. I'm not sure how I can help you. Is there something with wolfSSL that still needs attention?

AchimPieters, 2024-02-17:
Hello @gojimmypi,

I'm totally lost, can we take it from the top again?

# Reproduction steps

Open a terminal window on your mac.

```
docker pull espressif/idf:latest
```
- At this point idf (ESP-IDF v5.3-dev-2032-g4d90eedb6e)
```
git clone --recursive https://github.com/AchimPieters/esp32-homekit-demo.git
```
```
docker run -it -v ~/esp32-homekit-demo:/project -w /project espressif/idf:latest
```
```
cd examples/led
```
```
idf.py set-target esp32
```
```
idf.py menuconfig
```
- Select `StudioPieters` and then `(mysid) WIFI SSID` and fill in your Wi-Fi Network name, then select `(mypassword) WiFI Password` and fill in your Wi-Fi Network password.
- Select `component  config` and then `homekit` and put a tick at `debug output`
- Then press `ESC` until you are asked `Save Configuration?` and select `(Y)es`
```
idf.py build
```
And then the errors appear:

```
it/CMakeFiles/__idf_homekit.dir/src/crypto.c.obj -c /project/components/homekit/src/crypto.c
/project/components/homekit/src/crypto.c:59:19: error: unknown type name 'Srp'
   59 | int wc_SrpSetKeyH(Srp *srp, byte *secret, word32 size) {
      |                   ^~~
/project/components/homekit/src/crypto.c:80:1: error: unknown type name 'Srp'
   80 | Srp *crypto_srp_new() {
      | ^~~
/project/components/homekit/src/crypto.c: In function 'crypto_srp_new':
/project/components/homekit/src/crypto.c:81:9: error: unknown type name 'Srp'
   81 |         Srp *srp = malloc(sizeof(Srp));
      |         ^~~
/project/components/homekit/src/crypto.c:81:34: error: 'Srp' undeclared (first use in this function); did you mean 'srp'?
   81 |         Srp *srp = malloc(sizeof(Srp));
      |                                  ^~~
      |                                  srp
/project/components/homekit/src/crypto.c:81:34: note: each undeclared identifier is reported only once for each function it appears in
/project/components/homekit/src/crypto.c:84:17: error: implicit declaration of function 'wc_SrpInit' [-Werror=implicit-function-declaration]
   84 |         int r = wc_SrpInit(srp, SRP_TYPE_SHA512, SRP_CLIENT_SIDE);
      |                 ^~~~~~~~~~
/project/components/homekit/src/crypto.c:84:33: error: 'SRP_TYPE_SHA512' undeclared (first use in this function); did you mean 'WC_HASH_TYPE_SHA512'?
   84 |         int r = wc_SrpInit(srp, SRP_TYPE_SHA512, SRP_CLIENT_SIDE);
      |                                 ^~~~~~~~~~~~~~~
      |                                 WC_HASH_TYPE_SHA512
/project/components/homekit/src/crypto.c:84:50: error: 'SRP_CLIENT_SIDE' undeclared (first use in this function)
   84 |         int r = wc_SrpInit(srp, SRP_TYPE_SHA512, SRP_CLIENT_SIDE);
      |                                                  ^~~~~~~~~~~~~~~
/project/components/homekit/src/crypto.c:89:12: error: request for member 'keyGenFunc_cb' in something not a structure or union
   89 |         srp->keyGenFunc_cb = wc_SrpSetKeyH;
      |            ^~
/project/components/homekit/src/crypto.c:89:30: error: 'wc_SrpSetKeyH' undeclared (first use in this function)
   89 |         srp->keyGenFunc_cb = wc_SrpSetKeyH;
      |                              ^~~~~~~~~~~~~
/project/components/homekit/src/crypto.c: At top level:
/project/components/homekit/src/crypto.c:95:22: error: unknown type name 'Srp'
[... 22 lines cut ...]
      |         ^~~
/project/components/homekit/src/crypto.c: In function 'crypto_chacha20poly1305_decrypt':
/project/components/homekit/src/crypto.c:282:29: error: 'CHACHA20_POLY1305_AEAD_AUTHTAG_SIZE' undeclared (first use in this function)
  282 |         if (message_size <= CHACHA20_POLY1305_AEAD_AUTHTAG_SIZE) {
      |                             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/project/components/homekit/src/crypto.c:298:17: error: implicit declaration of function 'wc_ChaCha20Poly1305_Decrypt'; did you mean 'crypto_chacha20poly1305_decrypt'? [-Werror=implicit-function-declaration]
  298 |         int r = wc_ChaCha20Poly1305_Decrypt(
      |                 ^~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                 crypto_chacha20poly1305_decrypt
/project/components/homekit/src/crypto.c: In function 'crypto_chacha20poly1305_encrypt':
/project/components/homekit/src/crypto.c:315:37: error: 'CHACHA20_POLY1305_AEAD_AUTHTAG_SIZE' undeclared (first use in this function)
  315 |         size_t len = message_size + CHACHA20_POLY1305_AEAD_AUTHTAG_SIZE;
      |                                     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/project/components/homekit/src/crypto.c:323:17: error: implicit declaration of function 'wc_ChaCha20Poly1305_Encrypt'; did you mean 'crypto_chacha20poly1305_encrypt'? [-Werror=implicit-function-declaration]
  323 |         int r = wc_ChaCha20Poly1305_Encrypt(
      |                 ^~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                 crypto_chacha20poly1305_encrypt
cc1: some warnings being treated as errors
[909/921] Building C object esp-idf/es...s/__idf_espressif__mdns.dir/mdns.c.obj
ninja: build stopped: subcommand failed.
ninja failed with exit code 1, output of the command is in the /project/examples/led/build/log/idf_py_stderr_output_471 and /project/examples/led/build/log/idf_py_stdout_output_471
root@6659c5f65e9d:/project/examples/led# 
```

I've tried the "old" fixes, but they resolve the problem "kindisch", maybebe we can have a look specific at the current setup en solve them step by step, to see if this solves the problems?

gojimmypi, 2024-02-20:
Hello @AchimPieters 

I have been able to replicate the compile problem you are seeing. I appreciate your persistence and patience as we work through these issues.

Try adding these to your wolfssl `user_settings.h` in `esp32-homekit-demo\components\wolfssl\include`:

```c
#define WOLFCRYPT_HAVE_SRP
#define HAVE_CHACHA
#define HAVE_POLY1305
#define WOLFSSL_BASE64_ENCODE
```

Let me know if that works for you.

AchimPieters, 2024-02-21:
@gojimmypi I added the lines and did a compile, everything worked fine until this error?

` app partition is too small for binary main.bin` is the new version of wolfssl that much bigger then the previous one?

```
FAILED: esp-idf/esptool_py/CMakeFiles/app_check_size /project/examples/led/build/esp-idf/esptool_py/CMakeFiles/app_check_size 
cd /project/examples/led/build/esp-idf/esptool_py && /opt/esp/python_env/idf5.3_py3.10_env/bin/python /opt/esp/idf/components/partition_table/check_sizes.py --offset 0x8000 partition --type app /project/examples/led/build/partition_table/partition-table.bin /project/examples/led/build/main.bin
Error: app partition is too small for binary main.bin size 0x100720:
  - Part 'factory' 0/0 @ 0x10000 size 0x100000 (overflow 0x720)
ninja: build stopped: subcommand failed.
ninja failed with exit code 1, output of the command is in the /project/examples/led/build/log/idf_py_stderr_output_471 and /project/examples/led/build/log/idf_py_stdout_output_471
root@18cf358e7bae:/project/examples/led# 
```

When changing the partition table to:

```
# ESP-IDF Partition Table
# Name,   Type, SubType, Offset,  Size,   Flags
nvs,      data, nvs,     0x9000,  0x5000,
phy_init, data, phy,     0xe000,  0x1000,
homekit,  data, homekit, 0xf000,  0x1000,
factory,  app,  factory, 0x10000, 2M,
```

I can compile without any error, but then when I want to add the accessory to HomeKit it gives this errror:

```
I (3851) IP_EVENT: WiFI ready
>>> HomeKit: Starting server
>>> HomeKit: Using existing accessory ID: 2B:BC:E2:25:EC:3F
>>> HomeKit: Configuring mDNS
>>> homekit_setup_mdns: Accessory Setup ID = 1QJ8
>>> homekit_run_server: Starting HTTP server
>>> homekit_server_accept_client: Free heap: 210860
>>> HomeKit: [Client 1] Got new client connection from 192.168.178.28
>>> homekit_client_process: [Client 1] Got 132 incoming data
>>> homekit_server_on_pair_setup: Pair Setup
>>> homekit_server_on_pair_setup: Free heap: 211248
>>> tlv_debug: Got following TLV values:
>>> tlv_debug: Type 0 value (1 bytes): \x00
>>> tlv_debug: Type 6 value (1 bytes): \x01
>>> HomeKit: [Client 1] Pair Setup Step 1/3
>>> homekit_server_on_pair_setup: Free heap: 211136
>>> crypto_srp_new: Initializing SRP
>>> homekit_server_on_pair_setup: [Client 1] Initializing crypto
>>> homekit_server_on_pair_setup: Free heap: 208304
>>> homekit_server_on_pair_setup: [Client 1] Using user-specified password: 338-77-883
>>> crypto_srp_init: Generating salt
>>> crypto_srp_init: Setting SRP username
>>> crypto_srp_init: Setting SRP params
>>> crypto_srp_init: Setting SRP password
>>> crypto_srp_init: Getting SRP verifier
>>> crypto_srp_init: Failed to get SRP verifier (code -1)
!!! HomeKit: [Client 1] Failed to initialize SRP
>>> client_sendv: [Client 1] Sending payload: HTTP/1.1 200 OK\x0D\x0AContent-Type: application/pairing+tlv8\x0D\x0ATransfer-Encoding: chunked\x0D\x0AConnection: keep-alive\x0D\x0A\x0D\x0A
>>> client_sendv: [Client 1] Sending payload: \x36\x0D\x0A\x06\x01\x02\x07\x01\x01\x0D\x0A
>>> client_sendv: [Client 1] Sending payload: \x30\x0D\x0A\x0D\x0A
>>> homekit_client_process: [Client 1] Finished processing
>>> HomeKit: [Client 1] Closing client connection from 192.168.178.28
```

so the known ` >>> crypto_srp_init: Failed to get SRP verifier (code -1)` as seen here: https://github.com/wolfSSL/wolfssl/issues/6870#issuecomment-1766407854

I could follow the steps above, but want to do this together with you, so we are and stay on the same page.

So what do you recommend doing.

gojimmypi, 2024-02-21:
Hi @AchimPieters 

>  app partition is too small for binary main.bin is the new version of wolfssl that much bigger then the previous one?

Well, yes, we turned on new features. We saw that error [last year](https://github.com/wolfSSL/wolfssl/issues/6870#issuecomment-1766693433). Curious that you ran out of memory, as I still had 2% available yesterday on my classic ESP32 and did not need to adjust the partition size.

btw - see my [defaults](https://github.com/wolfSSL/wolfssl/blob/fd2b80ec22849c1d3e318d640bd3a76f7dd491e7/IDE/Espressif/ESP-IDF/examples/template/sdkconfig.defaults#L32), in particular the [partitions_singleapp_large.csv](https://github.com/wolfSSL/wolfssl/blob/master/IDE/Espressif/ESP-IDF/examples/template/partitions_singleapp_large.csv) that may be helpful.

> when I want to add the accessory to HomeKit it gives this errror:  crypto_srp_init: Failed to get SRP verifier (code -1)

It looks like the [suggested](https://github.com/wolfSSL/wolfssl/issues/6870#issuecomment-1775565390) settings from dgarske are missing from your [user_settings.h](https://github.com/AchimPieters/esp32-homekit-demo/blob/066913c10705365b34b1819bc432d54fc3d52808/components/wolfssl/include/user_settings.h#L22):

```
If you are building with USE_FAST_MATH (tfc.c) then you need #define FP_MAX_BITS (8192 * 2).
If you are building with WOLFSSL_SP_MATH_ALL (sp_int.c) then set #define SP_INT_BITS 8192
```

Please add the `#define FP_MAX_BITS (8192 * 2)` back as before & let me know how it goes. 

I suggest [keeping](https://github.com/AchimPieters/esp32-homekit-demo/blob/066913c10705365b34b1819bc432d54fc3d52808/components/wolfssl/include/user_settings.h#L190) `USE_FAST_MATH` as that's the only wolfSSL math library supporting hardware accelerated encryption on the ESP32 at this time.

Seems you are very close! It would be great to see this all working as desired.

AchimPieters, 2024-02-22:
@gojimmypi 

Okay, that did something. Now I still get "Unable to add accessory" and in the terminal:

```
I (4262) IP_EVENT: WiFI ready
>>> HomeKit: Starting server
>>> HomeKit: Using existing accessory ID: 30:EC:BB:7A:1F:40
>>> HomeKit: Configuring mDNS
>>> homekit_setup_mdns: Accessory Setup ID = 1QJ8
>>> homekit_run_server: Starting HTTP server
>>> homekit_server_accept_client: Free heap: 210832
>>> HomeKit: [Client 1] Got new client connection from 192.168.178.28
>>> homekit_client_process: [Client 1] Got 132 incoming data
>>> homekit_server_on_pair_setup: Pair Setup
>>> homekit_server_on_pair_setup: Free heap: 211220
>>> tlv_debug: Got following TLV values:
>>> tlv_debug: Type 0 value (1 bytes): \x00
>>> tlv_debug: Type 6 value (1 bytes): \x01
>>> HomeKit: [Client 1] Pair Setup Step 1/3
>>> homekit_server_on_pair_setup: Free heap: 211112
>>> crypto_srp_new: Initializing SRP
>>> homekit_server_on_pair_setup: [Client 1] Initializing crypto
>>> homekit_server_on_pair_setup: Free heap: 202136
>>> homekit_server_on_pair_setup: [Client 1] Using user-specified password: 338-77-883
>>> crypto_srp_init: Generating salt
>>> crypto_srp_init: Setting SRP username
>>> crypto_srp_init: Setting SRP params
>>> crypto_srp_init: Setting SRP password
>>> crypto_srp_init: Getting SRP verifier
>>> crypto_srp_init: Setting SRP verifier
>>> crypto_srp_get_public_key: Calculating public key
>>> client_sendv: [Client 1] Sending payload: HTTP/1.1 200 OK\x0D\x0AContent-Type: application/pairing+tlv8\x0D\x0ATransfer-Encoding: chunked\x0D\x0AConnection: keep-alive\x0D\x0A\x0D\x0A
>>> client_sendv: [Client 1] Sending payload: \x31\x39\x39\x0D\x0A\x03\xFF\x1E\xCD\xCB\xE9\x02\x7A\x15\x06\xDB\x98\x8C\xC5\xA5\x8A\xCE\x94\xAC\xE7\x9B\xFD\x3E\xCE\x6A\x71\x2E\x09\x61\xC3\xB1\xF9\x26\x70\xEC\x3C\xCA\x81\x0F\x51\x52\x43\x36\x4C\xC2\xB8\x28\x77\x32\xFB\x4B\xBC\x8E\xB5\x26\xE8\x16\xFF\x8E\x67\xA5\x99\x37\x3D\x9D\x20\x9E\xDE\x08\x34\xF0\xB9\x20\xE8\x37\x4E\xE9\x8F\xFE\x0F\x03\xE0\x7A\x44\x4C\x0A\xA5\xFB\x77\xA4\x57\x2B\xC9\xA3\x9B\x17\x96\xFA\x94\xBA\xEA\xCD\x37\x22\x4D\x31\xE0\x80\x34\x8D\x8D\x0F\x88\x84\x38\x76\xCE\x29\xEF\x7C\x1C\x58\xD4\x02\xF5\x6C\x5B\xD5\xF6\x2B\xCF\x0F\x2C\xFC\x49\xDA\x8E\xDB\xAE\xBC\xD3\xF0\xDD\x17\x10\x6E\x3D\xA4\x45\x2C\xF5\x6B\x38\x5D\x48\x3A\x21\xF1\xCB\xEB\xA1\xE4\x05\x06\xFD\xB8\x84\x4D\x60\x9D\xFE\xC4\xEB\x61\x1D\x65\xEB\xA7\xD4\xC1\xD0\x64\x7E\xD4\xF1\x59\xB4\xCE\x9E\x34\x07\xCA\x8E\x0D\x13\x65\xE0\x4E\x73\x94\xFE\xDF\x02\xE7\x86\x88\xFB\xAB\xC4\x9F\xAB\x10\x25\x81\xC9\xDF\x19\x08\x02\xEA\x1B\x0F\xD5\xC7\xC3\x76\x2D\x57\x77\xA5\x29\xBA\x77\x95\x7F\xB0\xC0\xE7\x69\x9F\xFE\x2D\x0F\xDC\x31\x02\x7D\xFA\x95\xC1\xBC\x4C\xD7\xE3\x1C\x94\x9B\x03\x81\x20\xB2\x15\xFD\xBE\xFB\x3B\xD2\xC3\x7A\x9E\xCA\xBF\xE1\x2B\x78\x54\xDA\xB8\xB4\x7D\x51\x53\x3D\xB6\xF2\x98\x19\xBB\xAD\x48\xDB\xE4\x84\xF8\xAA\x38\xDF\x1A\x18\x03\x95\x77\x2B\xD2\x62\xAC\xA0\x6E\x84\xB9\x5E\xE4\xD7\x0B\x38\x4F\x81\x74\xCF\x6F\x11\x91\x9A\xA8\xD8\x08\xAF\x83\x42\xC8\x74\x9A\x77\x82\x98\x18\xFF\x01\x2A\x0F\x17\xCC\xAC\x29\xD4\xF7\xF5\xC2\x7E\x77\x3F\x98\x52\x19\x9F\x89\xFC\x9D\xE1\xA3\x6F\x67\xEE\x6D\x69\x07\x40\xDA\x85\x7A\xCC\x5D\x76\x85\x54\x9C\xCF\x1D\xEC\x63\x3D\x37\xFC\xCF\xC2\xD3\x4E\x10\x02\x10\x7E\x21\xAF\xFD\xDA\x5A\x55\xE5\x8C\x71\x85\x6D\x89\x18\x30\xFF\x06\x01\x02\x0D\x0A
>>> client_sendv: [Client 1] Sending payload: \x30\x0D\x0A\x0D\x0A
>>> homekit_client_process: [Client 1] Finished processing
>>> homekit_client_process: [Client 1] Got 585 incoming data
>>> homekit_server_on_pair_setup: Pair Setup
>>> homekit_server_on_pair_setup: Free heap: 201500
>>> tlv_debug: Got following TLV values:
>>> tlv_debug: Type 6 value (1 bytes): \x03
>>> tlv_debug: Type 3 value (384 bytes): \xC1\x73\xA4\xDF\xCF\x66\xFD\xC3\x74\x21\x5C\x05\xB2\xB7\x47\x30\x9E\xAB\xCA\xA9\x88\x50\x3B\x20\x29\xE9\xCB\x9D\x38\xDD\xA5\x94\x9C\xE3\xE3\x7E\xC5\x28\xD5\x31\xC8\x3E\x8A\x0A\x6D\x8A\xE2\x0B\x1C\x3F\xF7\xEC\x6D\x45\xAD\xDD\x7B\xFF\x7A\xA1\x4A\xA3\x81\xBA\x9B\x7A\xC0\x0C\xF7\x50\x32\xA1\x52\xEE\xF0\x74\x27\x04\x3C\x58\x1B\xD2\xE9\x21\xF4\x64\xDA\x4A\x0A\xCC\x13\x20\xB2\x93\x57\x1B\x5C\xF2\x8B\x94\xDF\x37\x86\x3A\x46\x18\x3F\x47\x05\x98\xA5\x7C\xE3\x41\x33\x21\x1F\xD7\x8E\x4E\xD2\x8C\x55\x20\x4B\x1B\x9F\x29\x8C\x06\xBA\xCC\x76\x82\x44\x94\x53\xE2\xCF\xB1\x88\x1E\x82\x6A\xC9\x96\x5B\xF5\x59\x24\x8E\x6A\x9E\x67\x5A\x2F\xA2\x87\xF4\xE3\x31\x20\xA7\x2D\xF3\x29\x74\xBF\x91\xC0\xFF\x4B\x3C\x51\x13\x27\x25\xE0\xF6\x34\xCC\x5F\x88\xED\x36\x7E\xAA\xB8\x1F\x39\xE9\x21\x92\xED\x03\x49\xCE\xFD\xBA\x38\xD8\x38\x6B\x83\x7E\x5C\xB0\xD3\xAC\x79\x14\x24\xC4\xD3\xCE\x9F\xA0\xC0\x07\x7E\x53\xF3\xD8\xA0\x45\xC7\x8E\xC0\xE2\x71\x84\x04\x25\x39\x17\x40\xD4\xCA\xDF\x85\xD7\x33\x6A\xA3\x4F\xF3\x34\x72\x67\xCA\x2C\x78\x72\xFD\x09\xBB\x79\x4C\x99\x45\x69\xAC\x7E\xAC\xBD\xF2\x5E\x02\x02\x31\x2E\x89\x80\x84\x14\xE2\x02\x47\x41\x93\xE2\xB1\x38\x79\xFB\xE7\x97\x66\xEE\xD2\x68\xDA\x06\x46\x68\x39\x50\xD3\xBA\x7D\x1D\xB5\x87\xF1\x18\xCF\xBA\x76\x13\xAB\x51\xD4\x1D\x5F\x96\x07\x53\x23\xF8\x94\x3E\x0D\xEC\x1D\x20\xD0\x7D\x00\xF4\x8A\x0E\x7D\xB2\x92\x3E\x0E\x0C\xAF\xD9\xE8\x16\x37\xA4\x9B\x1A\x9E\xD7\x54\x18\x20\x38\x97\x30\x40\xAD\x36\xAC\x93\x8F\x41\x00\x1A\x3A\xA1\xAB\xB2\xFD\xCC\xED\xFF\x7A\x5F\xDF\x7C\x97\x98\x89\x46\x6F\x4D\xD3\x6F\x6E\xAE
>>> tlv_debug: Type 4 value (64 bytes): \x29\xD0\x37\xD9\xCE\x1C\x35\x4B\x00\x61\xC1\x75\xD8\xF0\xF6\x4B\x90\x7D\xB0\x32\x35\x77\x7E\xF0\xCF\x56\xBD\x0C\xD4\x2E\xA3\x96\x53\xD9\x87\xBB\x25\x52\xC3\x4F\xEC\x6A\x5C\x7F\x40\x46\x7A\x7E\x6B\xC3\x9E\x0C\x9D\x3B\x [... cut ...]
>>> client_sendv: [Client 1] Sending payload: \x34\x35\x0D\x0A\x06\x01\x04\x04\x40\x6A\x2C\x00\x97\x57\xF4\x26\x0B\xD5\x58\x1D\x79\xE0\x3B\x5F\x23\xF4\x22\xB5\x85\xEF\xC2\xB3\x72\x52\x16\x8A\x70\x3E\xB8\x41\xC1\x3E\x90\x67\xF9\x2F\x36\x40\x02\x73\x7E\x85\x85\xD6\xEB\x9B\x3A\xB2\xFB\x29\xA6\x18\x96\xA7\x4B\xE2\x37\xAE\x97\x21\x4B\x89\x8B\x0D\x0A
>>> client_sendv: [Client 1] Sending payload: \x30\x0D\x0A\x0D\x0A
>>> homekit_client_process: [Client 1] Finished processing
>>> HomeKit: [Client 1] Closing client connection from 192.168.178.28
```

The SRP works kindich...

gojimmypi, 2024-02-22:
Hi @AchimPieters 

It seems you are really close! Excellent. What's different from when you last had this [working](https://github.com/wolfSSL/wolfssl/issues/6870#issuecomment-1777632388)?

It's getting a bit more difficult for me to help, given that I don't have any of the Homekit hardware.

I do have some suggestions though:

- Try to disable hardware acceleration.

Recently while working on the [SSH to UART example](https://github.com/wolfSSL/wolfssh-examples/tree/main/Espressif/ESP32/ESP32-SSH-Server), I discovered a problem with SHA256 hardware acceleration as noted in https://github.com/wolfSSL/wolfssh-examples/pull/4. I'm working on a solution to both fix the code, and capture in [wolfCrypy test](https://github.com/wolfSSL/wolfssl/tree/master/IDE/Espressif/ESP-IDF/examples/wolfssl_test). The root cause was a hash interleave that I did not properly take into account when in `SINGLE_THREADED` mode when calculating an SSH signature. There could well be a similar problem in the SRP. 

The brute-force method to start with is to disable _all_ [hardware acceleration in the user_settings.h](https://github.com/wolfSSL/wolfssl/blob/8a68e6aaa1f2fc72e5145e469107929c9f03266c/IDE/Espressif/ESP-IDF/examples/template/components/wolfssl/include/user_settings.h#L220):

```
      #define NO_ESP32_CRYPT
      #define NO_WOLFSSL_ESP32_CRYPT_HASH
      #define NO_WOLFSSL_ESP32_CRYPT_AES
      #define NO_WOLFSSL_ESP32_CRYPT_RSA_PRI
      #define NO_WOLFSSL_ESP32_CRYPT_RSA_PRI_MP_MUL 
      #define NO_WOLFSSL_ESP32_CRYPT_RSA_PRI_MULMOD 
      #define NO_WOLFSSL_ESP32_CRYPT_RSA_PRI_EXPTMOD
```

- On the topic of tests: try using the [wolfCrypt test](https://github.com/wolfSSL/wolfssl/tree/master/IDE/Espressif/ESP-IDF/examples/wolfssl_test) app in your environment with your respective `user_settings.h`.

- Try turning on `DEBUG_WOLFSSL` (see [examples](https://github.com/wolfSSL/wolfssl/blob/8a68e6aaa1f2fc72e5145e469107929c9f03266c/IDE/Espressif/ESP-IDF/examples/wolfssl_test/components/wolfssl/include/user_settings.h#L355) for other debug options)

- Try sprinkling in some `printf` (or better: `ESP_LOGI(TAG, "message")`) breadcrumbs to see where it is failing.

- Try using a JTAG debugger with breakpoints & single-step to find exactly where things go sideways. I use the [VisualGDB](https://visualgdb.com/) extension for Visual Studio with the open-source [Tigard JTAG debugger](https://github.com/tigard-tools/tigard?tab=readme-ov-file#tigard). I've found this to be a very helpful and productive tool.

Let me know how that goes. I'd really like to see the your Homekit examples working again.

Best Regards

AchimPieters, 2024-02-23:
Tested:

ESP32 WROOM 32D - SUCCES!
ESP32C2 (ESP8684) - Still looking in too...
ESPC3 (ESP-C3-12F Kit):

```
I (3643) IP_EVENT: WiFI ready
>>> HomeKit: Starting server
>>> HomeKit: Using existing accessory ID: C5:D6:C1:32:8A:B4
>>> HomeKit: Configuring mDNS
>>> homekit_setup_mdns: Accessory Setup ID = 1QJ8
>>> homekit_run_server: Starting HTTP server
>>> homekit_server_accept_client: Free heap: 209740
>>> HomeKit: [Client 1] Got new client connection from 192.168.178.28
>>> homekit_client_process: [Client 1] Got 132 incoming data
>>> homekit_server_on_pair_setup: Pair Setup
>>> homekit_server_on_pair_setup: Free heap: 211476
>>> tlv_debug: Got following TLV values:
>>> tlv_debug: Type 0 value (1 bytes): \x00
>>> tlv_debug: Type 6 value (1 bytes): \x01
>>> HomeKit: [Client 1] Pair Setup Step 1/3
>>> homekit_server_on_pair_setup: Free heap: 211376
>>> crypto_srp_new: Initializing SRP
>>> homekit_server_on_pair_setup: [Client 1] Initializing crypto
>>> homekit_server_on_pair_setup: Free heap: 208564
>>> homekit_server_on_pair_setup: [Client 1] Using user-specified password: 338-77-883
>>> crypto_srp_init: Generating salt
>>> crypto_srp_init: Setting SRP username
>>> crypto_srp_init: Setting SRP params
>>> crypto_srp_init: Setting SRP password
>>> crypto_srp_init: Getting SRP verifier
>>> crypto_srp_init: Failed to get SRP verifier (code -1)
!!! HomeKit: [Client 1] Failed to initialize SRP
>>> client_sendv: [Client 1] Sending payload: HTTP/1.1 200 OK\x0D\x0AContent-Type: application/pairing+tlv8\x0D\x0ATransfer-Encoding: chunked\x0D\x0AConnection: keep-alive\x0D\x0A\x0D\x0A
>>> client_sendv: [Client 1] Sending payload: \x36\x0D\x0A\x06\x01\x02\x07\x01\x01\x0D\x0A
>>> client_sendv: [Client 1] Sending payload: \x30\x0D\x0A\x0D\x0A
>>> homekit_client_process: [Client 1] Finished processing
>>> HomeKit: [Client 1] Closing client connection from 192.168.178.28
```

The SRP error is back `Failed to get SRP verifier (code -1)` ?

gojimmypi, 2024-02-23:
Hi @AchimPieters  - That's awesome you have it working! 

Regarding your ESP-C3-12F Kit: The default hardware acceleration disabling is SoC-specific. Can you please confirm you've disabled it for the ESP32-C3 in your [user_settings.h](https://github.com/AchimPieters/esp32-homekit-demo/blob/69a6a806c4121d76769fb5eff584eec8c98e09a2/components/wolfssl/include/user_settings.h#L351) for the `CONFIG_IDF_TARGET_ESP32C3`?  

You may also consider moving the definitions completely outside of the [conditional SoC check](https://github.com/AchimPieters/esp32-homekit-demo/blob/69a6a806c4121d76769fb5eff584eec8c98e09a2/components/wolfssl/include/user_settings.h#L270) to disable all hardware acceleration for all devices.

I'll need to look into why the hardware acceleration passes all tests but apparently causes the SRP to fail. 

Well done on making such great progress! Cheers

AchimPieters, 2024-02-24:
@gojimmypi, You are awesome, couldn't do it without you! Thank you so much!

Regarding the User_settings.h here is my latest files: https://github.com/AchimPieters/esp32-homekit-demo.git

I'll keep working en testing this weekend, and will share my results!

AchimPieters, 2024-02-25:
@gojimmypi I have checked, double-checked, [Krajiceked](https://nl.wikipedia.org/wiki/Richard_Krajicek)...

But the `SRP verifier (code -1)` keeps a mystery. 

I also have ordered some ESP32XX Devkits for better debugging capabilities...

gojimmypi, 2024-02-25:
Hello @AchimPieters  -

Can you please confirm the SRP is fully operational with the hardware acceleration confirmed as _disabled_ for the respective target ESP32 SoC flavor?  (I'm not sure I understand the Richard Krajicek tennis link: essentially "_quadruple checked_"?).  Your prior comment seemed to indicate success, but the most recent comment indicates otherwise.

Unless running out of heap or stack, or something wrong with the calling application... I would expect the wolfSSL SRP to be 100% operational in software only mode. (and in theory with hardware acceleration, if everything there is working properly).

I'm assuming this is a hardware acceleration problem, similar to the issue I mentioned with the wolfSSH signature and the interleaved hash calcs not being handled properly. I recently updated the [wolfCrypt tests](https://github.com/wolfSSL/wolfssl/tree/master/wolfcrypt/test) for checking SHA interleaving in [#7262](https://github.com/wolfSSL/wolfssl/pull/7262). 

I believe I've resolved SHA interleave problem with another PR coming soon. It could well be the SRP has a similar or same problem.

I'll be giving this a high priority to resolve. I need to be 100% certain of the results your are seeing, particularly since I don't have the hardware to reproduce myself.

If you are ordering new ESP32 boards: one thing I should have mentioned that in addition to the Tigard _external_ JTAG, the newer Espressif boards with 2 USB ports typically also have JTAG built-in to the board. It is of course not as flexible as having a stand-alone JTAG, but at least that specific board can use the JTAG debugger quite easily.

Thanks for your help. Cheers

AchimPieters, 2024-02-26:
@gojimmypi the SRP is fully operational with the hardware acceleration confirmed as disabled for the respective target ESP32 SoC flavor, But Not for the ESP32C3 and other models.

The Krajicek, was maybe a bad word joke, sorry but that.... 

ETA for the new ESP boards with double USB ports for better debugging is 7 days, once I receive them, I can do a model by model test. Thank You for your continuous support, you're the best!

AchimPieters, 2024-03-05:
@gojimmypi, I received the modules ESP32, ESP32C2, ESP32C3 and ESP32S3.

As said the ESP32 works, but the other models got SRP errors, I attach a log from every model.
[ESP32S2.log](https://github.com/wolfSSL/wolfssl/files/14499163/ESP32S2.log)
[ESP32C3.log](https://github.com/wolfSSL/wolfssl/files/14499164/ESP32C3.log)
[ESP32C2.log](https://github.com/wolfSSL/wolfssl/files/14499165/ESP32C2.log)
[ESP32.log](https://github.com/wolfSSL/wolfssl/files/14499166/ESP32.log)

gojimmypi, 2024-03-05:
@AchimPieters Thank you for the additional details and congratulations on your new purchases of other ESP32 devices!

As you said the ESP32 works, but we disabled the hardware acceleration there, right? It appears that your [user_settings.h](https://github.com/AchimPieters/esp32-homekit-demo/blob/9ab98078a51d5fd91070a7bc3e5d103ea350bf32/components/wolfssl/include/user_settings.h#L349) also has hardware disabled for the ESP32-C3. Are you saying that even in software-only mode, you are still seeing SRP errors?

If indeed your are seeing software-only problems, I suggest turning on the stack smashing and heap corruption detection. Perhaps there's a low memory problem. The SRP is fairly robust and unlikely to have problems with hardware acceleration turned off.

I'm actively working on a solution to this & appreciate any additional information.

AchimPieters, 2024-03-06:
@gojimmypi 

_Are you saying that even in software-only mode, you are still seeing SRP errors?_
Yes, as we discussed above, I changed the [user_settings.h](https://github.com/AchimPieters/esp32-homekit-demo/blob/9ab98078a51d5fd91070a7bc3e5d103ea350bf32/components/wolfssl/include/user_settings.h#L349) and there are still SRP errors.

So I'm going to try to follow your advice and turn on the stack smashing and heap corruption detection, and keep you updated.

gojimmypi, 2024-03-06:
Hi @AchimPieters 

Thanks for the confirmation. I have a fix in the works for the SRP issue. It's my active development branch, but [this commit](https://github.com/gojimmypi/wolfssl/tree/2a1f26389e178426e89cd77dd82bec1939096574) should be relatively good if you'd like to take it for a test drive.

In particular, I have this updated [wolfssl_test](https://github.com/gojimmypi/wolfssl/tree/2a1f26389e178426e89cd77dd82bec1939096574/IDE/Espressif/ESP-IDF/examples/wolfssl_test) app that specifically includes a test for the SRP enabled. (and many other features enabled, for the most exhaustive ESP32 test ever). fwiw - I never saw an SRP failure in software-only mode. 

It is of course not the same as your actual Homekit app: Perhaps an updated test is in order if you've found something new. 

I hope to have a PR together in the relatively near future.

AchimPieters, 2024-03-21:
@gojimmypi Okay I have given it a try, but got this error:

```
[858/926] Building C object esp-idf/wo...df_wolfssl.dir/wolfcrypt/src/tfm.c.obj
FAILED: esp-idf/wolfssl/CMakeFiles/__idf_wolfssl.dir/wolfcrypt/src/tfm.c.obj 
ccache /opt/esp/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/xtensa-esp32-elf-gcc -DESP_PLATFORM -DIDF_VER=\"v5.3-dev-2032-g4d90eedb6e\" -DLIBWOLFSSL_VERSION_GIT_BRANCH=\"main\" -DLIBWOLFSSL_VERSION_GIT_HASH=\"1844b49d56e923e612beb14ae11174f0e59f9fc4\" -DLIBWOLFSSL_VERSION_GIT_HASH_DATE="\"'Sun Mar 10 16:22:33 2024 +0100'\"" -DLIBWOLFSSL_VERSION_GIT_ORIGIN=\"https://github.com/AchimPieters/esp32-homekit-demo.git\" -DLIBWOLFSSL_VERSION_GIT_SHORT_HASH=\"1844b49\" -DLIBWOLFSSL_VERSION_WOLFSSL_ROOT=\"/project/components/wolfssl\" -DSOC_MMU_PAGE_SIZE=CONFIG_MMU_PAGE_SIZE -DSOC_XTAL_FREQ_MHZ=CONFIG_XTAL_FREQ -DWOLFSSL_USER_SETTINGS_DIR=\"//project//examples//led//components//wolfssl//include//user_settings.h\" -D_GLIBCXX_HAVE_POSIX_SEMAPHORE -D_GLIBCXX_USE_POSIX_SEMAPHORE -D_GNU_SOURCE -D_POSIX_READER_WRITER_LOCKS -I/project/examples/led/build/config -I/project/components/wolfssl/include -I/project/components/wolfssl -I/project/components/wolfssl/wolfssl -I/project/components/wolfssl/wolfssl/wolfcrypt -I/project/components/wolfssl/wolfssl/wolfcrypt/port/Espressif -I/opt/esp/idf/components/freertos/FreeRTOS-Kernel/include/freertos -I/opt/esp/idf/components/newlib/platform_include -I/opt/esp/idf/components/freertos/config/include -I/opt/esp/idf/components/freertos/config/include/freertos -I/opt/esp/idf/components/freertos/config/xtensa/include -I/opt/esp/idf/components/freertos/FreeRTOS-Kernel/include -I/opt/esp/idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/include -I/opt/esp/idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/include/freertos -I/opt/esp/idf/components/freertos/esp_additions/include -I/opt/esp/idf/components/esp_hw_support/include -I/opt/esp/idf/components/esp_hw_support/include/soc -I/opt/esp/idf/components/esp_hw_support/include/soc/esp32 -I/opt/esp/idf/components/esp_hw_support/dma/include -I/opt/esp/idf/components/esp_hw_support/port/esp32/. -I/opt/esp/idf/components/esp_hw_support/port/esp32/include -I/opt/esp/idf/components/heap/include -I/opt/esp/idf/components/log/include -I/opt/esp/idf/components/soc/include -I/opt/esp/idf/components/soc/esp32 -I/opt/esp/idf/components/soc/esp32/include -I/opt/esp/idf/components/hal/platform_port/include -I/opt/esp/idf/components/hal/esp32/include -I/opt/esp/idf/components/hal/include -I/opt/esp/idf/components/esp_rom/include -I/opt/esp/idf/components/esp_rom/include/esp32 -I/opt/esp/idf/components/esp_rom/esp32 -I/opt/esp/idf/components/esp_common/include -I/opt/esp/idf/components/esp_system/include -I/opt/esp/idf/components/esp_system/port/soc -I/opt/esp/idf/components/esp_system/port/include/private -I/opt/esp/idf/components/xtensa/esp32/include -I/opt/esp/idf/components/xtensa/include -I/opt/esp/idf/components/xtensa/deprecated_include -I/opt/esp/idf/components/lwip/include -I/opt/esp/idf/components/lwip/include/apps -I/opt/esp/idf/components/lwip/include/apps/sntp -I/opt/esp/idf/components/lwip/lwip/src/include -I/opt/esp/idf/components/lwip/port/include -I/opt/esp/idf/components/lwip/port/freertos/include -I/opt/esp/idf/components/lwip/port/esp32xx/include -I/opt/esp/idf/components/lwip/port/esp32xx/include/arch -I/opt/esp/idf/components/lwip/port/esp32xx/include/sys -mlongcalls -Wno-frame-address  -DWOLFSSL_USER_SETTINGS -DWOLFSSL_CMAKE_SYSTEM_NAME_LINUX -fdiagnostics-color=always -ffunction-sections -fdata-sections -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-variable -Wno-error=unused-but-set-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -Wno-enum-conversion -gdwarf-4 -ggdb -Og -fno-shrink-wrap -fmacro-prefix-map=/project/examples/led=. -fmacro-prefix-map=/opt/esp/idf=/IDF -fstrict-volatile-bitfields -fno-jump-tables -fno-tree-switch-conversion -std=gnu17 -Wno-old-style-declaration -MD -MT esp-idf/wolfssl/CMakeFiles/__idf_wolfssl.dir/wolfcrypt/src/tfm.c.obj -MF esp-idf/wolfssl/CMakeFiles/__idf_wolfssl.dir/wolfcrypt/src/tfm.c.obj.d -o esp-idf/wolfssl/CMakeFiles/__idf_wolfssl.dir/wolfcrypt/src/tfm.c.obj -c /project/components/wolfssl/wolfcrypt/src/tfm.c
/project/components/wolfssl/wolfcrypt/src/tfm.c: In function '_fp_exptmod_nct':
/project/components/wolfssl/wolfcrypt/src/tfm.c:2467:6: error: implicit declaration of function 'WOLFSSL_MSG_EX' [-Werror=implicit-function-declaration]
 2467 |      WOLFSSL_MSG_EX("_fp_exptmod_nct XMALLOC failed: %d", (word32)sizeof(fp_int)*((1 << winsize) + 1));
      |      ^~~~~~~~~~~~~~
/project/components/wolfssl/wolfcrypt/src/tfm.c: In function 'fp_exptmod_nct':
/project/components/wolfssl/wolfcrypt/src/tfm.c:3313:7: error: implicit declaration of function 'WOLFSSL_MSG'; did you mean 'WOLFSSL_API'? [-Werror=implicit-function-declaration]
 3313 |       WOLFSSL_MSG("handle modulus of zero");
      |       ^~~~~~~~~~~
      |       WOLFSSL_API
cc1: some warnings being treated as errors
[863/926] Building C object esp-idf/wo...df_wolfssl.dir/wolfcrypt/src/srp.c.obj
ninja: build stopped: subcommand failed.
ninja failed with exit code 1, output of the command is in the /project/examples/led/build/log/idf_py_stderr_output_713 and /project/examples/led/build/log/idf_py_stdout_output_713
root@5afeccbad994:/project/examples/led# 
```

I don't know whether I did something wrong or something wrong with your branch?

gojimmypi, 2024-03-21:
Hi @AchimPieters 

My branch is under active development so at some point, there might have been inoperable code at some point.

When `WOLFSSL_MSG` and/or `WOLFSSL_MSG_EX` is missing, simply include the wolfSSL logging header:

```
#include <wolfssl/wolfcrypt/logging.h>
```

The [wolfcrypt TFM](https://github.com/wolfSSL/wolfssl/blob/master/wolfcrypt/src/tfm.c) library does not include this by default. I must have been testing something. Feel free to add the include or simply comment out those lines. They are only interesting with `#define DEBUG_WOLFSSL` in the `user_settings.h` and turned on with `wolfSSL_Debugging_ON();`

My [latest branch update](https://github.com/gojimmypi/wolfssl/tree/ED25519_SHA2_fix) is working well with SRP and includes an improved logging messages when heap allocation fails. It would be great for you to take it for a test drive.

AchimPieters, 2024-03-22:
@gojimmypi I've updated wolfssl to the latest branch update, but got the same error. Then I added `#include <wolfssl/wolfcrypt/logging.h>` but then I got this:

```
ir/src/internal.c.obj -c /project/components/wolfssl/src/internal.c
In file included from /project/components/wolfssl/wolfssl/wolfcrypt/logging.h:3,
                 from /project/components/wolfssl/include/user_settings.h:23,
                 from /project/components/wolfssl/wolfssl/wolfcrypt/settings.h:301,
                 from /project/components/wolfssl/src/internal.c:28:
/project/components/wolfssl/wolfssl/wolfcrypt/types.h:1293:46: error: missing binary operator before token "("
 1293 |     #if defined(HAVE_FIPS) && FIPS_VERSION_LT(5,3)
      |                                              ^
/project/components/wolfssl/wolfssl/wolfcrypt/types.h:1648:24: error: missing binary operator before token "("
 1648 |     #if FIPS_VERSION_GE(5,1)
      |                        ^
/project/components/wolfssl/wolfssl/wolfcrypt/settings.h:1006: warning: "XMALLOC" redefined
 1006 |         #define XMALLOC(s, h, type)  ((void)(h), (void)(type), pvPortMalloc((s)))
      | 
/project/components/wolfssl/wolfssl/wolfcrypt/types.h:575: note: this is the location of the previous definition
  575 |                 #define XMALLOC(s, h, t)     ((void)(h), (void)(t), wolfSSL_Malloc((s)))
      | 
/project/components/wolfssl/wolfssl/wolfcrypt/settings.h:1007: warning: "XFREE" redefined
 1007 |         #define XFREE(p, h, type)    ((void)(h), (void)(type), vPortFree((p)))
      | 
/project/components/wolfssl/wolfssl/wolfcrypt/types.h:579: note: this is the location of the previous definition
  579 |                     #define XFREE(p, h, t)       {void* xp = (p); (void)(h); (void)(t); if (xp) wolfSSL_Free(xp);}
      | 
/project/components/wolfssl/wolfssl/wolfcrypt/settings.h:1012: warning: "XREALLOC" redefined
 1012 |                 #define XREALLOC(p, n, h, t) ((void)(h), (void)(t), realloc((p), (n)))
      | 
/project/components/wolfssl/wolfssl/wolfcrypt/types.h:581: note: this is the location of the previous definition
  581 |                 #define XREALLOC(p, n, h, t) ((void)(h), (void)(t), wolfSSL_Realloc((p), (n)))
      | 
ninja: build stopped: subcommand failed.
ninja failed with exit code 1, output of the command is in the /project/examples/led/build/log/idf_py_stderr_output_7284 and /project/examples/led/build/log/idf_py_stdout_output_7284
root@9eb0400574ee:/project/examples/led# 
```

gojimmypi, 2024-03-22:
Hi @AchimPieters - I'll take a look at this today. Perhaps try not using the include and only commenting out the `WOLFSSL_MSG` messages.

AchimPieters, 2024-03-31:
@gojimmypi upgraded today to 5.7.0 and got new errors:

```
In file included from /project/components/homekit/src/server.c:35:
/project/components/homekit/src/crypto.h:60:25: error: unknown type name 'ed25519_key'
   60 | int crypto_ed25519_init(ed25519_key *key);
      |                         ^~~~~~~~~~~
/project/components/homekit/src/crypto.h:61:1: error: unknown type name 'ed25519_key'
   61 | ed25519_key *crypto_ed25519_new();
      | ^~~~~~~~~~~
/project/components/homekit/src/crypto.h:62:29: error: unknown type name 'ed25519_key'
   62 | int crypto_ed25519_generate(ed25519_key *key);
      |                             ^~~~~~~~~~~
/project/components/homekit/src/crypto.h:63:26: error: unknown type name 'ed25519_key'
   63 | void crypto_ed25519_free(ed25519_key *key);
      |                          ^~~~~~~~~~~
/project/components/homekit/src/crypto.h:66:9: error: unknown type name 'ed25519_key'
   66 |         ed25519_key *key,
      |         ^~~~~~~~~~~
/project/components/homekit/src/crypto.h:70:15: error: unknown type name 'ed25519_key'
   70 |         const ed25519_key *key,
      |               ^~~~~~~~~~~
/project/components/homekit/src/crypto.h:75:9: error: unknown type name 'ed25519_key'
   75 |         ed25519_key *key,
      |         ^~~~~~~~~~~
/project/components/homekit/src/crypto.h:79:15: error: unknown type name 'ed25519_key'
   79 |         const ed25519_key *key,
      |               ^~~~~~~~~~~
/project/components/homekit/src/crypto.h:84:15: error: unknown type name 'ed25519_key'
   84 |         const ed25519_key *key,
      |               ^~~~~~~~~~~
/project/components/homekit/src/crypto.h:89:15: error: unknown type name 'ed25519_key'
   89 |         const ed25519_key *key,
      |               ^~~~~~~~~~~
/project/components/homekit/src/crypto.h:96:28: error: unknown type name 'curve25519_key'
   96 | int crypto_curve25519_init(curve25519_key *key);
      |                            ^~~~~~~~~~~~~~
/project/components/homekit/src/crypto.h:97:28: error: unknown type name 'curve25519_key'
   97 | int crypto_curve25519_done(curve25519_key *key);
      |                            ^~~~~~~~~~~~~~
/project/components/homekit/src/crypto.h:98:32: error: unknown type name 'curve25519_key'
   98 | int crypto_curve25519_generate(curve25519_key *key);
      |                                ^~~~~~~~~~~~~~
/project/components/homekit/src/crypto.h:100:9: error: unknown type name 'curve25519_key'
  100 |         curve25519_key *key,
      |         ^~~~~~~~~~~~~~
/project/components/homekit/src/crypto.h:104:15: error: unknown type name 'curve25519_key'
  104 |         const curve25519_key *key,
      |               ^~~~~~~~~~~~~~
/project/components/homekit/src/crypto.h:108:15: error: unknown type name 'curve25519_key'
  108 |         const curve25519_key *private_key,
      |               ^~~~~~~~~~~~~~
/project/components/homekit/src/crypto.h:109:15: error: unknown type name 'curve25519_key'
  109 |         const curve25519_key *public_key,
      |               ^~~~~~~~~~~~~~
In file included from /project/components/homekit/src/server.c:36:
/project/components/homekit/src/pairing.h:14:9: error: unknown type name 'ed25519_key'
   14 |         ed25519_key device_key;
      |         ^~~~~~~~~~~
In file included from /project/components/homekit/src/server.c:37:
/project/components/homekit/src/storage.h:13:46: error: unknown type name 'ed25519_key'
   13 | int homekit_storage_save_accessory_key(const ed25519_key *key);
      |                                              ^~~~~~~~~~~
/project/components/homekit/src/storage.h:14:40: error: unknown type name 'ed25519_key'
   14 | int homekit_storage_load_accessory_key(ed25519_key *key);
      |                                        ^~~~~~~~~~~
/project/components/homekit/src/storage.h:17:62: error: unknown type name 'ed25519_key'
   17 | int homekit_storage_add_pairing(const char *device_id, const ed25519_ke *device_key, byte permissions);
      |                                                              ^~~~~~~~~~
/project/components/homekit/src/server.c:127:9: error: unknown type name 'ed25519_key'
[... 48 lines cut ...]
 4064 |                         wc_Sha512Hash((const unsigned char *)data, data_size-1, shaHash);
      |                         ^~~~~~~~~~~~~
      |                         wc_Sha384Hash
/project/components/homekit/src/server.c:4063:39: warning: unused variable 'shaHash' [-Wunused-variable]
 4063 |                         unsigned char shaHash[SHA512_DIGEST_SIZE];
      |                                       ^~~~~~~
/project/components/homekit/src/server.c: At top level:
/project/components/homekit/src/server.c:4092:36: error: unknown type name 'ed25519_key'
 4092 | int homekit_accessory_key_generate(ed25519_key *key) {
      |                                    ^~~~~~~~~~~
/project/components/homekit/src/server.c: In function 'homekit_server_task':
/project/components/homekit/src/server.c:4114:29: error: implicit declaration of function 'homekit_storage_load_accessory_key'; did you mean 'homekit_storage_load_accessory_id'? [-Werror=implicit-function-declaration]
 4114 |                         r = homekit_storage_load_accessory_key(&server->accessory_key);
      |                             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                             homekit_storage_load_accessory_id
[... cut ...]
      |                 ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                 homekit_accessory_id_generate
cc1: some warnings being treated as errors
[915/923] Building C object esp-idf/es...s/__idf_espressif__mdns.dir/mdns.c.obj
ninja: build stopped: subcommand failed.
ninja failed with exit code 1, output of the command is in the /project/examples/led/build/log/idf_py_stderr_output_701 and /project/examples/led/build/log/idf_py_stdout_output_701
root@6717049ab60a:/project/examples/led# 
```

AchimPieters, 2024-03-31:
@gojimmypi changed some settings:

```
/* user_settings.h
 *
 * Copyright (C) 2006-2023 wolfSSL Inc.
 *
 * This file is part of wolfSSL.
 *
 * wolfSSL is free software; you can redistribute it and/or modify
 * it under the terms of the GNU General Public License as published by
 * the Free Software Foundation; either version 2 of the License, or
 * (at your option) any later version.
 *
 * wolfSSL is distributed in the hope that it will be useful,
 * but WITHOUT ANY WARRANTY; without even the implied warranty of
 * MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
 * GNU General Public License for more details.
 *
 * You should have received a copy of the GNU General Public License
 * along with this program; if not, write to the Free Software
 * Foundation, Inc., 51 Franklin Street, Fifth Floor, Boston, MA 02110-1335, USA
 */

#include <sdkconfig.h> /* essential to chip set detection */

#undef WOLFSSL_ESPIDF
#undef WOLFSSL_ESP32
#undef WOLFSSL_ESPWROOM32SE
#undef WOLFSSL_ESP32
#undef WOLFSSL_ESP8266

#define WOLFSSL_ESPIDF

#define WOLFCRYPT_HAVE_SRP
#define HAVE_CHACHA
#define HAVE_POLY1305
#define WOLFSSL_BASE64_ENCODE

/* The Espressif sdkconfig will have chipset info.
**
** Possible values:
**
**   CONFIG_IDF_TARGET_ESP32
**   CONFIG_IDF_TARGET_ESP32S2
**   CONFIG_IDF_TARGET_ESP32S3
**   CONFIG_IDF_TARGET_ESP32C3
**   CONFIG_IDF_TARGET_ESP32C6
*/

/* Optionally enable some wolfSSH settings */
#ifdef ESP_ENABLE_WOLFSSH
    /* The default SSH Windows size is massive for an embedded target. Limit it: */
    #define DEFAULT_WINDOW_SZ 2000

    /* These may be defined in cmake for other examples: */
    #undef  WOLFSSH_TERM
    #define WOLFSSH_TERM

    #undef  DEBUG_WOLFSSH
    #define DEBUG_WOLFSSH

    #undef  WOLFSSL_KEY_GEN
    #define WOLFSSL_KEY_GEN

    #undef  WOLFSSL_PTHREADS
    #define WOLFSSL_PTHREADS

    #define WOLFSSH_TEST_SERVER
    #define WOLFSSH_TEST_THREADING
[... 436 lines cut ...]
 *  #define WC_NO_CACHE_RESISTANT
 *  #define WC_AES_BITSLICED
 *  #define HAVE_AES_ECB
 *  #define HAVE_AES_DIRECT
 */
```

ESP32 works, but the ESP32C2 gives me the same error:

```
>>> crypto_srp_init: Generating salt
>>> crypto_srp_init: Setting SRP username
>>> crypto_srp_init: Setting SRP params
>>> crypto_srp_init: Setting SRP password
>>> crypto_srp_init: Getting SRP verifier
>>> crypto_srp_init: Failed to get SRP verifier (code -1)
!!! HomeKit: [Client 1] Failed to initialize SRP
>>> client_sendv: [Client 1] Sending payload: HTTP/1.1 200 OK\x0D\x0AContent-Type: application/pairing+tlv8\x0D\x0ATransfer-Encoding: chunked\x0D\x0AConnection: keep-alive\x0D\x0A\x0D\x0A
>>> client_sendv: [Client 1] Sending payload: \x36\x0D\x0A\x06\x01\x02\x07\x01\x01\x0D\x0A
>>> client_sendv: [Client 1] Sending payload: \x30\x0D\x0A\x0D\x0A
>>> homekit_client_process: [Client 1] Finished processing
>>> HomeKit: [Client 1] Closing client connection from 192.168.178.28
```

going to try something else, to see if it works.

gojimmypi, 2024-03-31:
Hi @AchimPieters - thank you so much for providing the `user_settings.h` and for detailing the ESP32 _works_ and the ESP32-C2 _does not work_ with SRP.

The first thing that comes to mind with one device working but not another is your memory configuration: Note that the C2 has less RAM: 272 KB of SRAM (16 KB dedicated to cache) [*1](https://www.espressif.com/en/products/socs/esp32-c2) vs the ESP32 that has 520 KB SRAM [*2](https://www.espressif.com/sites/default/files/documentation/esp32-wroom-32e_esp32-wroom-32ue_datasheet_en.pdf).

Could you please provide all of the settings in your `sdkconfig.h` that include the text `stack`, `heap`, and `IRAM`. You might also consider turning on stack smashing and/or heap corruption detection.

Also, I didn't see any setting for session caching. See the [user_settings.h for the TLS Client Example](https://github.com/wolfSSL/wolfssl/blob/d930825a924efb22e0631dd6a963292040a4d00c/IDE/Espressif/ESP-IDF/examples/wolfssl_client/components/wolfssl/include/user_settings.h#L59). Smaller cache can save memory.

```
#define SMALL_SESSION_CACHE
```

That said, the most recent wolfSSL v5.6.0 does not include all of my changes. One thing in particular with is SHA interleaving improvements that I am still testing. I'm not referring to the HW interleave capability, but rather the current HW implementation with multiple, concurrent SHA calculations that typically occur in something such as SRP. I recently added https://github.com/wolfSSL/wolfssl/pull/7262 that helps detect this.  See the Espressif-specific [wolfssl_test](https://github.com/wolfSSL/wolfssl/tree/master/IDE/Espressif/ESP-IDF/examples/wolfssl_test) example.

AchimPieters, 2024-04-01:
@gojimmypi Her is the complete `sdkconfig.h` [sdkconfig.txt](https://github.com/wolfSSL/wolfssl/files/14821736/sdkconfig.txt) 

 Turning on stack smashing and/or heap corruption detection, didn't provide any useful information.
 
 I added Smaller cache: [user_settings.txt](https://github.com/wolfSSL/wolfssl/files/14821938/user_settings.txt) but gave me the same error:
 
```
 >>> homekit_server_accept_client: Free heap: 89624
>>> HomeKit: [Client 1] Got new client connection from 192.168.178.28
>>> homekit_client_process: [Client 1] Got 132 incoming data
>>> homekit_server_on_pair_setup: Pair Setup
>>> homekit_server_on_pair_setup: Free heap: 91344
>>> tlv_debug: Got following TLV values:
>>> tlv_debug: Type 0 value (1 bytes): \x00
>>> tlv_debug: Type 6 value (1 bytes): \x01
>>> HomeKit: [Client 1] Pair Setup Step 1/3
>>> homekit_server_on_pair_setup: Free heap: 91252
>>> crypto_srp_new: Initializing SRP
>>> homekit_server_on_pair_setup: [Client 1] Initializing crypto
>>> homekit_server_on_pair_setup: Free heap: 88492
>>> homekit_server_on_pair_setup: [Client 1] Using user-specified password: 338-77-883
>>> crypto_srp_init: Generating salt
>>> crypto_srp_init: Setting SRP username
>>> crypto_srp_init: Setting SRP params
>>> crypto_srp_init: Setting SRP password
```

gojimmypi, 2024-04-06:
Hello @AchimPieters 

TL;DR: The version of wolfSSL you are using does not have my SRP fix. Try [this one](https://github.com/gojimmypi/wolfssl/tree/6f92cc98968d03baf14aaa54a8e6919d60365347).

Here are my notes:

Heads up the `Icon` file is still problematic on Visual Studio / Windows:

```text
C:\workspace>git clone --recursive https://github.com/AchimPieters/esp32-homekit-demo.git esp32-homekit-demo-gojimmypi
Cloning into 'esp32-homekit-demo-gojimmypi'...
remote: Enumerating objects: 4725, done.
remote: Counting objects: 100% (359/359), done.
remote: Compressing objects: 100% (257/257), done.
remote: Total 4725 (delta 136), reused 166 (delta 98), pack-reused 4366Receiving objects: 100% (4725/4725), 39.09 MiB | Receiving objects: 100% (4725/4725), 40.87 MiB | 6.44 MiB/s, done.

Resolving deltas: 100% (1815/1815), done.
error: invalid path 'Icon?'
fatal: unable to checkout working tree
warning: Clone succeeded, but checkout failed.
You can inspect what was checked out with 'git status'
and retry with 'git restore --source=HEAD :/'
```

I tried to submit a [request](https://developercommunity.visualstudio.com/t/Bad-filenames-result-in-Git-not-detectin/10489725) on the Visual Studio Developer Community without much luck.

When using Linux and adding an upstream like this:

```
git remote add upstream https://github.com/AchimPieters/esp32-homekit-demo.git
```

I saw this `remote error: upload-pack: not our ref` error:

```

gojimmypi:/mnt/c/workspace/esp32-homekit-demo-gojimmypi
$ git fetch upstream main
remote: Enumerating objects: 997, done.
remote: Counting objects: 100% (390/390), done.
remote: Compressing objects: 100% (199/199), done.
remote: Total 997 (delta 212), reused 255 (delta 187), pack-reused 607
Receiving objects: 100% (997/997), 15.65 MiB | 2.25 MiB/s, done.
Resolving deltas: 100% (463/463), completed with 90 local objects.
From https://github.com/AchimPieters/esp32-homekit-demo
 * branch            main       -> FETCH_HEAD
 * [new branch]      main       -> upstream/main
Fetching submodule components/homekit
fatal: remote error: upload-pack: not our ref 80ae7716b49e7005c441a3286a6a56af8fb42aff
Errors during submodule fetch:
        components/homekit
```

So I ended up having to deleting the entire repo and forking a fresh one.

Ha! Today I also learned one cannot have two forks of the same repo, even if the first fork is renamed.

In an any case, my [new fork of esp32-homekit-demo](https://github.com/gojimmypi/esp32-homekit-demo) seems to be much happier.

Upon inspection, the first thing I noticed is that the [esp32-homekit-demo wolfssl component](https://github.com/AchimPieters/esp32-homekit-demo/tree/main/components/wolfssl) appears to be a copy of the [wolfSSL v5.7.0 Espressif Managed Component](https://components.espressif.com/components/wolfssl/wolfssl/versions/5.7.0?language=en).

Please note that not only have my SRP changes [mentioned above](https://github.com/wolfSSL/wolfssl/issues/7210#issuecomment-2013235616)  _not_ made it into the `v5.7.0` release, I've not even yet created a PR to merge them into wolfssl. 

The good news, is that my [9x SRP testing. (all passing) commit](https://github.com/gojimmypi/wolfssl/commit/6f92cc98968d03baf14aaa54a8e6919d60365347) at `commit 6f92cc9` appears to be passing the SRP test for all flavors of ESP32 on [my testing jig](https://x.com/gojimmypi/status/1733644618049421602) including an ESP8266 that I later added to the center.

See the [wolfssl_test](https://github.com/gojimmypi/wolfssl/tree/6f92cc98968d03baf14aaa54a8e6919d60365347/IDE/Espressif/ESP-IDF/examples/wolfssl_test) app at that point in history on my fork.

How to proceed? For me, I'd clone wolfssl at the [root level](https://github.com/AchimPieters/esp32-homekit-demo) of the project, with the same parent directory as `images`, `Documentation`, etc. Be sure to remove the existing one.

Then in the `components` directory, only a few files are needed as seen in the [example](https://github.com/gojimmypi/wolfssl/tree/6f92cc98968d03baf14aaa54a8e6919d60365347/IDE/Espressif/ESP-IDF/examples/wolfssl_test/components/wolfssl):
[... 21 lines cut ...]
        #endif

        #if (CONFIG_ESP_MAIN_TASK_STACK_SIZE < ESP_SRP_MINIMUM_STACK)
            #warning "WOLFCRYPT_HAVE_SRP enabled with small stack size"
        #endif
    #endif
#else
    #warning "CONFIG_ESP_MAIN_TASK_STACK_SIZE not defined!"
#endif
```

The 28K is a very bare-minimum amount for the `wolfssl_test` exercise. Any other application would likely need
additional overhead.

Here's the output from my [test script](https://github.com/gojimmypi/wolfssl/blob/6f92cc98968d03baf14aaa54a8e6919d60365347/IDE/Espressif/ESP-IDF/examples/wolfssl_test/testAll.sh) showing the SRP passing on all devices:

[image]

Note I've also added some additional `sdkconfig.default.[device]` files that adjusts the memory as appropriate.

Let me know if you are able to test with my version of wolfSSL.

I am certain we can get your Apple Homekit project working on every flavor of ESP32.

Thank you again for your patience and persistence.  Cheers.

AchimPieters, 2024-04-06:
@gojimmypi thank you for the extensive work you have done, at the moment I am ill (Notting serious), but gave it a quick go: you can find the branch here: https://github.com/AchimPieters/esp32-homekit-demo/tree/RSP-WOLFSSL

But now I got these errors:

```
In file included from /project/components/wolfssl/wolfssl/wolfcrypt/setIn file included from /project/components/wolfssl/wolfssl/wolfcrypt/settings.h:301,
                 from /project/components/homekit/src/crypto.c:3:
/project/components/wolfssl/include/user_settings.h:541:22: warning: #warning "RSA may be difficult with less than 10KB Stack "/ [-Wcpp]
  541 |                     #warning "RSA may be difficult with less than 10KB Stack "/
      |                      ^~~~~~~
In file included from /project/components/homekit/src/crypto.c:4:
/project/components/wolfssl/include/user_settings.h:541:22: warning: #warning "RSA may be difficult with less than 10KB Stack "/ [-Wcpp]
  541 |                     #warning "RSA may be difficult with less than 10KB Stack "/
      |                      ^~~~~~~
/project/components/homekit/src/crypto.c: In function 'crypto_chacha20poly1305_decrypt':
/project/components/homekit/src/crypto.c:282:29: error: 'CHACHA20_POLY1305_AEAD_AUTHTAG_SIZE' undeclared (first use in this function)
  282 |         if (message_size <= CHACHA20_POLY1305_AEAD_AUTHTAG_SIZE) {
      |                             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/project/components/homekit/src/crypto.c:282:29: note: each undeclared identifier is reported only once for each function it appears in
/project/components/homekit/src/crypto.c:298:17: error: implicit declaration of function 'wc_ChaCha20Poly1305_Decrypt'; did you mean 'crypto_chacha20poly1305_decrypt'? [-Werror=implicit-function-declaration]
  298 |         int r = wc_ChaCha20Poly1305_Decrypt(
      |                 ^~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                 crypto_chacha20poly1305_decrypt
/project/components/homekit/src/crypto.c: In function 'crypto_chacha20poly1305_encrypt':
/project/components/homekit/src/crypto.c:315:37: error: 'CHACHA20_POLY1305_AEAD_AUTHTAG_SIZE' undeclared (first use in this function)
  315 |         size_t len = message_size + CHACHA20_POLY1305_AEAD_AUTHTAG_SIZE;
      |                                     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/project/components/homekit/src/crypto.c:323:17: error: implicit declaration of function 'wc_ChaCha20Poly1305_Encrypt'; did you mean 'crypto_chacha20poly1305_encrypt'? [-Werror=implicit-function-declaration]
  323 |         int r = wc_ChaCha20Poly1305_Encrypt(
      |                 ^~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                 crypto_chacha20poly1305_encrypt
/project/components/homekit/src/crypto.c: In function 'crypto_curve25519_init':
/project/components/homekit/src/crypto.c:466:17: error: implicit declaration of function 'wc_curve25519_init'; did you mean 'wc_ed25519_init'? [-Werror=implicit-function-declaration]
  466 |         int r = wc_curve25519_init(key);
      |                 ^~~~~~~~~~~~~~~~~~
      |                 wc_ed25519_init
/project/components/homekit/src/crypto.c: In function 'crypto_curve25519_done':
/project/components/homekit/src/crypto.c:478:9: error: implicit declaration of function 'wc_curve25519_free'; did you mean 'wc_ed25519_free'? [-Werror=implicit-function-declaration]
  478 |         wc_curve25519_free(key);
      |         ^~~~~~~~~~~~~~~~~~
      |         wc_ed25519_free
/project/components/homekit/src/crypto.c: In function 'crypto_curve25519_generat':
/project/components/homekit/src/crypto.c:496:13: error: implicit declaration of function 'wc_curve25519_make_key'; did you mean 'wc_ed25519_make_key'? [-Werror=implicit-function-declaration]
  496 |         r = wc_curve25519_make_key(&rng, 32, key);
      |             ^~~~~~~~~~~~~~~~~~~~~~
      |             wc_ed25519_make_key
/project/components/homekit/src/crypto.c: In function 'crypto_curve25519_import_public':
/project/components/homekit/src/crypto.c:507:16: error: implicit declaration of function 'wc_curve25519_import_public_ex'; did you mean 'wc_ed25519_import_public_ex'? [-Werror=implicit-function-declaration]
  507 |         return wc_curve25519_import_public_ex(data, size, key, EC25519_LITTLE_ENDIAN);
      |                ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                wc_ed25519_import_public_ex
/project/components/homekit/src/crypto.c:507:64: error: 'EC25519_LITTLE_ENDIAN' undeclared (first use in this function)
  507 |         return wc_curve25519_import_public_ex(data, size, key, EC25519_LITTLE_ENDIAN);
      |                                                                ^~~~~~~~~~~~~~~~~~~~~
/project/components/homekit/src/crypto.c: In function 'crypto_curve25519_export_public':
/project/components/homekit/src/crypto.c:514:25: error: implicit declaration of function 'wc_curve25519_export_public_ex'; did you mean 'wc_ed25519_import_public_ex'? [-Werror=implicit-function-declaration]
  514 |                 int r = wc_curve25519_export_public_ex(
      |                         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                         wc_ed25519_import_public_ex
/project/components/homekit/src/crypto.c:517:25: error: 'EC25519_LITTLE_ENDIAN' undeclared (first use in this function)
  517 |                         EC25519_LITTLE_ENDIAN
      |                         ^~~~~~~~~~~~~~~~~~~~~
/project/components/homekit/src/crypto.c: In function 'crypto_curve25519_shared_secret':
[... cut ...]
/project/components/homekit/src/crypto.c:508:1: error: control reaches end of non-void function [-Werror=return-type]
  508 | }
      | ^
cc1: some warnings being treated as errors
ninja: build stopped: subcommand failed.
ninja failed with exit code 1, output of the command is in the /project/examples/led/build/log/idf_py_stderr_output_9900 and /project/examples/led/build/log/idf_py_stdout_output_9900
root@48772da10d14:/project/examples/led# 
```
Maybe It will give you an idea right away, but for me, I have to put it on hold, until i'm better.

AchimPieters, 2024-04-07:
@gojimmypi did some tweaking to `user_settings.h` and changed the sdkconfig files as suggested.
Here are my latest files: https://github.com/AchimPieters/esp32-homekit-demo/tree/RSP-WOLFSSL

But now I got a real strange one:

```
/project/components/wolfssl/wolfcrypt/src/port/Espressif/esp_sdk_wifi_lib.c:37:10: fatal error: esp_wifi.h: No such file or directory
   37 | #include <esp_wifi.h>
      |          ^~~~~~~~~~~~
compilation terminated.
[869/921] Building C object esp-idf/wolfssl/CMakeFiles/_....dir/wolfcrypt/src/port/Espressif/esp_sdk_time_lib.c.ob
FAILED: esp-idf/wolfssl/CMakeFiles/__idf_wolfssl.dir/wolfcrypt/src/port/Espressif/esp_sdk_time_lib.c.obj 

```
And
```
/project/components/wolfssl/wolfcrypt/src/port/Espressif/esp_sdk_time_lib.c:62:18: fatal error: esp_netif_sntp.h: No such file or directory
   62 |         #include <esp_netif_sntp.h>
      |                  ^~~~~~~~~~~~~~~~~~
compilation terminated.
[872/921] Building C object esp-idf/wolfssl/CMakeFiles/__idf_wolfssl.dir/wolfcrypt/src/tfm.c.obj
ninja: build stopped: subcommand failed.
Compilation failed because esp_sdk_wifi_lib.c (in "wolfssl" component) includes esp_wifi.h, provided by esp_wifi component(s).
However, esp_wifi component(s) is not in the requirements list of "wolfssl".
To fix this, add esp_wifi to PRIV_REQUIRES list of idf_component_register call in /project/components/wolfssl/CMakeLists.txt.
ninja failed with exit code 1, output of the command is in the /project/examples/led/build/log/idf_py_stderr_output_7752 and /project/examples/led/build/log/idf_py_stdout_output_7752
root@e3b704ffe19d:/project/examples/led# 
```

But when I'm looking into the source directory's:  the `/components/wolfssl/wolfcrypt/src/port/Espressif/esp_sdk_wifi_lib` AND the `components/wolfssl/wolfcrypt/src/port/Espressif/esp_sdk_time_lib.c` is also there?

I don't want to fiddle too much, because otherwise we don't remember what we did, don't you agree?

AchimPieters, 2024-04-08:
@gojimmypi Here are what I did and the results:

`git clone --recursive https://github.com/gojimmypi/esp32-homekit-demo.git esp32-homekit-demo-gojimmypi`
`cd  ./esp32-homekit-demo-gojimmypi`

`git remote add upstream https://github.com/AchimPieters/esp32-homekit-demo.git`

`git clone https://github.com/gojimmypi/wolfssl.git`

`cd wolfssl`
`git checkout a84feaba2454b4262f29dd9b1ee80465d00114c7`
`cd ..`

```
rm -rf ./components/wolfssl/examples
rm -rf ./components/wolfssl/src
rm -rf ./components/wolfssl/wolfssl
rm -rf ./components/wolfssl/wolfcrypt
rm     ./components/wolfssl/.component_hash
rm     ./components/wolfssl/idf_component.yml
```

`cp ./wolfssl/IDE/Espressif/ESP-IDF/examples/template/components/wolfssl/CMakeLists`.txt `./components/wolfssl/CMakeLists.txt`

### **ESP32**

`cd ./examples/led`
`idf.py set-target esp32`
`idf.py menuconfig`
`idf.py build`

- [x] Compling
- [x] Connecting to iPhone
- [x] Overal function

### **ESP32C2**

`cd ./examples/led`
`idf.py set-target esp32C2`
`idf.py menuconfig`
`idf.py build`

- [x] Compling
- [ ] Connecting to iPhone
- [ ] Overal function

```
>>> crypto_srp_init: Generating salt
>>> crypto_srp_init: Setting SRP username
>>> crypto_srp_init: Setting SRP params
>>> crypto_srp_init: Setting SRP password
>>> crypto_srp_init: Getting SRP verifier
>>> crypto_srp_init: Failed to get SRP verifier (code -1)
!!! HomeKit: [Client 1] Failed to initialize SRP
>>> client_sendv: [Client 1] Sending payload: HTTP/1.1 200 OK\x0D\x0AContent-Type: application/pairing+tlv8\x0D\x0ATransfer-Encoding: chunked\x0D\x0AConnection: keep-alive\x0D\x0A\x0D\x0A
>>> client_sendv: [Client 1] Sending payload: \x36\x0D\x0A\x06\x01\x02\x07\x01\x01\x0D\x0A
>>> client_sendv: [Client 1] Sending payload: \x30\x0D\x0A\x0D\x0A
>>> homekit_client_process: [Client 1] Finished processing
>>> HomeKit: [Client 1] Closing client connection from 192.168.178.28
```

Unfortunally I did not got the `wolf_test_task();` to work, or maybe I don't know how? What I did is adding it to my `main.c` code, it compiled but did not give any output?

gojimmypi, 2024-04-08:
@AchimPieters that's curious the SRP is failing on your ESP32-C2, but not the ESP32. Do you have any other flavors to test with?

Can you provide your stack and IRAM settings? Here are mine:

## IRAM
[image]
[image]

## Stack
[image]

## Heap
[image]

See also my:

-  [wolfssl_test user_settings.h](https://github.com/gojimmypi/wolfssl/blob/ED25519_SHA2_fix/IDE/Espressif/ESP-IDF/examples/wolfssl_test/components/wolfssl/include/user_settings.h) 
- [sdkconfig.defaults](https://github.com/gojimmypi/wolfssl/blob/ED25519_SHA2_fix/IDE/Espressif/ESP-IDF/examples/wolfssl_test/sdkconfig.defaults)

In particular, although the ESP8266 is configured for [#define FP_MAX_BITS MIN_FFDHE_FP_MAX_BITS](https://github.com/gojimmypi/wolfssl/blob/a84feaba2454b4262f29dd9b1ee80465d00114c7/IDE/Espressif/ESP-IDF/examples/wolfssl_test/components/wolfssl/include/user_settings.h#L131) due to limited memory, the C2 is using the full-sized [#define FP_MAX_BITS (8192 * 2)](https://github.com/gojimmypi/wolfssl/blob/a84feaba2454b4262f29dd9b1ee80465d00114c7/IDE/Espressif/ESP-IDF/examples/wolfssl_test/components/wolfssl/include/user_settings.h#L147).

I think we tried this before, but what happens if you [disabled all hardware acceleration](https://github.com/gojimmypi/wolfssl/blob/a84feaba2454b4262f29dd9b1ee80465d00114c7/IDE/Espressif/ESP-IDF/examples/wolfssl_test/components/wolfssl/include/user_settings.h#L524)? (There's [limited capability](https://github.com/gojimmypi/wolfssl/blob/a84feaba2454b4262f29dd9b1ee80465d00114c7/IDE/Espressif/ESP-IDF/examples/wolfssl_test/components/wolfssl/include/user_settings.h#L432) in the ESP32-C2 anyhow)

To disabled all HW acceleration:
```c
    #define NO_ESP32_CRYPT
    #define NO_WOLFSSL_ESP32_CRYPT_HASH
    #define NO_WOLFSSL_ESP32_CRYPT_AES
    #define NO_WOLFSSL_ESP32_CRYPT_RSA_PRI
```

Let's see what the results of the [wolfssl_test](https://github.com/gojimmypi/wolfssl/tree/ED25519_SHA2_fix/IDE/Espressif/ESP-IDF/examples/wolfssl_test) are on your device. Would you please build and flash that onto your device and let me know the results?

For completeness, here's my ESP32-C2 log showing a passing SRP test:

```
ESP-ROM:esp8684-api2-20220127
Build:Jan 27 2022
rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd5c80,len:0x16f0
load:0x403acb70,len:0xbb0
load:0x403aeb70,len:0x27e0
entry 0x403acb7a
I (21) boot: ESP-IDF v5.2-dev-3903-g66992aca7a-dirty 2nd stage bootloader
I (21) boot: compile time Apr  5 2024 14:59:47
I (22) boot: chip revision: v1.0
I (26) boot.esp32c2: MMU Page Size  : 32K
I (31) boot.esp32c2: SPI Speed      : 60MHz
I (36) boot.esp32c2: SPI Mode       : DIO
I (40) boot.esp32c2: SPI Flash Size : 2MB
I (45) boot: Enabling RNG early entropy source...
I (50) boot: Partition Table:
I (54) boot: ## Label            Usage          Type ST Offset   Length
I (61) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (69) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (76) boot:  2 factory          factory app      00 00 00010000 00177000
I (84) boot: End of partition table
I (88) esp_image: segment 0: paddr=00010020 vaddr=3c0a8020 size=2cf58h (184152)map
I (138) esp_image: segment 1: paddr=0003cf80 vaddr=3fcaa4e0 size=015bch (  5564 load
I (139) esp_image: segment 2: paddr=0003e544 vaddr=40380000 size=01ad4h (  6868 load
I (145) esp_image: segment 3: paddr=00040020 vaddr=42000020 size=a7570h (685424 map
I (305) esp_image: segment 4: paddr=000e7598 vaddr=40381ad4 size=08a00h ( 35328 load
I (318) boot: Loaded app from partition at offset 0x10000
I (318) boot: Disabling RNG early entropy source...
I (329) cpu_start: Unicore app
I (338) cpu_start: Pro cpu start user code
I (338) cpu_start: cpu freq: 120000000 Hz
I (338) cpu_start: Application information:
[... 92 lines cut ...]
------------------------------------------------------------------------------
I (909) wc_test: Here we go with srp_test!
I (914) wc_port: wc_InitMutex success heap = 163748
SRP      test passed!
error    test passed!
MEMORY   test passed!
base64   test passed!
base16   test passed!
AES      test passed!
SHA-224  test passed!
HMAC-SHA224 test passed!
AES192   test passed!
SHA-256  test passed!
SHA-224  test passed!
SHA      test passed!
ED25519  test passed!
SHA-224  test passed!
SHA-384  test passed!
SHA-512  test passed!
SHA-512/224  test passed!
SHA-512/256  test passed!
etc...
```

I expect the SRP test to pass on your device. We can then compare if there are any configs in the [user_settings.h](https://github.com/gojimmypi/wolfssl/blob/ED25519_SHA2_fix/IDE/Espressif/ESP-IDF/examples/wolfssl_test/components/wolfssl/include/user_settings.h) and/or [sdkconfig,defaults](https://github.com/gojimmypi/wolfssl/blob/ED25519_SHA2_fix/IDE/Espressif/ESP-IDF/examples/wolfssl_test/sdkconfig.defaults) that are different from yours. It is possible that something else is going on, and the current [SRP test](https://github.com/gojimmypi/wolfssl/blob/a84feaba2454b4262f29dd9b1ee80465d00114c7/wolfcrypt/test/test.c#L23874) is missing something that the Apple Homekit is doing.

AchimPieters, 2024-04-08:
@gojimmypi _Can you provide your stack and IRAM settings?_ Sure, here are mine:

**ESP32C2 IRAM:**

```
# CONFIG_GPIO_CTRL_FUNC_IN_IRAM is not set

CONFIG_GPTIMER_ISR_HANDLER_IN_IRAM=y
# CONFIG_GPTIMER_CTRL_FUNC_IN_IRAM is not set
# CONFIG_GPTIMER_ISR_IRAM_SAFE is not set

# CONFIG_I2C_ISR_IRAM_SAFE is not set

# CONFIG_LEDC_CTRL_FUNC_IN_IRAM is not set

# CONFIG_SPI_MASTER_IN_IRAM is not set
CONFIG_SPI_MASTER_ISR_IN_IRAM=y
# CONFIG_SPI_SLAVE_IN_IRAM is not set
CONFIG_SPI_SLAVE_ISR_IN_IRAM=y

# CONFIG_UART_ISR_IN_IRAM is not set

CONFIG_ESP_EVENT_POST_FROM_IRAM_ISR=y

CONFIG_PERIPH_CTRL_FUNC_IN_IRAM=y

CONFIG_GDMA_CTRL_FUNC_IN_IRAM=y

CONFIG_ESP_SPI_BUS_LOCK_ISR_FUNCS_IN_IRAM=y

# CONFIG_ESP_PANIC_HANDLER_IRAM is not set

CONFIG_ESP_WIFI_IRAM_OPT=y
# CONFIG_ESP_WIFI_EXTRA_IRAM_OPT is not set
CONFIG_ESP_WIFI_RX_IRAM_OPT=y

# CONFIG_ESP_WIFI_SLP_IRAM_OPT is not set

CONFIG_HAL_SPI_MASTER_FUNC_IN_IRAM=y
CONFIG_HAL_SPI_SLAVE_FUNC_IN_IRAM=y

# CONFIG_LWIP_IRAM_OPTIMIZATION is not set
# CONFIG_LWIP_EXTRA_IRAM_OPTIMIZATION is not set

CONFIG_POST_EVENTS_FROM_IRAM_ISR=y

CONFIG_ESP32_WIFI_IRAM_OPT=y
CONFIG_ESP32_WIFI_RX_IRAM_OPT=y

```

**ESP32C2 STACK:**

```
CONFIG_COMPILER_STACK_CHECK_MODE_NONE=y
# CONFIG_COMPILER_STACK_CHECK_MODE_NORM is not set
# CONFIG_COMPILER_STACK_CHECK_MODE_STRONG is not set
# CONFIG_COMPILER_STACK_CHECK_MODE_ALL is not set

CONFIG_ESP_SYSTEM_EVENT_TASK_STACK_SIZE=2304
CONFIG_ESP_MAIN_TASK_STACK_SIZE=3584

CONFIG_ESP_MINIMAL_SHARED_STACK_SIZE=2048

CONFIG_ESP_SYSTEM_HW_STACK_GUARD=y

CONFIG_ESP_IPC_TASK_STACK_SIZE=1024

CONFIG_ESP_TIMER_TASK_STACK_SIZE=3584

# CONFIG_FATFS_LFN_STACK is not set

# CONFIG_FREERTOS_CHECK_STACKOVERFLOW_NONE is not set
# CONFIG_FREERTOS_CHECK_STACKOVERFLOW_PTRVAL is not set
CONFIG_FREERTOS_CHECK_STACKOVERFLOW_CANARY=y

CONFIG_FREERTOS_IDLE_TASK_STACKSIZE=1536

CONFIG_FREERTOS_TIMER_TASK_STACK_DEPTH=2048

# CONFIG_FREERTOS_WATCHPOINT_END_OF_STACK is not set

CONFIG_FREERTOS_ISR_STACKSIZE=1536

CONFIG_LWIP_TCPIP_TASK_STACK_SIZE=3072

CONFIG_PTHREAD_TASK_STACK_SIZE_DEFAULT=3072
CONFIG_PTHREAD_STACK_MIN=768

CONFIG_MDNS_TASK_STACK_SIZE=4096

CONFIG_STACK_CHECK_NONE=y
# CONFIG_STACK_CHECK_NORM is not set
# CONFIG_STACK_CHECK_STRONG is not set
# CONFIG_STACK_CHECK_ALL is not set

CONFIG_SYSTEM_EVENT_TASK_STACK_SIZE=2304
CONFIG_MAIN_TASK_STACK_SIZE=3584

CONFIG_IPC_TASK_STACK_SIZE=1024
CONFIG_TIMER_TASK_STACK_SIZE=3584

CONFIG_TIMER_TASK_STACK_DEPTH=2048

CONFIG_TCPIP_TASK_STACK_SIZE=3072

CONFIG_ESP32_PTHREAD_TASK_STACK_SIZE_DEFAULT=3072
CONFIG_ESP32_PTHREAD_STACK_MIN=768
```

AchimPieters, 2024-04-08:
@gojimmypi the results of the [wolfssl_test](https://github.com/gojimmypi/wolfssl/tree/ED25519_SHA2_fix/IDE/Espressif/ESP-IDF/examples/wolfssl_test) are as following:

### **ESP32**

`docker run -it -v ~/esp32-homekit-demo-gojimmypi:/project -w /project espressif/idf:latest`

`cd  ./esp32-homekit-demo-gojimmypi`

`cd /wolfssl/IDE/Espressif/ESP-IDF/examples/wolfssl_test`

`idf.py set-target esp32`

`idf.py build`

`esptool.py erase_flash`

`python -m esptool --chip esp32 -b 460800 --before default_reset --after hard_reset write_flash --flash_mode dio --flash_size 2MB --flash_freq 40m 0x1000 build/bootloader/bootloader.bin 0x8000 build/partition_table/partition-table.bin 0x10000 build/wolfssl_test.bin`

Here are the results:
[ESP32 LOG.txt](https://github.com/wolfSSL/wolfssl/files/14908131/ESP32.LOG.txt)

### **ESP32C2**

`docker run -it -v ~/esp32-homekit-demo-gojimmypi:/project -w /project espressif/idf:latest`

`cd  ./esp32-homekit-demo-gojimmypi`

`cd /wolfssl/IDE/Espressif/ESP-IDF/examples/wolfssl_test`

`idf.py set-target esp32c2`

`idf.py build`

`esptool.py erase_flash`

`python -m esptool --chip esp32c2 -b 460800 --before default_reset --after hard_reset write_flash --flash_mode dio --flash_size 2MB --flash_freq 40m 0x1000 build/bootloader/bootloader.bin 0x8000 build/partition_table/partition-table.bin 0x10000 build/wolfssl_test.bin`

Here are the results:
[ESP32C2 LOG.txt ](https://github.com/wolfSSL/wolfssl/files/14908136/ESP32C2.LOG.txt)

gojimmypi, 2024-04-08:
> here are mine: ... ESP32C2 STACK:

aha! The first thing I notice is our [sdkconfig settings differences](https://github.com/gojimmypi/wolfssl/commit/aa4f9718ed1b853a8210fe74e3f8b9a33b370926), in particular the main task stack:

`CONFIG_ESP_MAIN_TASK_STACK_SIZE=10500` vs `CONFIG_ESP_MAIN_TASK_STACK_SIZE=3584`

That alone is not necessarily the root cause, as I see you have several freeRTOS tasks, such as [the one in led/main/main.c](https://github.com/AchimPieters/esp32-homekit-demo/blob/eaaa4fd04a5cebd7f899c9b3f18eb73e551df659/examples/led/main/main.c#L104) allocating only 2K stack.

See the [Espressif docs on `xTaskCreate`](https://docs.espressif.com/projects/esp-idf/en/v4.2/esp32s2/api-reference/system/freertos.html#_CPPv423xTaskCreatePinnedToCore14TaskFunction_tPCKcK8uint32_tPCv11UBaseType_tPC12TaskHandle_tK10BaseType_t).

There's also the [components/homekit/src/server.c](https://github.com/AchimPieters/esp32-homekit/blob/e798825d7814a563b06f3da084f0114c3e415dd6/src/server.c#L4358) that appears to allocate only [1664 bytes](https://github.com/AchimPieters/esp32-homekit/blob/e798825d7814a563b06f3da084f0114c3e415dd6/src/port.h#L35) if `ESP_IDF` is not defined.

I don't recognize the `ESP_IDF` macro. It is not defined in my `wolfssl_test` project, nor could I find one in any of the source files.

There's a cmake definition in the [esp32-button](https://github.com/AchimPieters/esp32-button/blob/a5308eb90a9232867e07eae829eb5fbff2960b99/CMakeLists.txt#L7), and the [homekit component](https://github.com/AchimPieters/esp32-homekit/blob/e798825d7814a563b06f3da084f0114c3e415dd6/CMakeLists.txt#L23) but I'm not certain that the task calling the wolfSSL SRP sees that & the appropriate stack size.

It might be a good idea to ensure we know the exact value of `SERVER_TASK_STACK` during the call. I would _think_ that it is always the `12288` value, as you indicated the code works for the ESP32. Still, I'd try to set it to an known value and put in a `ESP_LOGI()` to confirm.

It is challenging for me to not have any of the Apple Homekit products. Are you aware of any emulation environments that I might be able to use to test? It would make this exercise much easier.

You might also consider turning on stack smashing features to see if the code will tell us if indeed there's a stack problem. See the `CONFIG_STACK_CHECK_NONE` setting.

> Here are the results: [ESP32C2 LOG.txt](https://github.com/wolfSSL/wolfssl/files/14908136/ESP32C2.LOG.txt)

Excellent! That at least confirms the SRP test is passing on your ESP32C2. Even more evidence this is a stack setting issue.

AchimPieters, 2024-04-09:
@gojimmypi good to hear this gives more insight in the problem, regarding your question about an emulator, there are a few out there depending on which you have a good feeling, there is this one for example: https://dev.to/ianito/how-to-emulate-ios-on-linux-with-docker-4gj3

or

https://hackintosh.com

I'm going to look into the intel you provided...

gojimmypi, 2024-04-15:
Hi @AchimPieters - I know you mentioned you've not been feeling well... so no rush here.. just wondering if you have an update on the ESP32-C2 SRP Homekit failure that you were seeing? 

I took a brief look at some of the Apple emulators and simulators. One cannot deny the simplicity of testing on real hardware. If you are still seeing problems & have a loaner iPhone that I can for testing, that would be great! That's a very generous offer.  Please reach out to me via email: jim (at) wolfssl.com

In return I can probably arrange to send you some wolfssl swag & I'll of course return the device upon completion of our exercise.

AchimPieters, 2024-04-16:
@gojimmypi no update yet, I'm going to try doing some test this weekend ;) I'm currently a little behind, but I'll be fine!

I will email you, to discuss some things regarding the iPhone.

AchimPieters, 2024-05-03:
@gojimmypi 

Okay, let's start from my repro at this moment: https://github.com/AchimPieters/esp32-homekit-demo with my current [user_settings.h](https://github.com/AchimPieters/esp32-homekit-demo/blob/main/components/wolfssl/include/user_settings.h) and go from here...

gojimmypi, 2024-05-04:
@AchimPieters - excellent! I believe I have all the issues worked out.

Having the iPhone was quite helpful. Thank you again.

See your PR https://github.com/AchimPieters/esp32-homekit-demo/pull/3 and wolfSSL https://github.com/wolfSSL/wolfssl/pull/7505.

For reference, here are my instructions to test on my fork:

```
# set TEST_DIR to fully qualified path:
# export TEST_DIR=~/yourdir
export TEST_DIR=/mnt/c/test-homekit

cd $TEST_DIR

mkdir test-homekit
cd    test-homekit

# get a copy of the gojimmypi fork of wolfSSL (see PR)
git clone https://github.com/gojimmypi/wolfssl.git wolfssl-gojimmypi
cd wolfssl-gojimmypi
git checkout PR-Apple-Homekit-SRP-fix
cd ..

# if wolfSSL was fetched to `wolfssl` or 'wolfssl-$USER` or `wolfssl-master`
# then cmake would find it if in parent directory of example
# since it is not, we set the WOLFSSL_ROOT source code directory an environment variable:

export WOLFSSL_ROOT=$TEST_DIR/wolfssl-gojimmypi

# get a copy of the gojimmypi esp32-homekit-demo fork (see PR)
git clone --recursive https://github.com/gojimmypi/esp32-homekit-demo.git esp32-homekit-demo-gojimmypi
cd  ./esp32-homekit-demo-gojimmypi
git remote add upstream https://github.com/AchimPieters/esp32-homekit-demo.git
git checkout PR-wolfssl

# Windows users may need:
git config core.fileMode false

cd ./examples/led

# Set path for your ESP-IDF, shown here for v5.2 in WSL for VisualGDB
WRK_IDF_PATH=/mnt/c/SysGCC/esp32/esp-idf/v5.2

. ${WRK_IDF_PATH}/export.sh

idf.py set-target esp32c2
idf.py menuconfig
idf.py build
```

AchimPieters, 2024-05-05:
@gojimmypi 

That's fantastic news! It's great to see progress being made. However, despite following all your instructions, I wasn't able to reproduce it. Even after accepting your merge request and trying again, I still couldn't get it to work. I suspect the difference in operating systems, Windows versus macOS, might be causing the issue.

I had to revert the merge again. Additionally, the changes to the local, static, stale copy of wolfSSL are causing complications, along with the fact that I'm using Docker.

I'd prefer to use wolfSSL via https://components.espressif.com/ because I've built this repro for beginners, so they can start with creating a HomeKit device in a simple and educational way. For this reason, I'd like to offer a repro that includes all the necessary components to start without any extra steps.

Once again, it's amazing that it worked for you! Now, I'll need to figure out how to make it work on my platform.

gojimmypi, 2024-05-05:
@AchimPieters 

> That's fantastic news! It's great to see progress being made. 

Yes! I'm confident we'll be able to get this working properly.

>  after accepting your merge request and trying again, I still couldn't get it to work

I'll need a little more than "_couldn't get it to work_", please.  New error messages? Some of the changes to your code in my repo are essential to getting the ESP2-C2 to work, as well as changes to wolfSSL:  Key to this working is also the merge of https://github.com/wolfSSL/wolfssl/pull/7505 into wolfSSL, which has not occurred yet.

You can continue to test with [my branch](https://github.com/gojimmypi/esp32-homekit-demo/tree/PR-wolfssl), if you'd like. Any other option is unlikely to succeed without code changes.

> the changes to the local, static, stale copy of wolfSSL are causing complications.

I'll need specifics to "_complications_" in order to be able to help.

There *must* be changes to wolfSSL in order for the ESP32-C2 to work properly for all your Homekit examples and all the Espressif targets.  I'm particularly interested in the value of `WOLFSSL_ROOT`.

> I suspect the difference in operating systems, Windows versus macOS, might be causing the issue

I'm using the Espressif v5.2. If you are using the same version, I think it is highly unlikely (though not impossible) that OS difference would cause problems. I need to see the errors you are encountering.

> I'd prefer to use wolfSSL via https://components.espressif.com/

Yes, I agree, however that's not going to happen today, nor in the immediate future, at least not for the official wolfSSL. I do however, have a [staging instance of wolfssl](https://components-staging.espressif.com/components/gojimmypi/mywolfssl) called `mywolfssl` that _could_ be used for preliminary testing of that. But I would need to publish an update, and I'd prefer to do that from the official wolfSSL branch _after_ my PR 7505 there gets merged. I want to make sure this is working for you _before_ updating those published libraries.

Using the managed component would also address any possible license issues of _distributing_ wolfSSL source code with your project or anyone else that uses it and distributes it.

I suggest trying my [PR-wolfssl](https://github.com/AchimPieters/esp32-homekit-demo) branch of your repo, even making wolfSSL a temporary, static local copy with my [wolfSSL PR Branch](https://github.com/gojimmypi/wolfssl/tree/PR-Apple-Homekit-SRP-fix) called  `PR-Apple-Homekit-SRP-fix`. Baby steps. Let's get that working for you and go from there.

I assumed you'd do the testing as noted in https://github.com/AchimPieters/esp32-homekit-demo/pull/3 instruction _before_ merging. The instructions there are for the forks and branches.

The most important thing to look for is the value of `WOLFSSL_ROOT`. I wonder if that's being set and possibly overridden in your docker environment? It is important, and I display it during the processing of the wolfSSL cmake file. Look for instances of `WOLFSSL_ROOT` in the cmake output. The most important one being the value. For instance for me:

```
-- WOLFSSL_EXTRA_PROJECT_DIR=C:/workspace/wolfssl-gojimmypi/src/
-- WOLFSSL_FOUND_IDF=0
-- WOLFSSL_PROJECT_DIR=C:/workspace/wolfssl-gojimmypi/IDE/Espressif/ESP-IDF/examples/wolfssl_client/components/wolfssl
-- WOLFSSL_ROOT=C:/workspace/wolfssl-gojimmypi
```

We need to also ensure you are not using wolfSSL in the ESP-IDF components. (see `WOLFSSL_FOUND_IDF=0`, above). The `WOLFSSL_ROOT` path should address that. 

When following those instructions, I'd like to see the value of `WOLFSSL_ROOT` environment variable from this step, and proof that the path exists and contains my latest update to wolfSSL:

```
export WOLFSSL_ROOT=$TEST_DIR/wolfssl-gojimmypi
``` 

I'd like to see full build/flash logs and any relevant device output on the serial monitor, please. I realize that we're 9 time zones apart and it is later for you at the moment, but if you can get that to me today, I'll work on it today.

Thanks so much for your persistence. We're really close, and I am completely confident we'll have a solid solution soon.

AchimPieters, 2024-05-05:
@gojimmypi  Her is what I tried so far:

Followed your instructions, as far as the are applicable on my system (macOS), I made a  shell script (.sh) file and run it:
```
# set TEST_DIR to fully qualified path:
# export TEST_DIR=~/yourdir
export TEST_DIR=/mnt/c/test-homekit

cd $TEST_DIR

mkdir test-homekit
cd    test-homekit

# get a copy of the gojimmypi fork of wolfSSL (see PR)
git clone https://github.com/gojimmypi/wolfssl.git wolfssl-gojimmypi
cd wolfssl-gojimmypi
git checkout PR-Apple-Homekit-SRP-fix
cd ..

# if wolfSSL was fetched to `wolfssl` or 'wolfssl-$USER` or `wolfssl-master`
# then cmake would find it if in parent directory of example
# since it is not, we set the WOLFSSL_ROOT source code directory an environment variable:

export WOLFSSL_ROOT=$TEST_DIR/wolfssl-gojimmypi

# get a copy of the gojimmypi esp32-homekit-demo fork (see PR)
git clone --recursive https://github.com/gojimmypi/esp32-homekit-demo.git esp32-homekit-demo-gojimmypi
cd  ./esp32-homekit-demo-gojimmypi
git remote add upstream https://github.com/AchimPieters/esp32-homekit-demo.git
git checkout PR-wolfssl

```
with this as result:

```
Last login: Sun May  5 17:16:26 on ttys000
achimpieters@Achims-iMac ~ % bash run.sh
run.sh: line 5: cd: /mnt/c/test-homekit: No such file or directory
Cloning into 'wolfssl-gojimmypi'...
remote: Enumerating objects: 184809, done.
remote: Counting objects: 100% (2598/2598), done.
remote: Compressing objects: 100% (1188/1188), done.
remote: Total 184809 (delta 1632), reused 2132 (delta 1323), pack-reused 182211
Receiving objects: 100% (184809/184809), 628.44 MiB | 21.62 MiB/s, done.
Resolving deltas: 100% (135634/135634), done.
Updating files: 100% (2730/2730), done.
branch 'PR-Apple-Homekit-SRP-fix' set up to track 'origin/PR-Apple-Homekit-SRP-fix'.
Switched to a new branch 'PR-Apple-Homekit-SRP-fix'
Cloning into 'esp32-homekit-demo-gojimmypi'...
remote: Enumerating objects: 4829, done.
remote: Counting objects: 100% (463/463), done.
remote: Compressing objects: 100% (316/316), done.
remote: Total 4829 (delta 212), reused 238 (delta 143), pack-reused 4366
Receiving objects: 100% (4829/4829), 40.91 MiB | 18.90 MiB/s, done.
Resolving deltas: 100% (1891/1891), done.
Submodule 'components/esp32-button' (https://AchimPieters@github.com/AchimPieters/esp32-button.git) registered for path 'components/esp32-button'
Submodule 'components/esp32-library' (https://AchimPieters@github.com/UncleRus/esp-idf-lib.git) registered for path 'components/esp32-library'
Submodule 'components/homekit' (https://AchimPieters@github.com/AchimPieters/esp32-homekit.git) registered for path 'components/homekit'
Cloning into '/Users/achimpieters/test-homekit/esp32-homekit-demo-gojimmypi/components/esp32-button'...
remote: Enumerating objects: 45, done.        
remote: Counting objects: 100% (45/45), done.        
remote: Compressing objects: 100% (41/41), done.        
remote: Total 45 (delta 18), reused 19 (delta 3), pack-reused 0        
Receiving objects: 100% (45/45), 105.84 KiB | 1.86 MiB/s, done.
Resolving deltas: 100% (18/18), done.
Cloning into '/Users/achimpieters/test-homekit/esp32-homekit-demo-gojimmypi/components/esp32-library'...
remote: Enumerating objects: 13128, done.        
remote: Counting objects: 100% (1783/1783), done.        
remote: Compressing objects: 100% (327/327), done.        
remote: Total 13128 (delta 1513), reused 1555 (delta 1447), pack-reused 11345        
[... 208 lines cut ...]
Call Stack (most recent call first):
  /project/esp32-homekit-demo-gojimmypi/components/wolfssl/CMakeLists.txt:182 (FIND_WOLFSSL_DIRECTORY)

-- Configuring incomplete, errors occurred!
See also "/project/esp32-homekit-demo-gojimmypi/examples/led/build/CMakeFiles/CMakeOutput.log".
cmake failed with exit code 1
root@7ee758f7e4a3:/project/esp32-homekit-demo-gojimmypi/examples/led# 

```

As you can see the error:

```
-- Starting FIND_WOLFSSL_DIRECTORY
CMake Error at /project/esp32-homekit-demo-gojimmypi/components/wolfssl/CMakeLists.txt:100 (message):
  WOLFSSL_ROOT Environment Variable not defined and wolfssl path not found in
  any parent.
Call Stack (most recent call first):
  /project/esp32-homekit-demo-gojimmypi/components/wolfssl/CMakeLists.txt:182 (FIND_WOLFSSL_DIRECTORY)

```

So I looked at 'set(WOLFSSL_ROOT "c:/workspace/wolfssl-gojimmypi")' You use 'c:/.....' That doesnt work on my mac and tried to change it to several paths but without anny result...

There has to be an easier way, as said here above 'because I've built this repro for beginners', Just say your reply, gonna read that now ;)

gojimmypi, 2024-05-05:
> There has to be an easier way, as said here above 'because I've built this repro for beginners'

Oh, and as for that: yes, I completely agree. I had not expected you to merge the changes without testing. For an interim solution, I think setting the environment variable for the location of wolfSSL source code is fairly straightforward for beginners. 

```
export WOLFSSL_ROOT=/path/to/wolfssl/
```

I've confirmed that simply commenting out the `set(WOLFSSL_ROOT "c:/workspace/wolfssl-gojimmypi")` also works (instead of setting a path in the cmake file) as long as wolfSSL and _my fork_  of `esp32-homekit-demo` are both cloned from the same directory (but in your case, wolfssl not named with the `-gojimmypi` suffix).

The cmake file will automatically find `wolfssl`, `wolfssl-$USER`, or `wolfSSL-master` in the parent directories of an example.

I've updated my branch [commenting out that path](https://github.com/gojimmypi/esp32-homekit-demo/commit/eb85553d7ecdaa9a614d290ed93f2925fe3089b4
).

Once my wolfSSL PR https://github.com/wolfSSL/wolfssl/pull/7505 gets merged and there's a new, official stable release (probably in a couple of months), the [wofssl Managed Component](https://components.espressif.com/components/wolfssl/wolfssl) is definitely the way to go, as you stated.

Let's get it confirmed working for you and then we can do some polishing.

AchimPieters, 2024-05-05:
@gojimmypi I have made progress!! 

Here is what I did:

I took my working repro, then I cloned:

```
git clone https://github.com/gojimmypi/wolfssl.git wolfssl-gojimmypi
cd wolfssl-gojimmypi
git checkout PR-Apple-Homekit-SRP-fix
```

Replaced the directories, `examples`, `scr`, `wolfcrypt` and `wolfssl` in my 'wolfssl' directory under components. Then I changed the `user_settings.h` after the changes you made. I also changed the other files, as you did in your pull request.

Then I started compiling as always, I saw these warnings for the ESP32 module:

```
[778/935] Building C object esp-idf/wo...s/__idf_wolfssl.dir/src/ssl_load.c.obj
/project/components/wolfssl/src/ssl_load.c:56:10: warning: #warning ssl_bn.c does not need to be compiled separately from ssl.c [-Wcpp]
   56 |         #warning ssl_bn.c does not need to be compiled separately from ssl.c
      |          ^~~~~~~
[780/935] Building C object esp-idf/wo.../__idf_wolfssl.dir/src/ssl_p7p12.c.obj
/project/components/wolfssl/src/ssl_p7p12.c:37:10: warning: #warning ssl_p7p12.c does not need to be compiled separately from ssl.c [-Wcpp]
   37 |         #warning ssl_p7p12.c does not need to be compiled separately from ssl.c
      |          ^~~~~~~
[781/935] Building C object esp-idf/wo...s/__idf_wolfssl.dir/src/ssl_sess.c.obj
/project/components/wolfssl/src/ssl_sess.c:31:10: warning: #warning ssl_sess.c does not need to be compiled separately from ssl.c [-Wcpp]
   31 |         #warning ssl_sess.c does not need to be compiled separately from ssl.c
      |          ^~~~~~~

```

But it compiled and worked under HomeKit as usual. Then the test for the ESP32C2 and once again the warnings:

```
[704/859] Building C object esp-idf/wo...s/__idf_wolfssl.dir/src/ssl_load.c.obj
/project/components/wolfssl/src/ssl_load.c:56:10: warning: #warning ssl_bn.c does not need to be compiled separately from ssl.c [-Wcpp]
   56 |         #warning ssl_bn.c does not need to be compiled separately from ssl.c
      |          ^~~~~~~
[705/859] Building C object esp-idf/wo.../__idf_wolfssl.dir/src/ssl_p7p12.c.obj
/project/components/wolfssl/src/ssl_p7p12.c:37:10: warning: #warning ssl_p7p12.c does not need to be compiled separately from ssl.c [-Wcpp]
   37 |         #warning ssl_p7p12.c does not need to be compiled separately from ssl.c
      |          ^~~~~~~
[706/859] Building C object esp-idf/wo...s/__idf_wolfssl.dir/src/ssl_sess.c.obj
/project/components/wolfssl/src/ssl_sess.c:31:10: warning: #warning ssl_sess.c does not need to be compiled separately from ssl.c [-Wcpp]
   31 |         #warning ssl_sess.c does not need to be compiled separately from ssl.c
      |          ^~~~~~~
[858/859] Generating binary image from built executable
```

But it compiled and worked under HomeKit!! so progress!

gojimmypi, 2024-05-05:
Hi @AchimPieters !

> it compiled and worked under HomeKit!!

YAY! That's excellent!!

> Replaced the directories, examples, scr, wolfcrypt and wolfssl in my 'wolfssl' directory under components

oof, that's a bit of a brute-force method, but I'm glad it worked for you!  `<happy dance />`

>  I saw these warnings for the ESP32 module [...]

Those warnings are because you didn't replace the `CMakeLists.txt` in your [components/wolfssl](https://github.com/AchimPieters/esp32-homekit-demo/tree/main/components/wolfssl) directory with the [latest version](https://github.com/gojimmypi/esp32-homekit-demo/blob/PR-wolfssl/components/wolfssl/CMakeLists.txt) that was included in my PR. See in particular the [set(COMPONENT_SRCEXCLUDE...](https://github.com/gojimmypi/esp32-homekit-demo/blob/eb85553d7ecdaa9a614d290ed93f2925fe3089b4/components/wolfssl/CMakeLists.txt#L388).

It's no big deal... it's just what it says: some files "_do not need to be compiled separately from ssl.c_". 

I'll refresh my fork from your changes & create a new PR soon.

The important thing is you are using the [changes in wolfSSL PR #7505](https://github.com/wolfSSL/wolfssl/pull/7505/files) (in particular the changes to `esp32_aes.c` and `esp32_sha.c`).... and that _those changes are working for you_!! hooray! Once those changes are merged, I'll refresh the [mywolfssl staging component](https://components-staging.espressif.com/components/gojimmypi/mywolfssl) and we can test with that. We are both in agreement that the Managed Components are by far the easiest to use for the beginner.

Thank you again for all your cooperation. I'm so glad it is working for you. Cheers!

gojimmypi, 2024-05-23:
Hello @AchimPieters -

I've created https://github.com/AchimPieters/esp32-homekit-demo/pull/6 that adds Apple HomeKit support using wolfSSL as a [Managed Component](https://components.espressif.com/components?q=namespace:wolfssl).

Pre-release version `5.7.1-Preview2e` or later must be used.

See: https://components.espressif.com/components/wolfssl/wolfssl/versions/5.7.1-preview2e

I've only updated the LED example. I've included instructions in the README on adding this to other examples. 

Basically just do this from the `main` directory:

```
idf.py add-dependency "wolfssl/wolfssl^5.7.1-preview2e"
```

and ensure the `Enable Apple HomeKit options` is checked with `idf.py menuconfig`, or as seen in in VisualGDB:

[image]

gojimmypi, 2024-06-21:
I'm closing this issue as I believe we have fixed this in https://github.com/wolfSSL/wolfssl/pull/7505 and the latest Managed Component as described above.. 

Please feel free to open a new issue if anything else needs attention.

closed 2024-06-21 by gojimmypi