wolfSSL/wolfssl #7213: How to compile chacha_asm.S for Windows x86_64?

Hippeys, 2024-02-05:
### Version

5.6.6

### Description

my user_settings.h include:
```
#undef  HAVE_CHACHA
#define HAVE_CHACHA

#undef  USE_INTEL_SPEEDUP
#define USE_INTEL_SPEEDUP
```
But when i try to link my app with WolfSSL static lib, compiled with those defines i have errors:

> Severity	Code	Description	Project	File	Line	Suppression State	Details
> Error	LNK2001	unresolved external symbol chacha_encrypt_avx2	E:\WORK\test\out\build\x64-release\test	E:\WORK\test\out\build\x64-release\wolfssl.lib(chacha.obj)	1		
> Error	LNK2001	unresolved external symbol chacha_encrypt_avx1	E:\WORK\test\out\build\x64-release\test	E:\WORK\test\out\build\x64-release\wolfssl.lib(chacha.obj)	1		
> Error	LNK2001	unresolved external symbol chacha_encrypt_x64	E:\WORK\test\out\build\x64-release\test	E:\WORK\test\out\build\x64-release\wolfssl.lib(chacha.obj)	1		
> Error	LNK1120	3 unresolved externals	E:\WORK\test\out\build\x64-release\test	E:\WORK\test\out\build\x64-release\cha_test.exe	1

As i understand this is because chacha_asm.S is not included inside .sln project, if i remove USE_INTEL_SPEEDUP everything is ok, but the thing is i want to use asm speedup for performace. 
Is there any way to compile chacha_asm.S for windows app?

dgarske, 2024-02-05:
Hi @Hippeys ,

Thank you for the report. I am not sure we have the .asm assembly format required for Visual Studio to compile the chacha_asm.S directly. I believe that ChaCha speedup is relatively new and we support it with MinGW64 and GCC.

I'm assigning this ticket to @SparkiDev who can provide details on this.

FYI: Here is an example of how we typically pull in .asm for the Visual Studio project:
https://github.com/wolfSSL/wolfssl/blob/master/wolfssl.vcxproj#L352
However that won't work for this file.

Thanks,
David Garske, wolfSSL

Hippeys, 2024-02-05:
Hi @dgarske 
Thank you for the response. I've tried to build windows x64 version of static library on linux machine(ubuntu) using mingw, here are my steps:
```
sudo mkdir /home/roma/.vs/wolfssl
cd /home/roma/.vs/wolfssl
sudo wget https://github.com/wolfSSL/wolfssl/releases/download/v5.6.6-stable/wolfssl-5.6.6.tar.gz
sudo tar -xvzf wolfssl-5.6.6.tar.gz
cd wolfssl-5.6.6

sudo ./configure --host=x86_64-w64-mingw32 --enable-static=yes --enable-shared=no --enable-intelasm --enable-aesni --enable-sp --enable-sp-asm --enable-errorstrings=no --enable-aesgcm-stream --enable-aesctr  --enable-sys-ca-certs=no --enable-crypttests=no --enable-examples=no --enable-supportedcurves=no --enable-filesystem=no --enable-base64encode=no --enable-asn-print=no --enable-dh=no --enable-oaep=no --enable-brainpool=no --enable-ecc=no --enable-eccshamir=no --enable-benchmark=no --enable-aescbc=no --enable-pwdbased=no --enable-webserver=no --enable-webclient=no --enable-dtls=no --enable-dtls-mtu=no --enable-tls13-draft18=no --enable-tls13=no --enable-quic=no --enable-postauth=no --enable-leantls=no  --enable-leanpsk=no --enable-tlsv12=no --enable-cryptonly  --enable-sha224=no --enable-sha3=no --enable-sha512=no --enable-sha384=no  --prefix="/home/roma/.vs/wolfssl/linux_mini"

sudo make
```

but there are many errors when build, please watch output.log in attachment
[output.log](https://github.com/wolfSSL/wolfssl/files/14168457/output.log)

Hippeys, 2024-02-05:
i have also tried to build chacha_asm.S in Visual Studio 2022 using [yasm plugin ](https://github.com/ShiftMediaProject/VSYASM), with minor changes. It compiles, but when application calls chacha asm functions it crashes, i think it is because of different calling convention and stack management between windows x64 and linux x64

dgarske, 2024-02-05:
Hi @Hippeys ,

I am able to reproduce on Win 10 with MSYS64. I am not able to reproduce on Ubuntu 22.

The compiler does not like the Intel .S assembly syntax around things like:

```
.type	Transform_Sha256_SSE2_Sha,@function
.size	Transform_Sha256_SSE2_Sha,.-Transform_Sha256_SSE2_Sha
```

We will investigate getting `--enable-itelasm --enable-aesni` working on MSYS64.

Thanks,
David Garske, wolfSSL

dgarske, 2024-02-05:
Hi @Hippeys ,

Can you tell us more about your project?

Thanks,
David Garske, wolfSSL

Hippeys, 2024-02-05:
Hi @dgarske,

I want to use it for my personal server, want to move from openssl because of huge bloat using evp, also WolfSSL's chacha20 much more performant for small input then openssl on linux using asm.
I made same for MSYS64, result pretty much the same as on ubuntu, i commented all .size and .type lines in .S files, it compiles, but when i try to use compiled .lib i have these errors:

> Severity	Code	Description	Project	File	Line	Suppression State	Details
> Error	LNK2001	unresolved external symbol ___chkstk_ms	E:\projects\common_new	E:\projects\common_new\wolfssl.lib(src_libwolfssl_la-pwdbased.o)	1		
> Error	LNK2001	unresolved external symbol ___chkstk_ms	E:\projects\common_new	E:\projects\common_new\wolfssl.lib(src_libwolfssl_la-sp_int.o)	1		
> Error	LNK2001	unresolved external symbol ___chkstk_ms	E:\projects\common_new	E:\projects\common_new\wolfssl.lib(src_libwolfssl_la-rsa.o)	1		
> Error	LNK2001	unresolved external symbol ___chkstk_ms	E:\projects\common_new	E:\projects\common_new\wolfssl.lib(src_libwolfssl_la-asn.o)	1		
> Error	LNK2001	unresolved external symbol ___chkstk_ms	E:\projects\common_new	E:\projects\common_new\wolfssl.lib(src_libwolfssl_la-sp_x86_64.o)	1		
> Error	LNK2001	unresolved external symbol chacha_encrypt_avx1@plt	E:\projects\common_new	E:\projects\common_new\wolfssl.lib(src_libwolfssl_la-chacha_asm.o)	1		
> Error	LNK2001	unresolved external symbol Transform_Sha256_AVX2@plt	E:\projects\common_new	E:\projects\common_new\wolfssl.lib(src_libwolfssl_la-sha256_asm.o)	1		
> Error	LNK2001	unresolved external symbol Transform_Sha256_AVX2_RORX@plt	E:\projects\common_new	E:\projects\common_new\wolfssl.lib(src_libwolfssl_la-sha256_asm.o)	1		
> Error	LNK2019	unresolved external symbol __imp__beginthreadex referenced in function wolfSSL_NewThread	E:\projects\common_new	E:\projects\common_new\wolfssl.lib(src_libwolfssl_la-wc_port.o)	1		
> Error	LNK2001	unresolved external symbol sp_2048_mul_16@plt	E:\projects\common_new	E:\projects\common_new\wolfssl.lib(src_libwolfssl_la-sp_x86_64_asm.o)	1		
> Error	LNK2001	unresolved external symbol sp_2048_mul_avx2_16@plt	E:\projects\common_new	E:\projects\common_new\wolfssl.lib(src_libwolfssl_la-sp_x86_64_asm.o)	1		
> Error	LNK2001	unresolved external symbol sp_2048_sqr_16@plt	E:\projects\common_new	E:\projects\common_new\wolfssl.lib(src_libwolfssl_la-sp_x86_64_asm.o)	1		
> Error	LNK2001	unresolved external symbol sp_2048_sqr_avx2_16@plt	E:\projects\common_new	E:\projects\common_new\wolfssl.lib(src_libwolfssl_la-sp_x86_64_asm.o)	1		
> Error	LNK2001	unresolved external symbol sp_2048_cond_sub_16@plt	E:\projects\common_new	E:\projects\common_new\wolfssl.lib(src_libwolfssl_la-sp_x86_64_asm.o)	1		
> Error	LNK2001	unresolved external symbol sp_2048_cond_sub_32@plt	E:\projects\common_new	E:\projects\common_new\wolfssl.lib(src_libwolfssl_la-sp_x86_64_asm.o)	1		
> Error	LNK2001	unresolved external symbol sp_3072_mul_12@plt	E:\projects\common_new	E:\projects\common_new\wolfssl.lib(src_libwolfssl_la-sp_x86_64_asm.o)	1		
> Error	LNK2001	unresolved external symbol sp_3072_mul_avx2_12@plt	E:\projects\common_new	E:\projects\common_new\wolfssl.lib(src_libwolfssl_la-sp_x86_64_asm.o)	1		
> Error	LNK2001	unresolved external symbol sp_3072_mul_24@plt	E:\projects\common_new	E:\projects\common_new\wolfssl.lib(src_libwolfssl_la-sp_x86_64_asm.o)	1		
> Error	LNK2001	unresolved external symbol sp_3072_mul_avx2_24@plt	E:\projects\common_new	E:\projects\common_new\wolfssl.lib(src_libwolfssl_la-sp_x86_64_asm.o)	1		
> Error	LNK2001	unresolved external symbol sp_3072_sqr_12@plt	E:\projects\common_new	E:\projects\common_new\wolfssl.lib(src_libwolfssl_la-sp_x86_64_asm.o)	1		
> Error	LNK2001	unresolved external symbol sp_3072_sqr_avx2_12@plt	E:\projects\common_new	E:\projects\common_new\wolfssl.lib(src_libwolfssl_la-sp_x86_64_asm.o)	1		
> Error	LNK2001	unresolved external symbol sp_3072_sqr_24@plt	E:\projects\common_new	E:\projects\common_new\wolfssl.lib(src_libwolfssl_la-sp_x86_64_asm.o)	1		
> Error	LNK2001	unresolved external symbol sp_3072_sqr_avx2_24@plt	E:\projects\common_new	E:\projects\common_new\wolfssl.lib(src_libwolfssl_la-sp_x86_64_asm.o)	1		
> Error	LNK2001	unresolved external symbol sp_3072_cond_sub_24@plt	E:\projects\common_new	E:\projects\common_new\wolfssl.lib(src_libwolfssl_la-sp_x86_64_asm.o)	1		
> Error	LNK2001	unresolved external symbol sp_3072_cond_sub_48@plt	E:\projects\common_new	E:\projects\common_new\wolfssl.lib(src_libwolfssl_la-sp_x86_64_asm.o)	1		
> Error	LNK2001	unresolved external symbol sp_2048_mul_32@plt	E:\projects\common_new	E:\projects\common_new\wolfssl.lib(src_libwolfssl_la-sp_x86_64_asm.o)	1		
> Error	LNK2001	unresolved external symbol sp_2048_mul_avx2_32@plt	E:\projects\common_new	E:\projects\common_new\wolfssl.lib(src_libwolfssl_la-sp_x86_64_asm.o)	1		
> Error	LNK2001	unresolved external symbol sp_2048_sqr_32@plt	E:\projects\common_new	E:\projects\common_new\wolfssl.lib(src_libwolfssl_la-sp_x86_64_asm.o)	1		
> Error	LNK2001	unresolved external symbol sp_2048_sqr_avx2_32@plt	E:\projects\common_new	E:\projects\common_new\wolfssl.lib(src_libwolfssl_la-sp_x86_64_asm.o)	1		
> Error	LNK2001	unresolved external symbol sp_4096_cond_sub_64@plt	E:\projects\common_new	E:\projects\common_new\wolfssl.lib(src_libwolfssl_la-sp_x86_64_asm.o)	1		
> Error	LNK1120	26 unresolved externals	E:\projects\common_new	E:\projects\common_new\cha_test.exe	1

SparkiDev, 2024-02-07:
Hi @Hippeys

You appear to be compiling with MINGW but are you compiling the library into a MSVC application?

If you want MSVC assembly files for ChaCha20 and Poly1305, we have untested versions.
Let me know if this is something you would like to finish developing.

Thanks,
Sean

Hippeys, 2024-02-07:
Hi @SparkiDev 

Yes, i tried to use somehow asm chacha20 implementation for windows, tried to use mingw winsys etc.
My final solution with Visual Studio: 
1. copied chacha_asm.S to [chacha_asm_win.S](https://github.com/Hippeys/wolfssl/blob/master/wolfcrypt/src/chacha_asm_win.S). 
2. inside chacha_asm_win.S. removed all unix stuff, added prologue/epilogue which converts windows x64 calling convention to linux like that:
```
        # save rdi and rsi
        pushq %rdi
        pushq %rsi

        # convert to windows call convention
        movq %rcx, %rdi
        movq %rdx, %rsi
        movq %r8, %rdx
        movq %r9, %rcx 

       # original function body

        #restoring rdi and rsi
        popq %rsi
        popq %rdi

        repz retq
```
3. installed [YASM plugin](https://github.com/ShiftMediaProject/VSYASM) for Visual Studio 2022
4. added chacha_asm_win.S into .sln, set gas parser for this file

Now it works ok, the huge pro for this solution is that you dont need to keep 2 implementations, at&t and masm, but obviously you would need yasm compiler and plugin to compile it.
Now chacha20 flies like a rocket :). Strangely on windows it works like 15-20% faster then linux, i thought i made some mistake, but i performed test: chacha_asm_win.S encrypts, openssl decrypts random bytes arrays, seems ok...

SparkiDev, 2024-02-08:
Hi Hippeys,

Great to hear you got it working!
The MSVC version is generated using the same scripts as generates the Unix version.
Let me know if you want to try it out.

Thanks,
Sean

Hippeys, 2024-03-05:
Hi @SparkiDev 

Of course, will be glad to try 👍 
Does it include both x64 and x86 versions?

Thank you

SparkiDev, 2024-03-12:
Hi @Hippeys,

I've put up a pull request that has the MSVC assembly files in it.
To use the assembly files you need to change IDE/WIN/user_settings.h and turn on the defines:
    USE_INTEL_CHACHA_SPEEDUP
    USE_INTEL_POLY1305_SPEEDUP

Only x64 assembly and it hasn't been tested.
Let me know what happens!

Sean

Hippeys, 2024-03-12:
Hi @SparkiDev 
I compiled with this code and definitions but have errors when linking:

> Severity	Code	Description	Project	File	Line	Suppression State	Details
> Error	LNK2001	unresolved external symbol chacha_encrypt_avx2	\aacommon_new	\wolfssl.lib(chacha.obj)	1		
> Error	LNK2001	unresolved external symbol chacha_encrypt_avx1	\aacommon_new	\wolfssl.lib(chacha.obj)	1		
> Error	LNK2001	unresolved external symbol chacha_encrypt_x64	\aacommon_new	\wolfssl.lib(chacha.obj)	1		
> Error	LNK2001	unresolved external symbol poly1305_setkey_avx	\aacommon_new	\wolfssl.lib(poly1305.obj)	1		
> Error	LNK2001	unresolved external symbol poly1305_block_avx	\aacommon_new	\wolfssl.lib(poly1305.obj)	1		
> Error	LNK2001	unresolved external symbol poly1305_setkey_avx2	\aacommon_new	\wolfssl.lib(poly1305.obj)	1		
> Error	LNK2001	unresolved external symbol poly1305_calc_powers_avx2	\aacommon_new	\wolfssl.lib(poly1305.obj)	1		
> Error	LNK2001	unresolved external symbol poly1305_final_avx2	\aacommon_new	\wolfssl.lib(poly1305.obj)	1		
> Error	LNK2001	unresolved external symbol poly1305_blocks_avx2	\aacommon_new	\wolfssl.lib(poly1305.obj)	1		
> Error	LNK2001	unresolved external symbol poly1305_final_avx	\aacommon_new	\wolfssl.lib(poly1305.obj)	1		
> Error	LNK2001	unresolved external symbol poly1305_blocks_avx	\aacommon_new	\wolfssl.lib(poly1305.obj)	1		
> Error	LNK1120	11 unresolved externals	\aacommon_new	\test.exe	1	

For example i checked chacha_asm.asm and as i see inside it WOLFSSL_X86_64_BUILD is not defined, if i remove IFDEF WOLFSSL_X86_64_BUILD and corresponding ENDIF - many compilation errors regarding registers etc.

SparkiDev, 2024-05-01:
PR has been updated to work.

Need to define:
    HAVE_CHACHA
    HAVE_POLY1305
    USE_INTEL_CHACHA_SPEEDUP
    USE_INTEL_POLY1305_SPEEDUP
    HAVE_CPUID
    HAVE_CPUID_INTEL

PR #7319 (2024-05-08): ChaCha20, Poly1305 ASM for MSVC
# Description

Make ChaCha20 and Poly1305 asm available for MSVC.

Fixes #7213

# Testing

Linux ASM still works.

# Checklist

 - [ ] added tests
 - [ ] updated/added doxygen
 - [ ] updated appropriate READMEs
 - [ ] Updated manual and documentation
closed 2024-05-08 by douzzer