wolfSSL/wolfssl #7244: [Bug]: `make check` fails on OpenBSD 7.4

job, 2024-02-13:
### Version

5.6.6

### Description

The following steps work:
```
$ ./autogen.sh
$ ./configure
$ make
```

However, this seems to fail:

```
feather$ make check
make -j7  check-recursive
make -j7   testsuite/testsuite.test   scripts/resume.test          scripts/tls13.test scripts/external.test  scripts/google.test  scripts/openssl.test scripts/ocsp.test  scripts/unit.test scripts/pem.test
`testsuite/testsuite.test' is up to date.
`scripts/resume.test' is up to date.
`scripts/tls13.test' is up to date.
`scripts/external.test' is up to date.
`scripts/google.test' is up to date.
`scripts/openssl.test' is up to date.
`scripts/ocsp.test' is up to date.
`scripts/unit.test' is up to date.
`scripts/pem.test' is up to date.
make -j7  check-TESTS
FAIL: scripts/resume.test
FAIL: scripts/external.test
FAIL: scripts/google.test
FAIL: scripts/openssl.test
FAIL: scripts/tls13.test
FAIL: scripts/pem.test
SKIP: scripts/ocsp.test
PASS: testsuite/testsuite.test
PASS: scripts/unit.test
============================================================================
Testsuite summary for wolfssl 5.6.6
============================================================================
# TOTAL: 9
# PASS:  2
# SKIP:  1
# XFAIL: 0
# FAIL:  6
# XPASS: 0
# ERROR: 0
============================================================================
See ./test-suite.log
Please report to https://github.com/wolfssl/wolfssl/issues
============================================================================
*** Error 1 in . (Makefile:8272 'test-suite.log': @case $- in *e*) set +e;; esac; {  mgn= red= grn= lgn= blu= brg= std=;  am__color_tests=no...)
*** Error 2 in . (Makefile:8283 'check-TESTS')
*** Error 2 in . (Makefile:8522 'check-am')
*** Error 1 in . (Makefile:8080 'check-recursive')
*** Error 2 in /home/job/source/wolfssl (Makefile:8524 'check')
```

### Reproduction steps

`$ make check`

### Relevant log output

```shell
=====================================
   wolfssl 5.6.6: ./test-suite.log
=====================================

# TOTAL: 9
# PASS:  2
# SKIP:  1
# XFAIL: 0
# FAIL:  6
# XPASS: 0
# ERROR: 0

.. contents:: :depth: 2

FAIL: scripts/resume
====================

./build-aux/test-driver[112]: ./scripts/resume.test: No such file or directory
FAIL scripts/resume.test (exit status: 1)

FAIL: scripts/tls13
===================

./build-aux/test-driver[112]: ./scripts/tls13.test: No such file or directory
FAIL scripts/tls13.test (exit status: 1)

FAIL: scripts/external
======================

./build-aux/test-driver[112]: ./scripts/external.test: No such file or directory
FAIL scripts/external.test (exit status: 1)

FAIL: scripts/google
====================

./build-aux/test-driver[112]: ./scripts/google.test: No such file or directory
FAIL scripts/google.test (exit status: 1)

FAIL: scripts/openssl
=====================

./build-aux/test-driver[112]: ./scripts/openssl.test: No such file or directory
FAIL scripts/openssl.test (exit status: 1)

SKIP: scripts/ocsp
==================

SNI is on, proceed with globalsign test

./scripts/ocsp.test[50]: ./scripts/ping.test: No such file or directory
./scripts/ocsp.test[61]: ./scripts/ping.test: No such file or directory

Both OCSP connection to globalsign and google failed
Test Failed!
SKIP scripts/ocsp.test (exit status: 77)

FAIL: scripts/pem
=================

./build-aux/test-driver[112]: ./scripts/pem.test: No such file or directory
FAIL scripts/pem.test (exit status: 1)
```

embhorn, 2024-02-13:
Hi @job 

Could you please try configuring with `--disable-shared` and retrying `make check`?

Is this from a clean clone of the repository?

job, 2024-02-13:
This is a clean clone.

`./configure --disable-shared` didn't solve it:

```
feather$ make check
make -j7  check-recursive
make -j7   testsuite/testsuite.test   scripts/resume.test          scripts/tls13.test scripts/external.test  scripts/google.test  scripts/openssl.test   scripts/unit.test scripts/pem.test
`testsuite/testsuite.test' is up to date.
`scripts/resume.test' is up to date.
`scripts/tls13.test' is up to date.
`scripts/external.test' is up to date.
`scripts/google.test' is up to date.
`scripts/openssl.test' is up to date.
`scripts/unit.test' is up to date.
`scripts/pem.test' is up to date.
make -j7  check-TESTS
FAIL: scripts/google.test
FAIL: scripts/openssl.test
FAIL: scripts/external.test
FAIL: scripts/tls13.test
FAIL: scripts/resume.test
FAIL: scripts/pem.test
PASS: testsuite/testsuite.test
PASS: scripts/unit.test
============================================================================
Testsuite summary for wolfssl 5.6.6
============================================================================
# TOTAL: 8
# PASS:  2
# SKIP:  0
# XFAIL: 0
# FAIL:  6
# XPASS: 0
# ERROR: 0
============================================================================
See ./test-suite.log
Please report to https://github.com/wolfssl/wolfssl/issues
============================================================================
*** Error 1 in . (Makefile:8272 'test-suite.log': @case $- in *e*) set +e;; esac; {  mgn= red= grn= lgn= blu= brg= std=;  am__color_tests=no...)
*** Error 2 in . (Makefile:8283 'check-TESTS')
*** Error 2 in . (Makefile:8522 'check-am')
*** Error 1 in . (Makefile:8080 'check-recursive')
*** Error 2 in /home/job/source/wolfssl (Makefile:8524 'check')
```

dgarske, 2024-02-13:
Hi @job ,

Does the `/bin/bash` file exist? I'm guessing the scripts are failing because bash isn't found. All our scripts use `#!/bin/bash`. I suspect yours is at `#!/usr/local/bin/bash`. 

If you replace some of those failing scripts with `#!/usr/local/bin/bash` or `#!/usr/bin/env bash` does it solve the issue?

Thanks,
David Garske, wolfSSL

embhorn, 2024-02-23:
@job 

Did David's suggestion resolve the issue?