wolfSSL/wolfssl #7258: [Bug]: ERROR -308: failed to connect to wolfSSL -1

1756816846, 2024-02-19:
### Version

wolfssl-5.6.6

### Description

ERROR -308: failed to connect to wolfSSL -1
The target port is open normally

### Reproduction steps

`//

#include <stdlib.h>
#include <stdio.h>
#include <string.h>

/* socket includes */
#include <sys/socket.h>
#include <arpa/inet.h>
#include <netinet/in.h>
#include <unistd.h>
#include <fcntl.h>
#include <errno.h>

/* wolfSSL */
#include <wolfssl/options.h>
#include <wolfssl/ssl.h>
#include <wolfssl/wolfio.h>

#define DEFAULT_PORT 21152
#define CONNECT_WAIT_SEC 2
#define SELECT_WAIT_SEC 1
#define SOCKET_INVALID -1
typedef int SOCKET_T;
enum {
    TEST_SELECT_FAIL,
    TEST_TIMEOUT,
    TEST_RECV_READY,
    TEST_SEND_READY,
    TEST_ERROR_READY
};

static int tcp_select(SOCKET_T socketfd, int to_sec, int rx)
{
    fd_set fds, errfds;
    fd_set* recvfds = NULL;
    fd_set* sendfds = NULL;
    SOCKET_T nfds = socketfd + 1;
    struct timeval timeout;
    int result;

    FD_ZERO(&fds);
    FD_SET(socketfd, &fds);
    FD_ZERO(&errfds);
    FD_SET(socketfd, &errfds);

    if (rx)
        recvfds = &fds;
    else
        sendfds = &fds;

    result = select(nfds, recvfds, sendfds, &errfds, &timeout);

    if (result == 0)
        return TEST_TIMEOUT;
    else if (result > 0) {
        if (FD_ISSET(socketfd, &fds)) {
            if (rx)
                return TEST_RECV_READY;
            else
                return TEST_SEND_READY;
        }
        else if (FD_ISSET(socketfd, &errfds))
            return TEST_ERROR_READY;
    }

    return TEST_SELECT_FAIL;
}

int main(int argc, char** argv)
{
    int                ret, err;
    int                sockfd = SOCKET_INVALID;
    struct sockaddr_in servAddr;
    char               buff[256];
    size_t             len;

    /* declare wolfSSL objects */
    WOLFSSL_CTX* ctx = NULL;
    WOLFSSL*     ssl = NULL;

    /* Check for proper calling convention */
    if (argc != 2) {
        printf("usage: %s <IPv4 address>\n", argv[0]);
        ret = 0;
        goto exit;
    }

    /* Initialize wolfSSL */
    wolfSSL_Init();
    //wolfSSL_Debugging_ON();

    /* Create a socket that uses an internet IPv4 address,
     * Sets the socket to be stream based (TCP),
     * 0 means choose the default protocol. */
    if ((sockfd = socket(AF_INET, SOCK_STREAM, 0)) == -1) {
        fprintf(stderr, "ERROR: failed to create the socket\n");
        ret = -1;
        goto exit;
    }

    /* Set the socket options to use nonblocking I/O */
    if (fcntl(sockfd, F_SETFL, O_NONBLOCK) == -1) {
        fprintf(stderr, "ERROR: failed to set non-blocking\n");
        ret = -1;
        goto exit;
    }

    /* Create and initialize WOLFSSL_CTX */
    if ((ctx = wolfSSL_CTX_new(wolfTLSv1_2_client_method())) == NULL) {
        fprintf(stderr, "ERROR: failed to create WOLFSSL_CTX\n");
        ret = -1;
        goto exit;
    }

    wolfSSL_CTX_set_verify(ctx, SSL_VERIFY_NONE, 0);

    /* Initialize the server address struct with zeros */
    memset(&servAddr, 0, sizeof(servAddr));

    /* Fill in the server address */
    servAddr.sin_family = AF_INET;             /* using IPv4      */
    servAddr.sin_port   = htons(DEFAULT_PORT); /* on DEFAULT_PORT */

    /* Get the server IPv4 address from the command line call */
    if (inet_pton(AF_INET, argv[1], &servAddr.sin_addr) != 1) {
        fprintf(stderr, "ERROR: invalid Address\n");
        ret = -1;
        goto exit;
    }

    /* Connect to the server */
    while (connect(sockfd, (struct sockaddr*) &servAddr, sizeof(servAddr))
           == -1) {
        if (errno == EAGAIN || errno == EWOULDBLOCK) {
            /* non-blocking connect, wait for read activity on socket */
            tcp_select(sockfd, CONNECT_WAIT_SEC, 1);
            continue;
        }
        else if (errno == EINPROGRESS || errno == EALREADY) {
            break;
        }
        fprintf(stderr, "ERROR: failed to connect %d\n\n", errno);
        ret = -1;
        goto exit;
    }

    /* Create a WOLFSSL object */
    if ((ssl = wolfSSL_new(ctx)) == NULL) {
        fprintf(stderr, "ERROR: failed to create WOLFSSL object\n");
        ret = -1;
        goto exit;
    }

    /* Attach wolfSSL to the socket */
    if ((ret = wolfSSL_set_fd(ssl, sockfd)) != WOLFSSL_SUCCESS) {
        fprintf(stderr, "ERROR: Failed to set the file descriptor\n");
        goto exit;
    }

    /* Connect to wolfSSL on the server side */
    do {
        ret = wolfSSL_connect(ssl);
        err = wolfSSL_get_error(ssl, ret);
        if (err == WOLFSSL_ERROR_WANT_READ)
            tcp_select(sockfd, SELECT_WAIT_SEC, 1);
    } while (err == WOLFSSL_ERROR_WANT_READ || err == WOLFSSL_ERROR_WANT_WRITE);

    if (ret != WOLFSSL_SUCCESS){
        fprintf(stderr, "ERROR %d: failed to connect to wolfSSL %d\n", err, ret);
        goto exit;
    }

    /* Get a message for the server from stdin */
    printf("Message for server: ");
    memset(buff, 0, sizeof(buff));
    if (fgets(buff, sizeof(buff), stdin) == NULL) {
        fprintf(stderr, "ERROR: failed to get message for server\n");
        ret = -1;
        goto exit;
    }
    len = strnlen(buff, sizeof(buff));

    /* Send the message to the server */
    do {
        ret = wolfSSL_write(ssl, buff, len);
        err = wolfSSL_get_error(ssl, ret);
    }
    while (err == WOLFSSL_ERROR_WANT_WRITE);
    if (ret < 0) {
        fprintf(stderr, "ERROR %d: failed to write\n", ret);
        goto exit;
    }

    /* Read the reply from server */
    memset(buff, 0, sizeof(buff));
    do {
        ret = wolfSSL_read(ssl, buff, sizeof(buff)-1);
        err = wolfSSL_get_error(ssl, ret);
    } while (err == WOLFSSL_ERROR_WANT_READ);
    if (ret < 0) {
        fprintf(stderr, "ERROR %d: failed to read\n", ret);
        goto exit;
    }

    /* Print to stdout any data the server sends */
    printf("Server: %s\n", buff);

    /* send close notify */
    do {
        ret = wolfSSL_shutdown(ssl);
        err = wolfSSL_get_error(ssl, 0);
        if (err == WOLFSSL_ERROR_WANT_READ)
            tcp_select(sockfd, SELECT_WAIT_SEC, 1);
    } while (ret == WOLFSSL_ERROR_WANT_READ ||
             ret == WOLFSSL_ERROR_WANT_WRITE);
    printf("Shutdown complete\n");

    ret = 0;

    exit:
    /* Cleanup and return */
    if (ssl)
        wolfSSL_free(ssl);      /* Free the wolfSSL object              */
    if (sockfd != SOCKET_INVALID)
        close(sockfd);          /* Close the socket   */
    if (ctx)
        wolfSSL_CTX_free(ctx);  /* Free the wolfSSL context object          */
    wolfSSL_Cleanup();          /* Cleanup the wolfSSL environment          */

    return ret;                 /* Return reporting a success               */
}
`

1756816846, 2024-02-19:
[image]

1756816846, 2024-02-19:
[image]

kareem-wolfssl, 2024-02-19:
Hi @1756816846 ,

Do you have anything else open on port 21152?
Please rebuild wolfSSL with --enable-debug, run wolfSSL_Debugging_ON() at the start of your program and attach debug logs here.

1756816846, 2024-02-19:
> 你好@1756816846,
> 
> 您在端口 21152 上还有其他开放的东西吗？ 请使用--enable-debug重建wolfSSL，在程序开始时运行wolfSSL_Debugging_ON()并在此处附加调试日志。

─# ./demo 172.16.0.107
wolfSSL Entering TLSv1_2_client_method_ex
wolfSSL Entering wolfSSL_CTX_new_ex
wolfSSL Entering wolfSSL_CertManagerNew
wolfSSL Leaving WOLFSSL_CTX_new, return 0
wolfSSL Entering wolfSSL_CTX_set_verify
wolfSSL Entering SSL_new
wolfSSL Leaving SSL_new, return 0
wolfSSL Entering SSL_set_fd
wolfSSL Entering SSL_set_read_fd
wolfSSL Leaving SSL_set_read_fd, return 1
wolfSSL Entering SSL_set_write_fd
wolfSSL Leaving SSL_set_write_fd, return 1
wolfSSL Entering SSL_connect()
wolfSSL Entering SendClientHello
Adding signature algorithms extension
growing output buffer
Signature Algorithms extension to write
Point Formats extension to write
Supported Groups extension to write
Encrypt-Then-Mac extension to write
EMS extension to write
Your IO Send callback is null, please set
wolfSSL Leaving SendClientHello, return -308
wolfSSL error occurred, error = -308
wolfSSL Entering SSL_get_error
wolfSSL Leaving SSL_get_error, return -308
ERROR -308: failed to connect to wolfSSL -1
wolfSSL Entering SSL_free
Free'ing client ssl
Shrinking output buffer
wolfSSL Entering ClientSessionToSession
CTX ref count not 0 yet, no free
wolfSSL Leaving SSL_free, return 0
wolfSSL Entering SSL_CTX_free
CTX ref count down to 0, doing full free
wolfSSL Entering wolfSSL_CertManagerFree
wolfSSL Leaving SSL_CTX_free, return 0
wolfSSL Entering wolfSSL_Cleanup
wolfSSL Entering wolfCrypt_Cleanup

kareem-wolfssl, 2024-02-19:
` Your IO Send callback is null, please set`
You need to set an IO send callback using `wolfSSL_SetIOSend`.  Check out our example here: https://github.com/wolfSSL/wolfssl-examples/blob/master/tls/client-tls-callback.c

1756816846, 2024-02-19:
Error: failed to connect

Still not working, I used the demo you sent and added/* Set the socket options to use nonblocking I/O*/

If (fcntl (sockfd, F_SETFL, O-NONBLOCK)==-1){

Fprintf (stderr, "Error: failed to set non blocking \ n");

Ret=-1;

Goto exit;

}

1756816846, 2024-02-19:
I have succeeded now

closed 2024-02-20 by 1756816846