wolfSSL/wolfssl #7263: [Bug]: undefined reference to `__aeabi_read_tp'

1756816846, 2024-02-21:
### Version

wolfssl-5.6.6

### Description

  CCLD     wolfcrypt/benchmark/benchmark
wolfcrypt/benchmark/benchmark.o: In function `bench_ecc':
benchmark.c:(.text.bench_ecc+0x98): undefined reference to `__aeabi_read_tp'
benchmark.c:(.text.bench_ecc+0x238): undefined reference to `__aeabi_read_tp'
wolfcrypt/benchmark/benchmark.o: In function `bench_eccMakeKey':
benchmark.c:(.text.bench_eccMakeKey+0x54): undefined reference to `__aeabi_read_tp'
wolfcrypt/benchmark/benchmark.o: In function `bench_dh':
benchmark.c:(.text.bench_dh+0x158): undefined reference to `__aeabi_read_tp'
benchmark.c:(.text.bench_dh+0x318): undefined reference to `__aeabi_read_tp'
wolfcrypt/benchmark/benchmark.o:benchmark.c:(.text.bench_rsa+0xcc): more undefined references to `__aeabi_read_tp' follow

### Reproduction steps

armv5l
download:https://www.uclibc.org/downloads/binaries/0.9.30.1/cross-compiler-armv5l.tar.bz2
./configure --host=arm-linux --prefix=/root/Desktop/build/xcompile/armv5l/ --enable-opensslextra --enable-shared --enable-static --disable-harden --enable-sni --disable-asm --disable-filesystem --disable-examples --disable-oldtls --enable-tls13

1756816846, 2024-02-21:
I used -- disable benchmark to compile Wolfssl successfully, but now there is an error in compiling the demo

└─# /etc/xcompile/armv5l/bin/armv5l-gcc -o demo demo.c -lwolfssl -static   -lpthread -lm -DWC_NO_HARDEN

In file included from /etc/xcompile/armv5l/bin/../include/wolfssl/wolfcrypt/hash.h:41,
                 from /etc/xcompile/armv5l/bin/../include/wolfssl/wolfcrypt/hmac.h:29,
                 from /etc/xcompile/armv5l/bin/../include/wolfssl/openssl/compat_types.h:31,
                 from /etc/xcompile/armv5l/bin/../include/wolfssl/ssl.h:42,
                 from demo.c:40:
/etc/xcompile/armv5l/bin/../include/wolfssl/wolfcrypt/sha512.h:141: error: expected specifier-qualifier-list before 'word64'
In file included from /etc/xcompile/armv5l/bin/../include/wolfssl/openssl/ssl.h:46,
                 from /etc/xcompile/armv5l/bin/../include/wolfssl/openssl/asn1.h:27,
                 from /etc/xcompile/armv5l/bin/../include/wolfssl/ssl.h:4178,
                 from demo.c:40:
/etc/xcompile/armv5l/bin/../include/wolfssl/openssl/crypto.h:71: error: expected ')' before 'opts'
In file included from demo.c:40:
/etc/xcompile/armv5l/bin/../include/wolfssl/ssl.h:4952: error: expected ')' before 'opts'

kareem-wolfssl, 2024-02-22:
Hi @1756816846 ,

It looks like you are linking wolfSSL to your application and seeing build errors there.  You need to include our options.h and settings.h headers before all other wolf/OpenSSL headers in all files which include these headers:
```
#include <wolfssl/options.h>
#include <wolfssl/wolfcrypt/settings.h>
```

I would also recommend removing WC_NO_HARDEN as this define should not be needed.

1756816846, 2024-02-22:
#include <wolfssl/options.h>
#include <wolfssl/wolfcrypt/settings.h>
#include <wolfssl/ssl.h>

/include/wolfssl/wolfcrypt/sha512.h:141: error: expected specifier-qualifier-list before 'word64'

dgarske, 2024-02-22:
Hi @1756816846 ,

The SHA2-512 requires a 64-bit type. Most 32-bit systems support a "long long" or "uint64_t". I suspect your compiler and configure is not properly detecting  "long long" as 64-bit.

You can try adding `CFLAGS="-DSIZEOF_LONG_LONG=8"` to the ./configure.

If that doesn't work please share your `config.log` from ./configure.

Thanks,
David Garske, wolfSSL

1756816846, 2024-02-22:
Yes, you have successfully helped me solve this problem. Thank you

closed 2024-02-22 by 1756816846