wolfSSL/wolfssl #7276: [Bug]: ClientHello's handshake version issue

SmallTown123, 2024-02-27:
### Contact Details

small_town_123@163.com

### Version

5.5.1

### Description

We found that the handshake version field of TLS1.3 can only be 0x0303, and will respond with an Alert message for any other content, is this a compatibility issue with other TLS implementation libraries?

anhu, 2024-02-27:
Hi @SmallTown123 , 

My name is Anthony and I am a member of the wolfSSL team. Please see the RFC for TLS 1.3: https://datatracker.ietf.org/doc/html/rfc8446#section-4.1.2 

In particular, I will quote a specific passage: 

> legacy_version:  In previous versions of TLS, this field was used for
      version negotiation and represented the highest version number
      supported by the client.  Experience has shown that many servers
      do not properly implement version negotiation, leading to "version
      intolerance" in which the server rejects an otherwise acceptable
      ClientHello with a version number higher than it supports.  In
      TLS 1.3, the client indicates its version preferences in the
      "supported_versions" extension ([Section 4.2.1](https://datatracker.ietf.org/doc/html/rfc8446#section-4.2.1)) and the
      legacy_version field MUST be set to 0x0303, which is the version
      number for TLS 1.2.  TLS 1.3 ClientHellos are identified as having
      a legacy_version of 0x0303 and a supported_versions extension
      present with 0x0304 as the highest version indicated therein.
      (See [Appendix D](https://datatracker.ietf.org/doc/html/rfc8446#appendix-D) for details about backward compatibility.)

So this is required to be this way. 

I hope this helps.  Please let me know if you need more clarifications. 

Warm regards, Anthony

anhu, 2024-02-27:
May ask, can you please let us know a bit about yourself and your project?  Is this project academic, professional or personal? We love to know how people are using our software so please feel free to let us know as much as you care to share. 

Warm regards, Anthony

SmallTown123, 2024-02-27:
Hi, Anthony. Thanks for your reply, we are using a deep differential fuzzing framework TLS-DeepDiffer, but sorry the work is not yet published and we are not yet able to provide more detailed information, we will get in touch with you if we have more security findings, thanks.

SmallTown123, 2024-02-27:
After your analysis, we feel that there may be some problems with other TLS implementation libraries in this regard, thanks!

anhu, 2024-02-27:
Excellent.  When you are able to show us your published work, please send a message to "facts at wolfssl.com" and "anthony at wolfssl.com".  I will now proceed to close this issue. 

Warm regards, Anthony

closed 2024-02-27 by anhu

SmallTown123, 2024-02-29:
Dear Anthony,

Still on the issue of the legacy_version field of TLS 1.3 ClientHello message.

I don't know if you have time to read the reply to this issue from OpenSSL. 
https://github.com/openssl/openssl/issues/23702  

And I'm also trying to figure out if the issue poses a potential security threat.
Thank you!

| |
small_town_123
|
|
***@***.***
|
---- Replied Message ----
| From | Anthony ***@***.***> |
| Date | 2/27/2024 23:55 |
| To | ***@***.***> |
| Cc | ***@***.***>,
***@***.***> |
| Subject | Re: [wolfSSL/wolfssl] [Bug]: ClientHello's handshake version issue (Issue #7276) |

Closed #7276 as completed.

anhu, 2024-02-29:
I think wolfSSL and OpenSSL can agree to different approaches.  We follow and are a bit more strict about what the specification says.  OpenSSL team chooses to be a bit more relaxed.  That's fine.