wolfSSL/wolfssl #7316: gpg --verify failure

ler762, 2024-03-11:
### Version

5.6.6

### Description

What did I do wrong?  gpg --verify says bad signature.

download the files:
```
https://github.com/wolfSSL/wolfssl/releases/download/v5.6.6-stable/wolfssl-5.6.6-stable.tar.gz.asc
https://github.com/wolfSSL/wolfssl/releases/download/v5.6.6-stable/wolfssl-5.6.6.tar.gz
```

check the files:
```
$ gpg --verify wolfssl-5.6.6-stable.tar.gz.asc wolfssl-5.6.6.tar.gz
gpg: Signature made Tue Dec 19 17:15:11 2023 EST
gpg:                using RSA key 0xEBC80E415CA29677
gpg: Can't check signature: public key not found

$ gpg --keyserver-options auto-key-retrieve --verify wolfssl-5.6.6-stable.tar.gz.asc wolfssl-5.6.6.tar.gz
gpg: Signature made Tue Dec 19 17:15:11 2023 EST
gpg:                using RSA key 0xEBC80E415CA29677
gpg: requesting key 0xEBC80E415CA29677 from hkps server keys.openpgp.org
gpg: key 0xEBC80E415CA29677: public key "wolfSSL <secure@wolfssl.com>" imported
gpg: Total number processed: 1
gpg:               imported: 1  (RSA: 1)
gpg: BAD signature from "wolfSSL <secure@wolfssl.com>"

$ gpg --list-keys "wolfSSL <secure@wolfssl.com>"
pub   2048R/0xEBC80E415CA29677 2015-01-12 [expires: 2024-12-01]
      Key fingerprint = A2A4 8E7B CB96 C5BE CB98  7314 EBC8 0E41 5CA2 9677
uid                            wolfSSL <secure@wolfssl.com>
uid                            Jacob Barthelmeh (wolfSSL) <secure@wolfssl.com>
sub   2048R/0x71FEBFE6B6A684BC 2015-01-12 [expires: 2024-12-01]
```

JacobBarthelmeh, 2024-03-11:
Hi @ler762 ,

Thank you for using wolfSSL! We added a tar bundle in addition to the auto-generated GitHub tar bundle. The gpg signature was created on wolfssl-5.6.6-stable.tar.gz (downloaded by clicking on `Source code
(tar.gz)` in the Release "Assets".

```
bash-3.2$ gpg --verify ~/Downloads/wolfssl-5.6.6-stable.tar.gz.asc ~/Downloads/wolfssl-5.6.6-stable.tar.gz
gpg: Signature made Wed Dec 20 05:15:11 2023 WIB
gpg:                using RSA key A2A48E7BCB96C5BECB987314EBC80E415CA29677
gpg: Good signature from "wolfSSL <secure@wolfssl.com>" [unknown]
gpg:                 aka "Jacob Barthelmeh (wolfSSL) <secure@wolfssl.com>" [unknown]
gpg: WARNING: This key is not certified with a trusted signature!
gpg:          There is no indication that the signature belongs to the owner.
Primary key fingerprint: A2A4 8E7B CB96 C5BE CB98  7314 EBC8 0E41 5CA2 9677
```

ler762, 2024-03-11:
On Mon, Mar 11, 2024 at 3:48 PM JacobBarthelmeh wrote:

> Hi @ler762 <https://github.com/ler762> ,
>
> Thank you for using wolfSSL! We added a tar bundle in addition to the
> auto-generated GitHub tar bundle. The gpg signature was created on
> wolfssl-5.6.6-stable.tar.gz (downloaded by clicking on Source code
> (tar.gz) in the Release "Assets".
>

yup - that was it:

$ gpg --verify wolfssl-5.6.6-stable.tar.gz.asc wolfssl-5.6.6-stable.tar.gz
gpg: Signature made Tue Dec 19 17:15:11 2023 EST
gpg:                using RSA key 0xEBC80E415CA29677
gpg: Good signature from "wolfSSL ***@***.***>"
gpg:                 aka "Jacob Barthelmeh (wolfSSL) ***@***.***>"
gpg: WARNING: This key is not certified with a trusted signature!
gpg:          There is no indication that the signature belongs to the
owner.
Primary key fingerprint: A2A4 8E7B CB96 C5BE CB98  7314 EBC8 0E41 5CA2 9677

Thank you!

Message ID: ***@***.***>

>

closed 2024-03-11 by JacobBarthelmeh