wolfSSL/wolfssl #7322: [Bug]: Chacha20_poly1305 WolfSSL and OpenSSL produce same cipihertext but different tags

Hippeys, 2024-03-12:
### Version

5.66

### Description

WolfSSL ver: 5.6.6
OpenSSL ver: 3.2.0
Visual Studio x64 app for Windows

Testing WolfSSL's chacha20_poly1305 implementation i discovered that both produce same ciphertext out of same plaintext, using same key and iv, but poly1305 tag are different...

My sample code:
```
unsigned char plain[32];
unsigned char cipher_wolfssl[32];
unsigned char cipher_openssl[32];
unsigned char tag_wolfssl[16];
unsigned char tag_openssl[16];
unsigned char key[32];
unsigned char iv[12];

int openssl_len = 0;
EVP_CIPHER_CTX* octx = EVP_CIPHER_CTX_new();

aa_crypto_random(key, sizeof(key));
aa_crypto_random(iv, sizeof(iv));
aa_crypto_random(plain, sizeof(plain));

// WolfSSL
if (wc_ChaCha20Poly1305_Encrypt(key, iv, NULL, 0, plain, sizeof(plain), cipher_wolfssl, tag_wolfssl) != 0) {
    printf("wc_ChaCha20Poly1305_Encrypt failed!\n");
    return false;
}

// OpenSSL
if (EVP_EncryptInit_ex(octx, EVP_chacha20_poly1305(), NULL, NULL, NULL) != 1) {
    printf("EVP_EncryptInit_ex failed!\n");
    return false;
}

if (EVP_CIPHER_CTX_ctrl(octx, EVP_CTRL_AEAD_SET_IVLEN, sizeof(iv), 0) != 1) {
    printf("EVP_CIPHER_CTX_ctrl failed!\n");
    return false;
}

if (EVP_EncryptInit_ex(octx, NULL, NULL, key, iv) != 1) {
    printf("EVP_EncryptInit_ex failed(2)!\n");
    return false;
}

if (EVP_EncryptUpdate(octx, cipher_openssl, &openssl_len, plain, sizeof(plain)) != 1) {
    printf("EVP_EncryptUpdate failed!\n");
    return false;
}

if (EVP_EncryptFinal_ex(octx, plain + openssl_len, &openssl_len) != 1) {
    printf("EVP_EncryptFinal_ex failed!\n");
    return false;
}

if (EVP_CIPHER_CTX_ctrl(octx, EVP_CTRL_AEAD_GET_TAG, AA_CHACHAPOLY_TAG_SIZE, tag_openssl) != 1) {
    printf("EVP_CIPHER_CTX_ctrl failed(2)!\n");
    return false;
}

if (memcmp(cipher_wolfssl, cipher_openssl, sizeof(cipher_wolfssl) ) != 0 ) {
    printf("cipher_wolfssl != cipher_openssl\n");
    return false;
}

if (!aa_mem_cmp(tag_wolfssl, tag_openssl, sizeof(tag_wolfssl))) {
    printf("tag_wolfssl != tag_openssl");
    return false;
}
```

This code writes "tag_wolfssl != tag_openssl" to console. As you can see ciphertext are both same, only tags are different.

Hippeys, 2024-03-12:
this is also my user_settings.h:
```
#undef  NO_AES
#define NO_AES

#undef  NO_DSA
#define NO_DSA

#undef  HAVE_ECC
#define HAVE_ECC

#undef  HAVE_HASHDRBG
#define HAVE_HASHDRBG

#undef  WOLFSSL_HAVE_SP_ECC
#define WOLFSSL_HAVE_SP_ECC

#undef  WC_NO_ASYNC_THREADING
#define WC_NO_ASYNC_THREADING

#undef  NO_ERROR_STRINGS
#define NO_ERROR_STRINGS

#undef  NO_OLD_TLS
#define NO_OLD_TLS

#undef  WOLFSSL_NO_TLS12
#define WOLFSSL_NO_TLS12

#undef  WC_NO_RSA_OAEP
#define WC_NO_RSA_OAEP

#undef  NO_DH
#define NO_DH

#undef  WOLFSSL_NO_SHAKE128
#define WOLFSSL_NO_SHAKE128

#undef  WOLFSSL_NO_SHAKE256
#define WOLFSSL_NO_SHAKE256

#undef  NO_FILESYSTEM
#define NO_FILESYSTEM

#undef  NO_RC4
#define NO_RC4

#undef  NO_PSK
#define NO_PSK

#undef  NO_MD4
#define NO_MD4

#undef  WOLFCRYPT_ONLY
#define WOLFCRYPT_ONLY

#undef  NO_DES3
#define NO_DES3

#undef  NO_DO178
#define NO_DO178

#undef  NO_KDF
#define NO_KDF

#undef  NO_PWDBASED
#define NO_PWDBASED

#undef  NO_SIG_WRAPPER
#define NO_SIG_WRAPPER

#undef  USE_INTEL_SPEEDUP
#define USE_INTEL_SPEEDUP

#undef  WOLFSSL_HAVE_ATOMIC_H
#define WOLFSSL_HAVE_ATOMIC_H

#undef  HAVE_THREAD_LS
#define HAVE_THREAD_LS

#undef  ERROR_QUEUE_PER_THREAD
#define ERROR_QUEUE_PER_THREAD

#undef  TFM_TIMING_RESISTANT
#define TFM_TIMING_RESISTANT

#undef  ECC_TIMING_RESISTANT
#define ECC_TIMING_RESISTANT

#undef  WC_RSA_BLINDING
#define WC_RSA_BLINDING

#undef  WOLFSSL_USE_ALIGN
#define WOLFSSL_USE_ALIGN

#undef  WOLFSSL_ASN_TEMPLATE
#define WOLFSSL_ASN_TEMPLATE

#undef  HAVE_CHACHA
#define HAVE_CHACHA

#undef  HAVE_POLY1305
#define HAVE_POLY1305

#undef  HAVE_TLS_EXTENSIONS
#define HAVE_TLS_EXTENSIONS

#undef  HAVE_SNI
#define HAVE_SNI

#undef  HAVE_ENCRYPT_THEN_MAC
#define HAVE_ENCRYPT_THEN_MAC

#undef  WOLFSSL_HAVE_SP_RSA
#define WOLFSSL_HAVE_SP_RSA

#undef  WOLFSSL_SP_LARGE_CODE
#define WOLFSSL_SP_LARGE_CODE

#undef  WOLFSSL_SP
#define WOLFSSL_SP

#undef  WOLFSSL_SP_MATH_ALL
#define WOLFSSL_SP_MATH_ALL

#undef  WOLFSSL_SP_X86_64
#define WOLFSSL_SP_X86_64

#undef  GCM_TABLE_4BIT
#define GCM_TABLE_4BIT

#undef  WOLFSSL_TEST_STATIC_BUILD
#define WOLFSSL_TEST_STATIC_BUILD

#undef  HAVE_WC_INTROSPECTION
#define HAVE_WC_INTROSPECTION

#if defined (_WIN64 )
#undef  WOLFSSL_X86_64_BUILD
#define WOLFSSL_X86_64_BUILD

#undef  WOLFSSL_SP_ASM
#define WOLFSSL_SP_ASM

#undef  WOLFSSL_SP_X86_64_ASM
#define WOLFSSL_SP_X86_64_ASM

#undef  WOLFSSL_AESNI
#define WOLFSSL_AESNI
#endif
```

kareem-wolfssl, 2024-03-12:
Thanks for the report and the additional information.  I'm working on reproducing this here.  Will keep you updated.

Hippeys, 2024-03-12:
@kareem-wolfssl yes, on x64 linux, x86 windows all is ok, issue exists when targeting x64 windows.
I will send tag examples soon

kareem-wolfssl, 2024-03-12:
@Hippeys I'm also unable to reproduce on Windows x64, using Visual Studio 2022, wolfSSL 5.6.6 + OpenSSL 3.2.0.
I did have to modify your user_settings.h to not use Chacha/Poly assembly as these aren't supported in Visual Studio:
```
#define NO_CHACHA_ASM
#undef USE_INTEL_SPEEDUP
```
Can you confirm if you are using Chacha/Poly assembly in Visual Studio somehow?  Are you using Clang in VS?

kareem-wolfssl, 2024-03-12:
@Hippeys I was informed about your previous issue for Chacha/Poly assembly on Windows and I see my colleage Sean's PR for it.  I will give it a try here.

Hippeys, 2024-03-12:
@kareem-wolfssl I dont use masm code for chacha20 yet, so you dont need to try this yet. Yes, i tried now to undef USE_INTEL_SPEEDUP and works correctly. But the strange thing is that even if i remove asm code for chacha and poly1305 manually problem still exists. Steps to reproduce:
1. Clone current WolfSSL repository
2. inside sha256.c add line #undef USE_INTEL_SPEEDUP
3. inside poly1305.c add line #undef USE_INTEL_SPEEDUP
4. in user_settings.h add lines from my previous post and also add #define NO_CHACHA_ASM

So this will keep intel speedup for the rest of library, like for sp_x86_64 etc, but will remove asm code from sha256, poly1305 and chacha20 and will copile ok, but still WolfSSL generate different tag than OpenSSL.

kareem-wolfssl, 2024-03-12:
Thanks, I'm able to reproduce the issue here with your instructions.  I'm looking into this with the team.