wolfSSL/wolfssl #7326: How to bundle this lib with FetchContent/ExternalProject to find it later with pkg_check_modules?

Dich0tomy, 2024-03-12:
### Version

v5.6.6-stable

### Description

Hello, I'm using jwt-cpp for JWT tokens inside my application and it expects OpenSSL/LibreSSL/WolfSSL. Bundling each of them seems like a lot of painful work, but wolfSSL seemed the most feasible for me. This is what I'm trying now:

```cmake
# wolfssl
FetchContent_Declare(
	wolfssl
	GIT_REPOSITORY https://github.com/wolfssl/wolfssl
	GIT_TAG        v5.6.6-stable
)

FetchContent_GetProperties( wolfssl )
if(NOT wolfssl_POPULATED)
	FetchContent_Populate(wolfssl)

	set(WOLFSSL_EXAMPLES OFF CACHE INTERNAL "(Dont build wolfssl examples)")
	set(WOLFSSL_CRYPT_TESTS OFF CACHE INTERNAL "(Dont build wolfssl tests)")

	add_subdirectory( ${wolfssl_SOURCE_DIR} ${wolfssl_BINARY_DIR} )
endif()

# jwt-cpp
FetchContent_Declare(
	jwt-cpp
	GIT_REPOSITORY https://github.com/Thalhammer/jwt-cpp
	GIT_TAG        v0.7.0
)

FetchContent_GetProperties( jwt-cpp )
if(NOT jwt-cpp_POPULATED)
	FetchContent_Populate(jwt-cpp)

	set(JWT_BUILD_EXAMPLES OFF CACHE INTERNAL "(Dont build examples of jwt)")
	set(JWT_BUILD_TESTS OFF CACHE INTERNAL "(Dont build jwt tests)")
	set(JWT_BUILD_DOCS OFF CACHE INTERNAL "(Dont build jwt docs)")
	set(JWT_SSL_LIBRARY "wolfSSL" CACHE INTERNAL "(Use wolfSSL as cryptographic backend)")

	add_subdirectory( ${jwt-cpp_SOURCE_DIR} ${jwt-cpp_BINARY_DIR} )
endif()
```

The lib seems to compile fine, but jwt cannot find it in the end. The code responsible for finding looks like this:
```cmake
  find_package(PkgConfig REQUIRED)
  pkg_check_modules(wolfssl REQUIRED IMPORTED_TARGET wolfssl)
  list(TRANSFORM wolfssl_INCLUDE_DIRS APPEND "/wolfssl") # This is required to access OpenSSL compatibility API
```

I'm stuck on this for a few hours now and I cannot figure out how to bundle the lib with FetchContent and if it's even possible to do that sensibly. Bundling OpenSSL or LibreSSL gave me 10x more headache so I'm thinking wolfSSL is the closest to working here.

Any help is greatly appreciated, thanks from advance!

kareem-wolfssl, 2024-03-12:
Hi @Dich0tomy,

Please share the error you're getting from Cmake/jwt.  A full build log would be great if possible.

Dich0tomy, 2024-03-12:
My bad, sorry.

I cannot share the full log unfortunately, but the only relevant information here would be:
```
Generating user options header...
-- BUILD_SHARED_LIBS enabled: wolfcrypt/src/hmac.c;wolfcrypt/src/hash.c;wolfcrypt/src/cpuid.c;wolfcrypt/src/kdf.c;wolfcrypt/src/random.c;wolfcrypt/src/sha256.c;wolfcrypt/src/rsa.c;wolfcrypt/src/sp_int.c;wolfcrypt/src/aes.c;wolfcrypt/src/sha.c;wolfcrypt/src/sha512.c;wolfcrypt/src/sha3.c;wolfcrypt/src/logging.c;wolfcrypt/src/wc_port.c;wolfcrypt/src/error.c;wolfcrypt/src/wc_encrypt.c;wolfcrypt/src/signature.c;wolfcrypt/src/wolfmath.c;wolfcrypt/src/memory.c;wolfcrypt/src/dh.c;wolfcrypt/src/asn.c;wolfcrypt/src/coding.c;wolfcrypt/src/poly1305.c;wolfcrypt/src/md5.c;wolfcrypt/src/pwdbased.c;wolfcrypt/src/pkcs12.c;wolfcrypt/src/chacha.c;wolfcrypt/src/chacha20_poly1305.c;wolfcrypt/src/integer.c;wolfcrypt/src/ecc.c;src/internal.c;src/wolfio.c;src/keys.c;src/ssl.c;src/ocsp.c;src/tls.c;src/tls13.c
-- WOLFSSL_INCLUDE_DIRS is blank. No additional directories will be added.
-- Found PkgConfig: /nix/store/<hash>-pkg-config-wrapper-0.29.2/bin/pkg-config (found version "0.29.2")
-- Checking for module 'wolfssl'
--   No package 'wolfssl' found
CMake Error at /nix/store/<hash>-cmake-3.27.7/share/cmake-3.27/Modules/FindPkgConfig.cmake:607 (message):
  A required package was not found
Call Stack (most recent call first):
  /nix/store/<hash>-cmake-3.27.7/share/cmake-3.27/Modules/FindPkgConfig.cmake:829 (_pkg_check_modules_internal)
  build/_deps/jwt-cpp-src/CMakeLists.txt:61 (pkg_check_modules)
```

And that's it.

Btw. this is what I get during the generation phase run via
```
cmake -B build \
 -S . \
 -G Ninja \
 -DCMAKE_BUILD_TYPE=Debug
```

kareem-wolfssl, 2024-03-12:
Hi @Dich0tomy,

No worries and thanks for the additional information.
Can you rerun CMake with `-DCMAKE_FIND_DEBUG_MODE=ON`?  This will hopefully give more information on which paths CMake is searching in.

Dich0tomy, 2024-03-12:
Hmm I don't think that's very helpful:
```
CMake Debug Log at build/_deps/wolfssl-src/CMakeLists.txt:169 (find_package):
  find_package considered the following paths for FindThreads.cmake:

    /home/<project-dir>/build/_deps/wolfssl-src/cmake/modules/FindThreads.cmake

  The file was found at

    /nix/store/qacx56cg97hmwgh3gqz9a7qgdw82zmdp-cmake-3.27.7/share/cmake-3.27/Modules/FindThreads.cmake

Generating user options header...
-- BUILD_SHARED_LIBS enabled: wolfcrypt/src/hmac.c;wolfcrypt/src/hash.c;wolfcrypt/src/cpuid.c;wolfcrypt/src/kdf.c;wolfcrypt/src/random.c;wolfcrypt/src/sha256.c;wolfcrypt/src/rsa.c;wolfcrypt/src/sp_int.c;wolfcrypt/src/aes.c;wolfcrypt/src/sha.c;wolfcrypt/src/sha512.c;wolfcrypt/src/sha3.c;wolfcrypt/src/logging.c;wolfcrypt/src/wc_port.c;wolfcrypt/src/error.c;wolfcrypt/src/wc_encrypt.c;wolfcrypt/src/signature.c;wolfcrypt/src/wolfmath.c;wolfcrypt/src/memory.c;wolfcrypt/src/dh.c;wolfcrypt/src/asn.c;wolfcrypt/src/coding.c;wolfcrypt/src/poly1305.c;wolfcrypt/src/md5.c;wolfcrypt/src/pwdbased.c;wolfcrypt/src/pkcs12.c;wolfcrypt/src/chacha.c;wolfcrypt/src/chacha20_poly1305.c;wolfcrypt/src/integer.c;wolfcrypt/src/ecc.c;src/internal.c;src/wolfio.c;src/keys.c;src/ssl.c;src/ocsp.c;src/tls.c;src/tls13.c
-- WOLFSSL_INCLUDE_DIRS is blank. No additional directories will be added.
CMake Debug Log at /nix/store/qacx56cg97hmwgh3gqz9a7qgdw82zmdp-cmake-3.27.7/share/cmake-3.27/Modules/FetchContent.cmake:1618 (find_package):
  find_package considered the following paths for FindGit.cmake:

  The file was found at

    /nix/store/qacx56cg97hmwgh3gqz9a7qgdw82zmdp-cmake-3.27.7/share/cmake-3.27/Modules/FindGit.cmake

Call Stack (most recent call first):
  /nix/store/qacx56cg97hmwgh3gqz9a7qgdw82zmdp-cmake-3.27.7/share/cmake-3.27/Modules/FetchContent.cmake:1802:EVAL:2 (__FetchContent_directPopulate)
  /nix/store/qacx56cg97hmwgh3gqz9a7qgdw82zmdp-cmake-3.27.7/share/cmake-3.27/Modules/FetchContent.cmake:1802 (cmake_language)
  external/CMakeLists.txt:119 (FetchContent_Populate)

-- Found PkgConfig: /nix/store/3ck7yhjzm3h5m3984ni6mj6nbq4saya0-pkg-config-wrapper-0.29.2/bin/pkg-config (found version "0.29.2")
CMake Debug Log at build/_deps/jwt-cpp-src/CMakeLists.txt:60 (find_package):
  find_package considered the following paths for FindPkgConfig.cmake:

    /home/<project-dir>/build/_deps/jwt-cpp-src/cmake/FindPkgConfig.cmake

  The file was found at

    /nix/store/qacx56cg97hmwgh3gqz9a7qgdw82zmdp-cmake-3.27.7/share/cmake-3.27/Modules/FindPkgConfig.cmake

-- Checking for module 'wolfssl'
--   No package 'wolfssl' found
CMake Error at /nix/store/qacx56cg97hmwgh3gqz9a7qgdw82zmdp-cmake-3.27.7/share/cmake-3.27/Modules/FindPkgConfig.cmake:607 (message):
  A required package was not found
Call Stack (most recent call first):
  /nix/store/qacx56cg97hmwgh3gqz9a7qgdw82zmdp-cmake-3.27.7/share/cmake-3.27/Modules/FindPkgConfig.cmake:829 (_pkg_check_modules_internal)
  build/_deps/jwt-cpp-src/CMakeLists.txt:61 (pkg_check_modules)

```

Unless we are also talking about some output that's before?

Here's a basic repro with only these two: https://github.com/dich0tomy/repro

lealem47, 2024-03-21:
Hi @Dich0tomy,

Your use of `FetchContent_Declare` pulls in our source code to be built alongside your application but the lines below in jwt-cpp's CMakeLists.txt file assumes that wolfSSL is already installed on the system.
```
elseif(${JWT_SSL_LIBRARY} MATCHES "wolfSSL")
  find_package(PkgConfig REQUIRED)
  pkg_check_modules(wolfssl REQUIRED IMPORTED_TARGET wolfssl)
  list(TRANSFORM wolfssl_INCLUDE_DIRS APPEND "/wolfssl") # This is required to access OpenSSL compatibility API
endif()
```

You will either have to install wolfSSL onto the system before building jwt-cpp, or patch the CMakeLists.txt file so that it allows jwt-cpp and wolfSSL to be built alongside each other (remove the pkg check and instead use `target_link_libraries` to link to wolfSSL).

closed 2024-05-09 by Dich0tomy