wolfSSL/wolfssl #7344: [Suggestion] Consider defaulting WOLFSSL_ALT_CERT_CHAINS to ON ( Failed to verify CA from chain error )

RipleyTom, 2024-03-17:
### Version

v5.6.6

### Description

Recently Cloudflare changed their certificates(https://blog.cloudflare.com/upcoming-lets-encrypt-certificate-chain-change-and-impact-for-cloudflare-customers). This resulted in:
```
E NET: WOLFSSL: 1 : No CA signer to verify with
E NET: WOLFSSL: 1 : Failed to verify CA from chain
E NET: WOLFSSL: 0 : wolfSSL error occurred, error = -188
E NET: WOLFSSL: 2 : wolfSSL Entering SendAlert
E NET: WOLFSSL: 1 : SendAlert: 48 unknown_ca
```

The default certificate chain fails on "ISRG Root X1" referencing old DST X3 certificate which are not on modern trust stores and are not in the chain(Cloudflare does this for backward compatibility with older devices having the DST X3 certificate in their trust chain).

Enabling WOLFSSL_ALT_CERT_CHAINS fixes this as other chains are valid.
It took me a while to figure out and I thought it'd both be useful to post here if anyone else has the issue and to suggest setting it to ON by default as Cloudflare is a bit too big to have a ssl library fail with default config to connect to it?

dgarske, 2024-03-17:
Thank you for the suggestion. We are doing this. See https://github.com/wolfSSL/wolfssl/pull/7317

anhu, 2024-04-05:
Hi @RipleyTom , 

Please see the final comment in https://github.com/wolfSSL/wolfssl/pull/7317 . Unfortunately, I've abandoned the effort due to memory requirement side effects.  Of course this issue will remain in our issues log so people might be able to find it and learn your solution. 

I will now close this issue. 

Warm regards, Anthony

closed 2024-04-05 by anhu