wolfSSL/wolfssl #7384: DTLS1.3 and PQC not working as TLS1.3

RomainH27, 2024-04-02:
### Version

5.7.0-stable

### Description

Hello,

I was trying the Post Quantum Cryptography with WolfSSL and TLS1.3. 
I followed the installation steps and I managed to get the TLS1.3 client and server exemple running with PQC and PQKeys. I even checked in wireshark just to be sure but it is all working !
I read [here](https://www.wolfssl.com/whats-new-dtls-1-3/) that post quantum cryptography should work with DTLS1.3 as well.
So I tried to follow the few steps needed, at least I think, to make PQC work with DTLS1.3 (starting from [this exemple](https://github.com/wolfSSL/wolfssl-examples/blob/master/dtls/server-dtls13.c)).
I used the same certificates and keys than in the TLS1.3 client/server that I was able to run before.
I then added this to both the client and server :

```
# Added just after the WolfSSL_new()

ret = wolfSSL_UseKeyShare(ssl, WOLFSSL_P521_KYBER_LEVEL5);
if (ret < 0) {
    fprintf(stderr, "ERROR: failed to set the requested group to "
                    "P521_KYBER_LEVEL5.\n");
    ret = -1; goto cleanup;
}
```
My issue is that it still uses "SECP256R1" instead of "WOLFSSL_P521_KYBER_LEVEL5".
I tried to load each other a custom certificate, I double checked in the TLS1.3 version and it works.
I also considered the WolfSSL_set_cipher_list(), but it did not work and I think it is fine because there is not and PQ suite yet.

Would someone have a solution for this ?

RomainH27, 2024-04-04:
It appears that it's a server issue. When I run the client and tracks in the logs what the function just described above does, I can clearly see, just as in the TLS1.3 versions : 
```
wolfSSL Entering EccMakeKey
wolfSSL Leaving EccMakeKey, return 0
```
But, although the server and the client both return a success for this function, the DTLS 1.3 server has none of those lines in the logs. Maybe it is a bug or maybe I am doing something wrong here.

anhu, 2024-04-04:
Hi, 
Not exactly sure what you did or how you built so I will share what works for me. 

I build on linux like this: 
```
./configure --enable-experimental --with-liboqs --enable-dtls13 --enable-dtls --enable-dtls-frag-ch
make all check
```
Here is the command and resulting output for the example server: 
```
$ ./examples/server/server -u -v 4 --pqc P521_KYBER_LEVEL5
Using Post-Quantum KEM: P521_KYBER_LEVEL5
SSL version is DTLSv1.3
SSL cipher suite is TLS_AES_128_GCM_SHA256
SSL curve name is P521_KYBER_LEVEL5
Client message: hello wolfssl!
```

Here is the command and resulting output for the example server: 
```
$ ./examples/client/client  -u -v 4 --pqc P521_KYBER_LEVEL5
Using Post-Quantum KEM: P521_KYBER_LEVEL5
SSL version is DTLSv1.3
SSL cipher suite is TLS_AES_128_GCM_SHA256
SSL curve name is P521_KYBER_LEVEL5
I hear you fa shizzle!
```

I hope this helps!!  Can you please let us know about yourself and your project?  We love knowing how people use wolfSSL.  Anything you would care to share would be appreciated. 

Warm regards, Anthony

RomainH27, 2024-04-04:
Thank you for this answer. It worked for me as well, but what I am trying to do is to implement a DTLS1.3 client/server in C, that is why I made those tests and changes.
I am just a student working on post quantum cryptography and its integration in TLS and DTLS 1.3 for example !

anhu, 2024-04-04:
Hi @RomainH27 , 

examples/client/client and examples/server/server are both written in C.  Is there something about them that makes them unsatisfactory for your purpose? 

May I ask what school you attend?

RomainH27, 2024-04-05:
Hi,
No they are perfect I was not thinking the right way ! Thanks a lot 
It's a french school in Le Mans called ENSIM.

anhu, 2024-04-05:
Excellent.  I'm glad that you are happy with our example client and server.  

I will now proceed to close this issue.  That said, we always love knowing more about projects that are underway at academic institutions.  May I ask, is this a funded effort?  If you would like to share more information with me about you and your project, you can reach me at anthony@wolfssl.com. 

Warm regards, Anthony

closed 2024-04-05 by anhu