wolfSSL/wolfssl #7390: [Bug]: wolfSSL QUIC handshake failure

icing, 2024-04-03:
### Contact Details

stefan.eissing@gmail.com

### Version

all

### Description

As reported  in https://github.com/ngtcp2/ngtcp2/issues/1193. Fixing the returned ciper in https://github.com/wolfSSL/wolfssl/blob/c768f76d5a72e77507241c7b7d0666a846a0d631/src/quic.c#L997
from wolfSSL_EVP_aes_128_ctr to wolfSSL_EVP_aes_128_ccm fixes the assertion failure, but handshake still fails for some reason.

nbars, 2024-04-04:
Hey,

Does this mean that instead of using the CCM mode, the CTR mode was always used?

julek-wolfssl, 2024-04-24:
Hi @icing 

I cleaned up the EVP layer usage in quic in a PR here: https://github.com/wolfSSL/wolfssl/pull/7465. Thanks for the report.

Juliusz

closed 2024-04-25 by SparkiDev

icing, 2024-04-26:
QUIC handshake is now failing in tests at ngtcp2 and curl CI. Easiest to reproduce probably in ngtcp2.

Build https://github.com/ngtcp2/ngtcp2 as in the instructions with wolfssl and another tls lib (openssl-quictls or gnutls or boring). Run pytest:

```
> pytest -v -k test_01
examples/tests/test_01_handshake.py::TestHandshake::test_01_01_get[quictls-quictls] PASSED                                                                                                     [  6%]
examples/tests/test_01_handshake.py::TestHandshake::test_01_01_get[quictls-boringssl] PASSED                                                                                                   [ 12%]
examples/tests/test_01_handshake.py::TestHandshake::test_01_01_get[quictls-picotls] PASSED                                                                                                     [ 18%]
examples/tests/test_01_handshake.py::TestHandshake::test_01_01_get[quictls-wolfssl] FAILED                                                                                                     [ 25%]
examples/tests/test_01_handshake.py::TestHandshake::test_01_01_get[boringssl-quictls] PASSED                                                                                                   [ 31%]
examples/tests/test_01_handshake.py::TestHandshake::test_01_01_get[boringssl-boringssl] PASSED                                                                                                 [ 37%]
examples/tests/test_01_handshake.py::TestHandshake::test_01_01_get[boringssl-picotls] PASSED                                                                                                   [ 43%]
examples/tests/test_01_handshake.py::TestHandshake::test_01_01_get[boringssl-wolfssl] FAILED                                                                                                   [ 50%]
examples/tests/test_01_handshake.py::TestHandshake::test_01_01_get[picotls-quictls] PASSED                                                                                                     [ 56%]
examples/tests/test_01_handshake.py::TestHandshake::test_01_01_get[picotls-boringssl] PASSED                                                                                                   [ 62%]
examples/tests/test_01_handshake.py::TestHandshake::test_01_01_get[picotls-picotls] PASSED                                                                                                     [ 68%]
...
```

 You see that wolfssl can only connect if it is used in client *and* server. The problem seems to be cipher selection as in the logs you'll see ` pkt packet has incorrect reserved bits...pkt could not decrypt packet payload"