wolfSSL/wolfssl #7391: [Bug, openSSL compatibility]:

laurenzfg, 2024-04-03:
### Contact Details

my first name is laurenz. E-Mail [firstname].grote@rwth-aachen.de

### Version

reproduced in current HEAD c768f76d5a72e77507241c7b7d0666a846a0d63

### Description

Hello!

First and foremost thank you for the support provided so far. I am writing about an inconsistency with OpenSSL. In Wolf the function `wolfSSL_CTX_set1_groups_list` only accepts NIST curves (e.g. prime256) , but not Bernstein curves such as  `X25519`. This is because `populate_groups` in `ssl.c` refers to the dictionary `kNistCurves` in the same file, which lacks the Bernstein curves.
The analogous function `SSL_CTX_set1_groups_list` in OpenSSL accepts X25519, as also listed in their documentation: https://www.openssl.org/docs/manmaster/man3/SSL_CTX_set1_groups_list.html`

I would suggest to extend the dicitionary by the Non-NIST Curves such that `wolfSSL_CTX_set1_groups_list` a) is compatible to OpenSSL and b) `wolfSSL_CTX_set1_groups_list` can supersede `wolfSSL_CTX_set1_curves_list`.

My configuration is:

./autogen.sh && ./configure --with-liboqs=/liboqs --enable-nullcipher --enable-psk --enable-opensslextra --enable-debug --enable-tls13 --enable-ecccustcurves  --enable-brainpool --enable-curve25519 --enable-ed25519 --enable-curve448 --enable-ed448 CFLAGS="-DWOLFSSL_STATIC_RSA -DWOLFSSL_STATIC_DH -DKEEP_PEER_CERT -DHAVE_SECRET_CALLBACK -DHAVE_SUPPORTED_CURVES"

Many Thanks

Cheers

Laurenz

### Reproduction steps

Apply 
`std::string name = "X25519"; wolfSSL_CTX_set1_curves_list(ctx, name.c_str())`. Works!

Apply
`std::string name = "X25519"; wolfSSL_CTX_set1_groups_list(ctx, name.c_str()))`. Error `Unrecognized curve name in list`!

But (in OpenSSL 3.1.1)
`std::string name = "X25519"; SSL_CTX_set1_groups_list(ctx, name.c_str()))`. Works!

laurenzfg, 2024-04-03:
For the time being, I use this pattern as a bug fix:

` if ((wolfSSL_CTX_set1_curves_list(ctx, mappedName.c_str()) || wolfSSL_CTX_set1_groups_list(ctx, (char*) mappedName.c_str()))!= 1) {
`

julek-wolfssl, 2024-04-25:
Hi @laurenzfg 

thank you for the report. A fix and refactor will be included in https://github.com/wolfSSL/wolfssl/pull/7446.

Juliusz