wolfSSL/wolfssl #7414: [Bug]: X509_NAME_cmp returns != 0 for subject names that only differ on capitalization

colmenero, 2024-04-11:
### Contact Details

colmenero.sendra@gmail.com

### Version

5.5.1 (but I tested it on master, see reproducer)

### Description

The [X509 RFC](https://www.rfc-editor.org/rfc/rfc5280) states in [section 4.1.2.6](https://www.rfc-editor.org/rfc/rfc5280#section-4.1.2.6):

> Where it is non-empty, the subject field MUST contain an X.500 distinguished name (DN). Implementations of this specification MAY use the comparison rules in Section 7.1[...]

In that section we have:

> Comparisons of domainComponent attributes MUST be performed as specified in Section 7.3.

And in that one:

> Conforming implementations shall perform a case-insensitive exact match when comparing domainComponent attributes in distinguished names

So I would understand that capitalization does not matter when comparing subject names. For example, [OpenSSL's `X509_NAME_cmp` compares the canonical encoding representation, which is lowercase](https://github.com/openssl/openssl/blob/491bbb444c4b654de14adc7031eb44e88a101edb/crypto/x509/x509_cmp.c#L283). On the other hand, [wolfSSL compares directly the subject name strings](https://github.com/wolfSSL/wolfssl/blob/ff09f418c08f9a08019b6c3e215c617aedd97fc6/src/x509.c#L11162). There is a difference in behavior from both crypto libraries.

### Reproduction steps

[x509_name_cmp.zip](https://github.com/wolfSSL/wolfssl/files/14948773/x509_name_cmp.zip)

I've attached a reproducer that reads two certificates that have subject names only differing in capitalization. The reproducer then compares the subject names. You can see that OpenSSL returns that the subject names are equal, but wolfSSL says that they are different.

anhu, 2024-04-12:
Hi @colmenero , 

I was able to reproduce and fix the problem very easily thanks to your excellent reproducer.  I didn't even need to rebuild the applications!

Can you please let me know a bit about yourself and your project?  What are your goals?  What are you trying to achieve?  Is this for personal, professional or academic interest?  Any information you would be willing to share would be greatly appreciated.  

Warm regards, Anthony

colmenero, 2024-04-15:
Hi @anhu,

I'm glad to hear that it was issue to reproduce and fix.

This is for professional interest. I work for a company that develops a product that can use either OpenSSL or wolfSSL as crypto library. In fact, I believe we have wolfSSL support because we've previously contacted you through another person in the organization. In any case, you are always very responsive in GitHub and this was a small fix so I opening a bug report seemed like a good option.

Thanks for the help!

closed 2024-05-15 by SparkiDev