wolfSSL/wolfssl #7425: [Bug]: CMakeLists.txt is out of sync with latest `configure.ac`

iyanmv, 2024-04-15:
### Version

5.7.0

### Description

I maintain a [non-official package](https://aur.archlinux.org/packages/wolfssl-liboqs) for Arch Linux in the user repository ([AUR](https://aur.archlinux.org/)). I prefer using `cmake` and with the latest version I noticed that there is a mismatched between the configure flags and the `CMakeLists.txt`.

For example, this function to build solfSSL used to work but now fails with the latest version:

```bash
build() {
    local cmake_options=(
        -DCMAKE_INSTALL_PREFIX=/usr
        -DCMAKE_BUILD_TYPE=None
        -DWOLFSSL_CURVE25519=ON
        -DWOLFSSL_CURVE448=ON
        -DWOLFSSL_ED25519=ON
        -DWOLFSSL_ED448=ON
        -DWOLFSSL_REPRODUCIBLE_BUILD=ON
        -DWOLFSSL_OQS=ON
        -DWOLFSSL_EXAMPLES=ON
        -DWOLFSSL_CRYPT_TESTS=ON
        -DWARNING_C_FLAGS="$CFLAGS"
        -Wno-dev
        -B build
        -S $_pkgname-$pkgver-stable
    )

    cmake "${cmake_options[@]}"
    cmake --build build --verbose
}
```

The problem is that version 5.7.0 introduced the new `--enable-experimental` flag, which is required to use the option `WOLFSSL_OQS`

### Reproduction steps

```bash
cmake -Wno-dev -B build \
    -DCMAKE_INSTALL_PREFIX=/usr \
    -DCMAKE_BUILD_TYPE=None \
    -DWOLFSSL_CURVE25519=ON \
    -DWOLFSSL_CURVE448=ON \
    -DWOLFSSL_ED25519=ON \
    -DWOLFSSL_ED448=ON \
    -DWOLFSSL_REPRODUCIBLE_BUILD=ON \
    -DWOLFSSL_OQS=ON \
    -DWOLFSSL_EXAMPLES=ON \
    -DWOLFSSL_CRYPT_TESTS=ON
cmake --build build --verbose
```

### Relevant log output

```shell
In file included from /build/wolfssl-liboqs/src/wolfssl-5.7.0-stable/wolfssl/wolfcrypt/wc_port.h:29,
                 from /build/wolfssl-liboqs/src/wolfssl-5.7.0-stable/wolfcrypt/src/kdf.c:27:
/build/wolfssl-liboqs/src/wolfssl-5.7.0-stable/wolfssl/wolfcrypt/settings.h:3271:6: error: #error Experimental settings without WOLFSSL_EXPERIMENTAL_SETTINGS
 3271 |     #error Experimental settings without WOLFSSL_EXPERIMENTAL_SETTINGS
      |      ^~~~~
In file included from /build/wolfssl-liboqs/src/wolfssl-5.7.0-stable/wolfcrypt/src/sha256.c:45:
/build/wolfssl-liboqs/src/wolfssl-5.7.0-stable/wolfssl/wolfcrypt/settings.h:3271:6: error: #error Experimental settings without WOLFSSL_EXPERIMENTAL_SETTINGS
 3271 |     #error Experimental settings without WOLFSSL_EXPERIMENTAL_SETTINGS
      |      ^~~~~
In file included from /build/wolfssl-liboqs/src/wolfssl-5.7.0-stable/wolfssl/wolfcrypt/wc_port.h:29,
                 from /build/wolfssl-liboqs/src/wolfssl-5.7.0-stable/wolfcrypt/src/hmac.c:27:
/build/wolfssl-liboqs/src/wolfssl-5.7.0-stable/wolfssl/wolfcrypt/settings.h:3271:6: error: #error Experimental settings without WOLFSSL_EXPERIMENTAL_SETTINGS
 3271 |     #error Experimental settings without WOLFSSL_EXPERIMENTAL_SETTINGS
      |      ^~~~~
In file included from /build/wolfssl-liboqs/src/wolfssl-5.7.0-stable/wolfcrypt/src/hash.c:27:
/build/wolfssl-liboqs/src/wolfssl-5.7.0-stable/wolfssl/wolfcrypt/settings.h:3271:6: error: #error Experimental settings without WOLFSSL_EXPERIMENTAL_SETTINGS
 3271 |     #error Experimental settings without WOLFSSL_EXPERIMENTAL_SETTINGS
      |      ^~~~~
In file included from /build/wolfssl-liboqs/src/wolfssl-5.7.0-stable/wolfcrypt/src/random.c:32:
/build/wolfssl-liboqs/src/wolfssl-5.7.0-stable/wolfssl/wolfcrypt/settings.h:3271:6: error: #error Experimental settings without WOLFSSL_EXPERIMENTAL_SETTINGS
 3271 |     #error Experimental settings without WOLFSSL_EXPERIMENTAL_SETTINGS
      |      ^~~~~
In file included from /build/wolfssl-liboqs/src/wolfssl-5.7.0-stable/wolfcrypt/src/rsa.c:33:
/build/wolfssl-liboqs/src/wolfssl-5.7.0-stable/wolfssl/wolfcrypt/settings.h:3271:6: error: #error Experimental settings without WOLFSSL_EXPERIMENTAL_SETTINGS
 3271 |     #error Experimental settings without WOLFSSL_EXPERIMENTAL_SETTINGS
      |      ^~~~~
In file included from /build/wolfssl-liboqs/src/wolfssl-5.7.0-stable/wolfcrypt/src/cpuid.c:27:
/build/wolfssl-liboqs/src/wolfssl-5.7.0-stable/wolfssl/wolfcrypt/settings.h:3271:6: error: #error Experimental settings without WOLFSSL_EXPERIMENTAL_SETTINGS
 3271 |     #error Experimental settings without WOLFSSL_EXPERIMENTAL_SETTINGS
      |      ^~~~~
In file included from /build/wolfssl-liboqs/src/wolfssl-5.7.0-stable/wolfcrypt/src/sp_int.c:33:
/build/wolfssl-liboqs/src/wolfssl-5.7.0-stable/wolfssl/wolfcrypt/settings.h:3271:6: error: #error Experimental settings without WOLFSSL_EXPERIMENTAL_SETTINGS
 3271 |     #error Experimental settings without WOLFSSL_EXPERIMENTAL_SETTINGS
      |      ^~~~~
make[2]: *** [CMakeFiles/wolfssl.dir/build.make:104: CMakeFiles/wolfssl.dir/wolfcrypt/src/cpuid.c.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[2]: *** [CMakeFiles/wolfssl.dir/build.make:118: CMakeFiles/wolfssl.dir/wolfcrypt/src/kdf.c.o] Error 1
make[2]: *** [CMakeFiles/wolfssl.dir/build.make:132: CMakeFiles/wolfssl.dir/wolfcrypt/src/random.c.o] Error 1
make[2]: *** [CMakeFiles/wolfssl.dir/build.make:76: CMakeFiles/wolfssl.dir/wolfcrypt/src/hmac.c.o] Error 1
make[2]: *** [CMakeFiles/wolfssl.dir/build.make:146: CMakeFiles/wolfssl.dir/wolfcrypt/src/sha256.c.o] Error 1
make[2]: *** [CMakeFiles/wolfssl.dir/build.make:160: CMakeFiles/wolfssl.dir/wolfcrypt/src/rsa.c.o] Error 1
make[2]: *** [CMakeFiles/wolfssl.dir/build.make:90: CMakeFiles/wolfssl.dir/wolfcrypt/src/hash.c.o] Error 1
make[2]: *** [CMakeFiles/wolfssl.dir/build.make:174: CMakeFiles/wolfssl.dir/wolfcrypt/src/sp_int.c.o] Error 1
make[2]: Leaving directory '/build/wolfssl-liboqs/src/build'
make[1]: *** [CMakeFiles/Makefile2:99: CMakeFiles/wolfssl.dir/all] Error 2
make[1]: Leaving directory '/build/wolfssl-liboqs/src/build'
make: *** [Makefile:146: all] Error 2
```

iyanmv, 2024-04-15:
A quick but incomplete fix

```diff
diff --git a/CMakeLists.txt b/CMakeLists.txt
index 9aee6af3f..2574cdf46 100644
--- a/CMakeLists.txt
+++ b/CMakeLists.txt
@@ -534,22 +534,40 @@ endif()
 
 set(WOLFSSL_SLOW_MATH "yes")
 
+# Experimental features
+
+add_option(WOLFSSL_EXPERIMENTAL
+    "Enable experimental features (default: disabled)"
+    "no" "yes;no")
+
+# TODO: - Kyber
+#       - XMSS
+#       - LMS
+#       - dual-certs
+
 # liboqs
 add_option(WOLFSSL_OQS
     "Enable integration with the OQS (Open Quantum Safe) liboqs library (default: disabled)"
     "no" "yes;no")
 
-if (WOLFSSL_OQS)
-    find_package(OQS)
+if (WOLFSSL_EXPERIMENTAL)
 
-    if (OQS_FOUND)
-        list(APPEND WOLFSSL_LINK_LIBS ${OQS_LIBRARY})
-        list(APPEND WOLFSSL_INCLUDE_DIRS ${OQS_INCLUDE_DIR})
-        set(HAVE_LIBOQS 1)
-        list(APPEND WOLFSSL_DEFINITIONS
-          "-DHAVE_TLS_EXTENSIONS"
-          "-DHAVE_LIBOQS")
+    set(WOLFSSL_EXPERIMENTAL_SETTINGS 1)
+    list(APPEND WOLFSSL_DEFINITIONS "-DWOLFSSL_EXPERIMENTAL_SETTINGS")
+
+    if (WOLFSSL_OQS)
+        find_package(OQS)
+
+        if (OQS_FOUND)
+            list(APPEND WOLFSSL_LINK_LIBS ${OQS_LIBRARY})
+            list(APPEND WOLFSSL_INCLUDE_DIRS ${OQS_INCLUDE_DIR})
+            set(HAVE_LIBOQS 1)
+            list(APPEND WOLFSSL_DEFINITIONS
+            "-DHAVE_TLS_EXTENSIONS"
+            "-DHAVE_LIBOQS")
+        endif()
     endif()
+
 endif()
 
 # TODO: - Lean PSK
```

gojimmypi, 2024-04-16:
Hi @iyanmv and thank you for providing the details.

Indeed I've seen the error you mentioned, as I recently added Kyber support to the Espressif examples in https://github.com/wolfSSL/wolfssl/pull/7392 that needs `WOLFSSL_EXPERIMENTAL_SETTINGS`.

I tried running your `Reproduction Steps` code:

```
cmake -Wno-dev -B build \
    -DCMAKE_INSTALL_PREFIX=/usr \
    -DCMAKE_BUILD_TYPE=None \
    -DWOLFSSL_CURVE25519=ON \
    -DWOLFSSL_CURVE448=ON \
    -DWOLFSSL_ED25519=ON \
    -DWOLFSSL_ED448=ON \
    -DWOLFSSL_REPRODUCIBLE_BUILD=ON \
    -DWOLFSSL_OQS=ON \
    -DWOLFSSL_EXAMPLES=ON \
    -DWOLFSSL_CRYPT_TESTS=ON
cmake --build build --verbose
```

But I did not see the error. I did however encounter a `Could NOT find OQS (missing: OQS_INCLUDE_DIR OQS_LIBRARY)` that was quietly display and was otherwise successful:

```
$ cmake -Wno-dev -B build \
    -DCMAKE_INSTALL_PREFIX=/usr \
    -DCMAKE_BUILD_TYPE=None \
    -DWOLFSSL_CURVE25519=ON \
    -DWOLFSSL_CURVE448=ON \
    -DWOLFSSL_ED25519=ON \
    -DWOLFSSL_ED448=ON \
    -DWOLFSSL_REPRODUCIBLE_BUILD=ON \
    -DWOLFSSL_OQS=ON \
    -DWOLFSSL_EXAMPLES=ON \
    -DWOLFSSL_CRYPT_TESTS=ON
-- The C compiler identification is GNU 11.4.0
-- The ASM compiler identification is GNU
-- Found assembler: /usr/bin/cc
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Found WOLFSSL_ROOT = /mnt/c/workspace/wolfssl-gojimmypi-pr
-- Looking for arpa/inet.h

[ .. snip .. ]

-- Found Threads: TRUE
-- Could NOT find OQS (missing: OQS_INCLUDE_DIR OQS_LIBRARY)
-- Generating user options header...
-- BUILD_SHARED_LIBS enabled: wolfcrypt/src/hmac.c;wolfcrypt/src/hash.c;wolfcrypt/src/cpuid.c;wolfcrypt/src/kdf.c;wolfcrypt/src/random.c;wolfcrypt/src/sha256.c;wolfcrypt/src/rsa.c;wolfcrypt/src/sp_int.c;wolfcrypt/src/aes.c;wolfcrypt/src/sha.c;wolfcrypt/src/sha512.c;wolfcrypt/src/sha3.c;wolfcrypt/src/logging.c;wolfcrypt/src/wc_port.c;wolfcrypt/src/error.c;wolfcrypt/src/port/liboqs/liboqs.c;wolfcrypt/src/wc_encrypt.c;wolfcrypt/src/signature.c;wolfcrypt/src/wolfmath.c;wolfcrypt/src/memory.c;wolfcrypt/src/dh.c;wolfcrypt/src/asn.c;wolfcrypt/src/coding.c;wolfcrypt/src/poly1305.c;wolfcrypt/src/md5.c;wolfcrypt/src/pwdbased.c;wolfcrypt/src/pkcs12.c;wolfcrypt/src/chacha.c;wolfcrypt/src/chacha20_poly1305.c;wolfcrypt/src/integer.c;wolfcrypt/src/ecc.c;wolfcrypt/src/curve25519.c;wolfcrypt/src/ed25519.c;wolfcrypt/src/fe_operations.c;wolfcrypt/src/ge_operations.c;wolfcrypt/src/curve448.c;wolfcrypt/src/ed448.c;wolfcrypt/src/fe_448.c;wolfcrypt/src/ge_448.c;wolfcrypt/src/falcon.c;wolfcrypt/src/sphincs.c;wolfcrypt/src/dilithium.c;wolfcrypt/src/ext_kyber.c;src/internal.c;src/wolfio.c;src/keys.c;src/ssl.c;src/ocsp.c;src/tls.c;src/tls13.c
-- WOLFSSL_INCLUDE_DIRS is blank. No additional directories will be added.
-- Configuring done
-- Generating done
-- Build files have been written to: /mnt/c/workspace/wolfssl-gojimmypi-pr/build
gojimmypi:/mnt/c/workspace/wolfssl-gojimmypi-pr
$ cmake --build build --verbose
/usr/bin/cmake -S/mnt/c/workspace/wolfssl-gojimmypi-pr -B/mnt/c/workspace/wolfssl-gojimmypi-pr/build --check-build-system CMakeFiles/Makefile.cmake 0
/usr/bin/cmake -E cmake_progress_start /mnt/c/workspace/wolfssl-gojimmypi-pr/build/CMakeFiles /mnt/c/workspace/wolfssl-gojimmypi-pr/build//CMakeFiles/progress.marks
/usr/bin/gmake  -f CMakeFiles/Makefile2 all

[ .. snip .. ]

gmake[2]: Leaving directory '/mnt/c/workspace/wolfssl-gojimmypi-pr/build'
/usr/bin/gmake  -f CMakeFiles/wolfcryptbench.dir/build.make CMakeFiles/wolfcryptbench.dir/build
gmake[2]: Entering directory '/mnt/c/workspace/wolfssl-gojimmypi-pr/build'
[ 98%] Building C object CMakeFiles/wolfcryptbench.dir/wolfcrypt/benchmark/benchmark.c.o
/usr/bin/cc -DECC_SHAMIR -DECC_TIMING_RESISTANT -DGCM_TABLE_4BIT -DHAVE_AESGCM -DHAVE_CHACHA -DHAVE_CONFIG_H -DHAVE_CURVE25519 -DHAVE_CURVE448 -DHAVE_DH_DEFAULT_PARAMS -DHAVE_ECC -DHAVE_ED25519 -DHAVE_ED448 -DHAVE_ENCRYPT_THEN_MAC -DHAVE_EXTENDED_MASTER -DHAVE_FFDHE_2048 -DHAVE_HASHDRBG -DHAVE_HKDF -DHAVE_ONE_TIME_AUTH -DHAVE_POLY1305 -DHAVE_PTHREAD -DHAVE_REPRODUCIBLE_BUILD -DHAVE_SNI -DHAVE_SUPPORTED_CURVES -DHAVE_THREAD_LS -DHAVE_TLS_EXTENSIONS -DHAVE___UINT128_T -DNO_DES3 -DNO_DES3_TLS_SUITES -DNO_DSA -DNO_MD4 -DNO_PSK -DNO_RC4 -DTFM_ECC256 -DTFM_TIMING_RESISTANT -DWC_NO_ASYNC_THREADING -DWC_RSA_BLINDING -DWC_RSA_PSS -DWOLFSSL_BASE64_ENCODE -DWOLFSSL_DLL -DWOLFSSL_IGNORE_FILE_WARN -DWOLFSSL_NO_SHAKE128 -DWOLFSSL_PSS_LONG_SALT -DWOLFSSL_SHA224 -DWOLFSSL_SHA3 -DWOLFSSL_SHA384 -DWOLFSSL_SHA512 -DWOLFSSL_SHAKE256 -DWOLFSSL_SYS_CA_CERTS -DWOLFSSL_TLS13 -DWOLFSSL_USE_ALIGN -DWOLFSSL_X86_64_BUILD -D_POSIX_THREADS -I/mnt/c/workspace/wolfssl-gojimmypi-pr/build -I/mnt/c/workspace/wolfssl-gojimmypi-pr -Wall -Wextra -Wno-unused -Werror  -MD -MT CMakeFiles/wolfcryptbench.dir/wolfcrypt/benchmark/benchmark.c.o -MF CMakeFiles/wolfcryptbench.dir/wolfcrypt/benchmark/benchmark.c.o.d -o CMakeFiles/wolfcryptbench.dir/wolfcrypt/benchmark/benchmark.c.o -c /mnt/c/workspace/wolfssl-gojimmypi-pr/wolfcrypt/benchmark/benchmark.c
[100%] Linking C executable wolfcrypt/benchmark/benchmark
/usr/bin/cmake -E cmake_link_script CMakeFiles/wolfcryptbench.dir/link.txt --verbose=1
/usr/bin/cc -Wall -Wextra -Wno-unused -Werror  CMakeFiles/wolfcryptbench.dir/wolfcrypt/benchmark/benchmark.c.o -o wolfcrypt/benchmark/benchmark  -Wl,-rpath,/mnt/c/workspace/wolfssl-gojimmypi-pr/build libwolfssl.so [... cut ...]
AES-128-CBC-enc            340 MiB took 1.002 seconds,  339.281 MiB/s Cycles per byte =   8.20
AES-128-CBC-dec            395 MiB took 1.006 seconds,  392.525 MiB/s Cycles per byte =   7.09
AES-192-CBC-enc            120 MiB took 1.015 seconds,  118.209 MiB/s Cycles per byte =  23.54
[ .. etc .. ]

```

Pardon me for jumping in, as this is a bit out of my wheelhouse as an embedded developer, but I'm interested in the OQS & CMake features & would like to help.

Can you please confirm you are using the very latest code from the `master` branch? Thank you.

iyanmv, 2024-04-17:
No, I was using the signed tarball from the releases (version 5.7.0). But I can try from `master` branch. Regarding your error not finding `liboqs`, how did you install the library? I also used cmake, in particular I build with [this](https://aur.archlinux.org/packages/liboqs):

```bash
build() {
    cmake -G Ninja -B build -S "${pkgname}-${pkgver}"\
        -DBUILD_SHARED_LIBS=ON \
        -DCMAKE_BUILD_TYPE=None \
        -DCMAKE_INSTALL_PREFIX="${pkgdir}/usr" \
        -DOQS_ALGS_ENABLED=All \
        -DOQS_BUILD_ONLY_LIB=OFF \
        -DOQS_DIST_BUILD=ON \
        -DOQS_USE_OPENSSL=ON \
        -DOQS_USE_AES_OPENSSL=ON \
        -DOQS_USE_SHA2_OPENSSL=ON \
        -DOQS_USE_SHA3_OPENSSL=ON \
        -DOQS_OPT_TARGET=x86-64 \
        -DOQS_STRICT_WARNINGS=ON \
        -Wno-dev
    ninja -C build
    ninja -C build gen_docs
}

check() {
    ninja -C build run_tests
}

package() {
    ninja -C build install
    install -D -m0644 "${pkgname}-${pkgver}/LICENSE.txt" "${pkgdir}/usr/share/licenses/${pkgname}/LICENSE"
    install -d "${pkgdir}/usr/share/doc/"
    cp -r build/docs/html "${pkgdir}/usr/share/doc/${pkgname}"
}
``

gojimmypi, 2024-04-20:
> how did you install the library?

I used the [liboqs getting started](https://openquantumsafe.org/liboqs/getting-started.html). Previously, I thought it was already installed from my prior work. But in fact, it was not.

Using the latest master branch code, it appears some things might have been addressed since the 5.7.0 release, as I saw only one error, where you saw many:

```text
In file included from /mnt/c/workspace/wolfssl-master/wolfssl/wolfcrypt/wc_port.h:29,
                 from /mnt/c/workspace/wolfssl-master/wolfcrypt/src/hmac.c:27:
/mnt/c/workspace/wolfssl-master/wolfssl/wolfcrypt/settings.h:3341:6: error: #error Experimental settings without WOLFSSL_EXPERIMENTAL_SETTINGS
 3341 |     #error Experimental settings without WOLFSSL_EXPERIMENTAL_SETTINGS
      |      ^~~~~
gmake[2]: *** [CMakeFiles/wolfssl.dir/build.make:76: CMakeFiles/wolfssl.dir/wolfcrypt/src/hmac.c.o] Error 1
```
After I had liboqs properly installed, I was able to add `-DCMAKE_C_FLAGS="-DWOLFSSL_EXPERIMENTAL_SETTINGS"` like this:

```bash
 cmake -Wno-dev -B build \
    -DCMAKE_INSTALL_PREFIX=/usr \
    -DCMAKE_BUILD_TYPE=None \
    -DWOLFSSL_CURVE25519=ON \
    -DWOLFSSL_CURVE448=ON \
    -DWOLFSSL_ED25519=ON \
    -DWOLFSSL_ED448=ON \
    -DWOLFSSL_REPRODUCIBLE_BUILD=ON \
    -DWOLFSSL_OQS=ON \
    -DWOLFSSL_EXAMPLES=ON \
    -DWOLFSSL_CRYPT_TESTS=ON \
    -DCMAKE_C_FLAGS="-DWOLFSSL_EXPERIMENTAL_SETTINGS"
```
Although I suspect that was your point, above, with this:

```
+    set(WOLFSSL_EXPERIMENTAL_SETTINGS 1)
+    list(APPEND WOLFSSL_DEFINITIONS "-DWOLFSSL_EXPERIMENTAL_SETTINGS")
```

It is my understanding that any PQ libraries require the `WOLFSSL_EXPERIMENTAL_SETTINGS` definition for wolfSSL at this time.

I think it would be preferable to keep the experimental settings manual, deliberate, and conspicuously visible. 

Are proposing the  `WOLFSSL_EXPERIMENTAL_SETTINGS` be set quietly and automatically whenever enabling the OQS is enabled?

iyanmv, 2024-04-22:
> I think it would be preferable to keep the experimental settings manual, deliberate, and conspicuously visible.
> 
> Are proposing the `WOLFSSL_EXPERIMENTAL_SETTINGS` be set quietly and automatically whenever enabling the OQS is enabled?

I agree with the first sentence, but not with the second. I don't propose that, what I propose is adding a new cmake option, which has to be enabled explicitly to use any PQ features, so that cmake building is aligned with the new `configure.ac` (instead of having to pass the additional `-DCMAKE_C_FLAGS` as you proposes). What do you think?

gojimmypi, 2024-04-23:
Ah yes, I believe we are in agreement:

- Experimental settings should be manual, deliberate, and conspicuously visible.
- OQS needs to be enabled more consistently in cmake, not with `-DCMAKE_C_FLAGS=...`

I have some proposed changes, not yet in a PR [here](https://github.com/wolfSSL/wolfssl/compare/master...gojimmypi:wolfssl:PR-cmake-liboqs-kyber). Perhaps you can take a look?

The changes add these new cmake commandline options:

-    `-DWOLFSSL_CRYPT_TESTS=ON`
-    `-DWOLFSSL_KYBER=ON`
-    `-DWOLFSSL_EXPERIMENTAL=ON`

My changes are the basically the same as yours (thanks for providing that), along with some additional verbose messages and checks.

One thing I noticed is that `./wolfcrypt/test/testwolfcrypt` does not appear to actually test OQS or Kyber?

When I was adding Kyber support to the Espressif devices, I was testing with manual inspection in Wireshark, and not the `wolfssl_benchmark` nor `wolfssl_test` apps. @kareem-wolfssl: I'm not sure if perhaps new tests are needed to be added? 

Also: It appears the `configure.ac` [needs](https://github.com/wolfSSL/wolfssl/blob/3a89c452b388218dde54d6430df42fff23efc776/configure.ac#L1155) `Shake128` and `Shake256` for Kyber. I only needed to manually enable `SHA3` for Espressif, and only explicitly added `SHA3` in my proposed cmake changes. (even though it looks like it is [enabled by default](https://github.com/wolfSSL/wolfssl/blob/3a89c452b388218dde54d6430df42fff23efc776/configure.ac#L3426) for some architectures)

The same with `liboqs`: it appears that `OPENSSL_EXTRA` may be [enabled](https://github.com/wolfSSL/wolfssl/blob/3a89c452b388218dde54d6430df42fff23efc776/configure.ac#L1143) automatically?

Perhaps @anhu can also take a look at this before I create the PR.

@iyanmv - I see you added some TODO items as well:

```text
+# TODO: - Kyber
+#       - XMSS
+#       - LMS
+#       - dual-certs
```

I also implemented the Kyber one since it is something I'm familiar with, but moved the XMSS, LMS and Dual-Certs to the [existing cmake TODO list](https://github.com/wolfSSL/wolfssl/blob/3a89c452b388218dde54d6430df42fff23efc776/CMakeLists.txt#L555).

Are all of those new TODO items something you need immediately, or just the `liboqs` feature for now?

## Testing

Here is a script I used to test:

```bash
rm -rf ./build

echo "--------------------------------"
echo "No OQS, No Experimental"
echo "--------------------------------"
cmake -Wno-dev -B build \
    -DCMAKE_INSTALL_PREFIX=/usr     \
    -DCMAKE_BUILD_TYPE=None         \
    -DWOLFSSL_CURVE25519=ON         \
    -DWOLFSSL_CURVE448=ON           \
    -DWOLFSSL_ED25519=ON            \
    -DWOLFSSL_ED448=ON              \
    -DWOLFSSL_REPRODUCIBLE_BUILD=ON \
    -DWOLFSSL_EXAMPLES=ON           \
    -DWOLFSSL_CRYPT_TESTS=ON        \
                                      || exit 1
cmake --build build --verbose

rm -rf ./build
echo "--------------------------------"
echo "No OQS, Experimental"
echo "--------------------------------"
cmake -Wno-dev -B build \
    -DCMAKE_INSTALL_PREFIX=/usr     \
    -DCMAKE_BUILD_TYPE=None         \
    -DWOLFSSL_CURVE25519=ON         \
    -DWOLFSSL_CURVE448=ON           \
[... 83 lines cut ...]
                                      || exit 1
cmake --build build --verbose

rm -rf ./build
echo "--------------------------------"
echo "OQS, Kyber, Experimental"
echo "--------------------------------"
cmake -Wno-dev -B build \
    -DCMAKE_INSTALL_PREFIX=/usr     \
    -DCMAKE_BUILD_TYPE=None         \
    -DWOLFSSL_CURVE25519=ON         \
    -DWOLFSSL_CURVE448=ON           \
    -DWOLFSSL_ED25519=ON            \
    -DWOLFSSL_ED448=ON              \
    -DWOLFSSL_REPRODUCIBLE_BUILD=ON \
    -DWOLFSSL_EXAMPLES=ON           \
    -DWOLFSSL_CRYPT_TESTS=ON        \
    -DWOLFSSL_OQS=ON                \
    -DWOLFSSL_KYBER=ON              \
    -DWOLFSSL_EXPERIMENTAL=ON       \
                                      || echo -e "\nOk: failure as expected cannot enable both Kyber and OQS at the same time\n"
cmake --build build --verbose

echo "Success: Kyber and OQS tests completed."
```

gojimmypi, 2024-05-04:
Hi @iyanmv - I've created https://github.com/wolfSSL/wolfssl/pull/7480 to address cmake LIBOQS and Kyber.

Please take it for a test drive and let me know if it works as desired for you.

iyanmv, 2024-05-06:
@gojimmypi Cool, thanks! I will give it a try this evening.

iyanmv, 2024-05-08:
@gojimmypi works perfectly, thanks again!

closed 2024-05-13 by dgarske