wolfSSL/wolfssl #7427: Both `--enable-lms` and `--enable-xmss`?

iyanmv, 2024-04-15:
### Version

5.7.0

### Description

Super interesting to see experimental support for stateful hashed-based signatures. I wanted to test both but it seems like I cannot enable both at the same time, but I couldn't find any note about this in the docs.

I'm compiling with this:

```bash
prepare() {
    cd $_pkgname-$pkgver-stable
    ./autogen.sh

    # https://github.com/wolfSSL/wolfssl-examples/tree/master/pq/stateful_hash_sig
    cd ../xmss-reference
    git apply -v ../wolfssl-xmss.patch
}

build() {
    # Build XMSS library
    cd xmss-reference
    make xmss_lib.a

    # Build wolfSSL
    # cmake is currently broken: https://github.com/wolfSSL/wolfssl/issues/7425
    CFLAGS="${CFLAGS} -I/usr/include/cisco-hash-sigs"
    cd ../$_pkgname-$pkgver-stable
    ./configure --prefix=/usr \
        --enable-experimental \
        --enable-reproducible-build \
        --enable-benchmark \
        --enable-curve25519 \
        --enable-ed25519 \
        --enable-curve448 \
        --enable-ed448 \
        --enable-examples \
        --enable-crypttests \
        --enable-kyber \
        --enable-lms \
        --enable-xmss \
        --enable-dual-alg-certs \
        --with-liboqs=/usr/include \
        --with-liblms=/usr/lib \
        --with-libxmss=../xmss-reference
    make
}
```

And this is the error I get:

```
==> Starting build()...
/usr/bin/gcc -Wall -O2 -Wextra -Wpedantic   -c -o params.o params.c
/usr/bin/gcc -Wall -O2 -Wextra -Wpedantic   -c -o thash.o thash.c
/usr/bin/gcc -Wall -O2 -Wextra -Wpedantic   -c -o hash_address.o hash_address.c
/usr/bin/gcc -Wall -O2 -Wextra -Wpedantic   -c -o wots.o wots.c
/usr/bin/gcc -Wall -O2 -Wextra -Wpedantic   -c -o xmss.o xmss.c
/usr/bin/gcc -Wall -O2 -Wextra -Wpedantic   -c -o xmss_core_fast.o xmss_core_fast.c
/usr/bin/gcc -Wall -O2 -Wextra -Wpedantic   -c -o xmss_commons.o xmss_commons.c
/usr/bin/gcc -Wall -O2 -Wextra -Wpedantic   -c -o utils.o utils.c
ar rcs xmss_lib.a params.o thash.o hash_address.o wots.o xmss.o xmss_core_fast.o xmss_commons.o utils.o
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether the compiler supports GNU C... yes
[... 486 lines cut ...]
In file included from wolfcrypt/benchmark/benchmark.c:178:
./wolfssl/wolfcrypt/lms.h:37:15: note: previous declaration of ‘write_private_key_cb’ with type ‘write_private_key_cb’ {aka ‘int (*)(const unsigned char *, unsigned int,  void *)’}
   37 | typedef int (*write_private_key_cb)(const byte * priv, word32 privSz, void *context);
      |               ^~~~~~~~~~~~~~~~~~~~
./wolfssl/wolfcrypt/xmss.h:165:26: error: conflicting types for ‘read_private_key_cb’; have ‘enum wc_XmssRc (*)(byte *, word32,  void *)’ {aka ‘enum wc_XmssRc (*)(unsigned char *, unsigned int,  void *)’}
  165 | typedef enum wc_XmssRc (*read_private_key_cb)(byte* priv, word32 privSz,
      |                          ^~~~~~~~~~~~~~~~~~~
./wolfssl/wolfcrypt/lms.h:38:15: note: previous declaration of ‘read_private_key_cb’ with type ‘read_private_key_cb’ {aka ‘int (*)(unsigned char *, unsigned int,  void *)’}
   38 | typedef int (*read_private_key_cb)(byte * priv, word32 privSz, void *context);
      |               ^~~~~~~~~~~~~~~~~~~
  CC       wolfcrypt/src/src_libwolfssl_la-hash.lo
  CC       wolfcrypt/src/src_libwolfssl_la-cpuid.lo
  CC       wolfcrypt/src/src_libwolfssl_la-kdf.lo
  CC       wolfcrypt/src/src_libwolfssl_la-random.lo
make[2]: *** [Makefile:6490: wolfcrypt/benchmark/benchmark.o] Error 1
make[2]: *** Waiting for unfinished jobs....
  CC       wolfcrypt/src/src_libwolfssl_la-sha256.lo
make[2]: *** [Makefile:7942: wolfcrypt/test/testsuite_testsuite_test-test.o] Error 1
make[2]: Leaving directory '/build/wolfssl-liboqs/src/wolfssl-5.7.0-stable'
make[1]: *** [Makefile:8182: all-recursive] Error 1
make[1]: Leaving directory '/build/wolfssl-liboqs/src/wolfssl-5.7.0-stable'
make: *** [Makefile:4839: all] Error 2
```

Removing either `--enable-lms` or `--enable-xmss` fixes the problem.

anhu, 2024-06-28:
Hi, 
It seems that https://github.com/wolfSSL/wolfssl/pull/7480 might have fixed this.  Can you please update and see if the problem is still there? 

Warm regards, Anthony

iyanmv, 2024-07-15:
Not really. Now I get the following error:
```
checking for libxmss... configure: error: libxmss isn't found.
            If it's already installed, specify its path using --with-libxmss=/dir/
```
Even though I did use the flag `--with-libxmss=../xmss-reference`.

iyanmv, 2024-07-15:
In addition, cmake build is again broken for me unfortunately (I only did a super quick test)... Although it's okay since I move to autotools anyways.

closed 2025-11-11 by anhu