wolfSSL/wolfssl #7432: Unexpected behavior with --enable-sslextra

0xg0nz0, 2024-04-16:
### Version

v5.7.0-stable

### Description

I found that if you only specify `--enable-opensslextra` in CMake, OpenSSL support is only partially activated and you get compile errors. You need to specify `--enable-opensslall` *and* `--enable-opensslextra` for it to work.

This specifically came up when I was compiling lws with wolfSSL's OpenSSL compatibility layer:

https://libwebsockets.org/lws-api-doc-master/html/md_READMEs_README_8build.html#wolf

I would think that "extra is supported" implies "base support is enabled" and you should not need both, that extra is a superset. If you disagree I'll report this to lws instead as a documentation bug.

kareem-wolfssl, 2024-04-16:
Hi @0xg0nz0 ,

OpenSSL extra is a subset of our OpenSSL compatibility layer APIs, it covers the most commonly used APIs.  OpenSSL all enables all of our OpenSSL compatibility layer APIs.  When building third-party applications like libwebsockets, best practice is to use our dedicated configure argument for the third-party project, in this case `./configure --enable-libwebsockets`.  This will enable OpenSSL all, extra and a few other build flags required for libwebsockets.

0xg0nz0, 2024-04-16:
Excellent, I will raise an issue with lws regarding their documentation. Thank you!

closed 2024-04-16 by 0xg0nz0