wolfSSL/wolfssl #7464: [Bug]: 	C:/Users/EDS/Desktop/wallfssl/wolfssl/IDE/IAR-EWARM/Projects/lib/ewarm/Exe/wolfSSL-Lib.a', needed by 'C:/Users/EDS/Desktop/wallfssl/wolfssl/IDE/IAR-EWARM/Projects/benchmark/ewarm/Exe/benchmark.out', missing and no known rule to make it		when make build of IAR-EWARM then coming this issue file not wolfSSL-Lib.a

SantoshKumarEDS, 2024-04-24:
### Contact Details

santosh.kumar@eds-india.com

### Version

1.3

### Description

issue wolfSSL-Lib.a this file not found

dgarske, 2024-04-24:
Hi @SantoshKumarEDS ,

Please share details for reproducing. We test continually our IAR example on Windows. Did you follow the README? https://github.com/wolfSSL/wolfssl/tree/master/IDE/IAR-EWARM
Did you build the wolfssl-lib.a here?: https://github.com/wolfSSL/wolfssl/blob/master/IDE/IAR-EWARM/Projects/lib/wolfSSL-Lib.ewp

Thanks,
David Garske, wolfSSL

SantoshKumarEDS, 2024-04-24:
[IAR-EWARM.zip](https://github.com/wolfSSL/wolfssl/files/15096356/IAR-EWARM.zip)
@dgarske 

pls check it this

when I try then comes issue on compile time

Thanks,
Santosh Kumar

dgarske, 2024-08-22:
Hi @SantoshKumarEDS ,

Were you able to get the IAR project building? Sorry, I lost track of this issue. If you are still having issues please let me know. 

Thanks,
David Garske, wolfSSL

dgarske, 2024-11-15:
Hi @SantoshKumarEDS ,

Since I haven't heard back from you I will go ahead and close this issue. We have continuous tests for IAR-EWARM so I expect you have been able to resolve this issue.

Thanks,
David Garske, wolfSSL

SantoshKumarEDS, 2024-11-15:
Hi David,

I hope you're doing well.

We are still working on the same project, but I’m happy to inform you that we’ve resolved the issue and everything is now working successfully in IAR-EWARM.

Regarding the time license for CBC encryption, we don’t require it at the moment. If the need arises in the future, I will reach out to you.

Thank you once again for your support.

Best regards,
Santosh Kumar

---

Let me know if you need any more adjustments!

---- On Fri, 15 Nov 2024 22:19:33 +0530 ***@***.*** wrote ----

Hi @SantoshKumarEDS ,

Since I haven't heard back from you I will go ahead and close this issue. We have continuous tests for IAR-EWARM so I expect you have been able to resolve this issue.

Thanks,
David Garske, wolfSSL