wolfSSL/wolfssl #7508: EC_method support missing despite RSA_method being present

schwabe, 2024-05-07:
### Version

master of 2024-05-07

### Description

While removing the OpenSSL 1.0.2 compatibility in OpenVPN (https://gerrit.openvpn.net/c/openvpn/+/559) I noticed that wolfSSL nows fails with missing EC_method support. It seems wolfSSL implements the RSA_method but not the EC_method

julek-wolfssl, 2024-05-07:
Thanks @schwabe for the report. I'll look into fixing this.

Juliusz

julek-wolfssl, 2024-05-08:
Hi @schwabe 
I implemented stubs for the missing API at https://github.com/wolfSSL/wolfssl/pull/7511. I needed to make the following changes to your patchset to get it compiling:
```
diff --git a/src/openvpn/crypto_openssl.c b/src/openvpn/crypto_openssl.c
index 64ad3469..8f2724c1 100644
--- a/src/openvpn/crypto_openssl.c
+++ b/src/openvpn/crypto_openssl.c
@@ -1372,7 +1372,7 @@ out:
 
     return ret;
 }
-#elif !defined(LIBRESSL_VERSION_NUMBER)
+#elif !defined(LIBRESSL_VERSION_NUMBER) && !defined(WOLFSSL_VERSION)
 bool
 ssl_tls1_PRF(const uint8_t *seed, int seed_len, const uint8_t *secret,
              int secret_len, uint8_t *output, int output_len)
diff --git a/src/openvpn/ssl_openssl.c b/src/openvpn/ssl_openssl.c
index efdd8829..3e18a5ba 100644
--- a/src/openvpn/ssl_openssl.c
+++ b/src/openvpn/ssl_openssl.c
@@ -2145,7 +2145,7 @@ print_server_tempkey(SSL *ssl, char *buf, size_t buflen)
     EVP_PKEY_free(pkey);
 }
 
-#if !defined(LIBRESSL_VERSION_NUMBER) \
+#if (!defined(LIBRESSL_VERSION_NUMBER) && OPENSSL_VERSION_NUMBER >= 0x1010000fL) \
     || (defined(LIBRESSL_VERSION_NUMBER) && LIBRESSL_VERSION_NUMBER >= 0x3090000fL)
 /**
  * Translate an OpenSSL NID into a more human readable name
```
Should I submit this patch to the mailing list or can you include it in your patchset?

Juliusz

dgarske, 2024-05-13:
Hi @schwabe ,

The PR #7511 has been merged. Please let us know if you have any further issues. Please mark this closed. Also let us know if you plan on implementing the patch suggested by Juliusz directly or if we should submit to the mailing list.

Thanks,
David Garske, wolfSSL

closed 2024-05-13 by dgarske

schwabe, 2024-05-15:
For the first problem of the TLS prf: https://gerrit.openvpn.net/c/openvpn/+/560 should take of that problem.